realsenseai/librealsense #12703: 10:56:17 [Error] ./src/core/../librealsense-exception.h:52 - xioctl(VIDIOC_G_CTRL) failed Last Error: Invalid argument

danielhertz, 2024-02-26:
| | |
|---|---|
|**librealsense**|2.54.2 RELEASE|
|**OS**|Linux|
|**Name**|Intel RealSense D415|
|**Serial Number**|104122061474|
|**Firmware Version**|5.15.1|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|D400|
|**Asic Serial Number**|108123023920|
|**Firmware Update Id**|108123023920|

Please provide a description of the problem

MartyG-RealSense, 2024-02-26:
Hi @danielhertz  A **xioctl(VIDIOC_G_CTRL)** error usually indicates that there is a conflict between the librealsense SDK and the Linux kernel,  Which kernel version are you using, please?

danielhertz, 2024-02-26:
@MartyG-RealSense hello, ty so much for ur help, im using karnel version 6.5.0-21-generic

MartyG-RealSense, 2024-02-26:
Kernel 6.5 support has been added to the **development** branch of librealsense - please see https://github.com/IntelRealSense/librealsense/commit/9f4f0b4140b335966656df194004afba1b423c75

The development branch becomes the next 'master' latest release of the SDK when published.  The next SDK master release is planned for the Q1 2024 window (January to March).  

https://github.com/IntelRealSense/librealsense/tree/development

In the meantime, in situations where librealsense is conflicting with the kernel, the SDK can bypass the kernel if the SDK is built from source code with CMake with the build flag **-DFORCE_RSUSB_BACKEND=TRUE** included in the CMake build instruction.  This bypass is useful when a recent kernel is not yet supported by librealsense or when the computer being used is an 'exotic' configuration such as an industrial computing board.

danielhertz, 2024-03-05:
@MartyG-RealSense ty so much for the comment and support.
ok, i downgraded my ubuntu version from 22.04 to 20.04 and my kernel now is 5.15 and its still the same error, i can see the poinclouds, rgb and the rgb(also in cheese), i tried it in the windows partition and its not having the same one. 
im using laptop msi i7-11.
im banging my head for some time and wish i could know whats the problem

MartyG-RealSense, 2024-03-05:
The Windows version of librealsense runs on a different backend to the Linux version.  The Windows SDK's backend is Microsoft Media Foundation and the default Linux backend of the SDK is V4L2.

Which method did you use to install librealsense on Linux, please?

danielhertz, 2024-03-06:
from the https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md with the master

danielhertz, 2024-03-09:
well i dont have the error but other accured by this set
r those cruicial for the ros wrapper ? 

14:25:19 [Info]  - librealsense version: 2.54.2

14:25:19 [Debug] /home/daniel/dev/librealsense/src/rsusb-backend/rsusb-backend.cpp:32 - Creating UVC Device from path: 4-8-5
14:25:19 [Info] /home/daniel/dev/librealsense/src/rs.cpp:2697 - Framebuffer size changed to 1920 x 948
14:25:19 [Info] /home/daniel/dev/librealsense/src/rs.cpp:2697 - Window size changed to 1920 x 948
14:25:19 [Info] /home/daniel/dev/librealsense/src/rs.cpp:2697 - Scale Factor is now 1
14:25:19 [Debug] /home/daniel/dev/librealsense/src/rsusb-backend/rsusb-backend.cpp:32 - Creating UVC Device from path: 4-8-5
14:25:19 [Debug] /home/daniel/dev/librealsense/src/ds/d400/d400-factory.cpp:1185 - d400_try_fetch_usb_device(...) failed.
14:25:19 [Info] /home/daniel/dev/librealsense/src/context.cpp:336 - Found 1 RealSense devices (mask 0xfe)
14:25:19 [Debug] /home/daniel/dev/librealsense/src/ds/ds-private.h:295 - Loaded Valid Table: version [mjr.mnr]: 0x0300, type 25, size 496, CRC: 91b71f34
14:25:21 [Debug] /home/daniel/dev/librealsense/src/hw-monitor.cpp:196 - hwmon command 0x7d( 0 0 0 0 ) failed (response -21= No data to return)
14:25:21 [Debug] /home/daniel/dev/librealsense/src/types.cpp:74 - recoverable_exception: hwmon command 0x7d( 0 0 0 0 ) failed (response -21= No data to return)
14:25:21 [Debug] /home/daniel/dev/librealsense/src/hw-monitor.cpp:196 - hwmon command 0x7d( 0 0 0 0 ) failed (response -21= No data to return)
14:25:21 [Debug] /home/daniel/dev/librealsense/src/types.cpp:74 - recoverable_exception: hwmon command 0x7d( 0 0 0 0 ) failed (response -21= No data to return)
14:25:21 [Debug] /home/daniel/dev/librealsense/src/ds/ds-private.h:295 - Loaded Valid Table: version [mjr.mnr]: 0x0202, type 32, size 240, CRC: e247a118
14:25:21 [Debug] /home/daniel/dev/librealsense/src/ds/d400/d400-private.cpp:260 - 
 [0] = 242.649	 [1] = 616.414	 [2] = 616.832	 [3] = 2.8026e-45	

14:25:21 [Info]  - Intel RealSense D435 was selected as a default device
14:25:21 [Debug] /home/daniel/dev/librealsense/src/rsusb-backend/rsusb-backend.cpp:32 - Creating UVC Device from path: 3-10-4
14:25:21 [Error] /home/daniel/dev/librealsense/src/libusb/handle-libusb.h:53 - failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
14:25:21 [Error] /home/daniel/dev/librealsense/src/sensor.cpp:661 - acquire_power failed: failed to set power state
14:25:21 [Warn] /home/daniel/dev/librealsense/src/sensor.cpp:1291 - Failed to add Backlight Compensation control for RGB Camera
14:25:21 [Error] /home/daniel/dev/librealsense/src/libusb/handle-libusb.h:53 - failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
14:25:21 [Error] /home/daniel/dev/librealsense/src/sensor.cpp:661 - acquire_power failed: failed to set power state
14:25:21 [Warn] /home/daniel/dev/librealsense/src/sensor.cpp:1291 - Failed to add Brightness control for RGB Camera
14:25:21 [Error] /home/daniel/dev/librealsense/src/libusb/handle-libusb.h:53 - failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
14:25:21 [Error] /home/daniel/dev/librealsense/src/sensor.cpp:661 - acquire_power failed: failed to set power state
14:25:21 [Warn] /home/daniel/dev/librealsense/src/sensor.cpp:1291 - Failed to add Contrast control for RGB Camera
14:25:21 [Error] /home/daniel/dev/librealsense/src/libusb/handle-libusb.h:53 - failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
14:25:21 [Error] /home/daniel/dev/librealsense/src/sensor.cpp:661 - acquire_power failed: failed to set power state
14:25:21 [Warn] /home/daniel/dev/librealsense/src/sensor.cpp:1291 - Failed to add Exposure control for RGB Camera
14:25:21 [Error] /home/daniel/dev/librealsense/src/libusb/handle-libusb.h:53 - failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
14:25:21 [Error] /home/daniel/dev/librealsense/src/sensor.cpp:661 - acquire_power failed: failed to set power state
14:25:21 [Warn] /home/daniel/dev/librealsense/src/sensor.cpp:1291 - Failed to add Gamma control for RGB Camera
14:25:21 [Error] /home/daniel/dev/librealsense/src/libusb/handle-libusb.h:53 - failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
14:25:21 [Error] /home/daniel/dev/librealsense/src/sensor.cpp:661 - acquire_power failed: failed to set power state
14:25:21 [Warn] /home/daniel/dev/librealsense/src/sensor.cpp:1291 - Failed to add Hue control for RGB Camera
14:25:21 [Error] /home/daniel/dev/librealsense/src/libusb/handle-libusb.h:53 - failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
14:25:21 [Error] /home/daniel/dev/librealsense/src/sensor.cpp:661 - acquire_power failed: failed to set power state
14:25:21 [Warn] /home/daniel/dev/librealsense/src/sensor.cpp:1291 - Failed to add Saturation control for RGB Camera
14:25:21 [Error] /home/daniel/dev/librealsense/src/libusb/handle-libusb.h:53 - failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
14:25:21 [Error] /home/daniel/dev/librealsense/src/sensor.cpp:661 - acquire_power failed: failed to set power state
14:25:21 [Warn] /home/daniel/dev/librealsense/src/sensor.cpp:1291 - Failed to add Sharpness control for RGB Camera
14:25:21 [Error] /home/daniel/dev/librealsense/src/libusb/handle-libusb.h:53 - failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
14:25:21 [Error] /home/daniel/dev/librealsense/src/sensor.cpp:661 - acquire_power failed: failed to set power state
[... cut ...]
14:25:23 [Debug] /home/daniel/dev/librealsense/src/hw-monitor.cpp:196 - hwmon command 0x7d( 0 0 0 0 ) failed (response -21= No data to return)
14:25:23 [Debug] /home/daniel/dev/librealsense/src/types.cpp:74 - recoverable_exception: hwmon command 0x7d( 0 0 0 0 ) failed (response -21= No data to return)
14:25:23 [Debug] /home/daniel/dev/librealsense/src/hw-monitor.cpp:196 - hwmon command 0x7d( 0 0 0 0 ) failed (response -21= No data to return)
14:25:23 [Debug] /home/daniel/dev/librealsense/src/types.cpp:74 - recoverable_exception: hwmon command 0x7d( 0 0 0 0 ) failed (response -21= No data to return)

danielhertz, 2024-03-09:
@MartyG-RealSense  sorry for dudnt tagged u for the comment above

MartyG-RealSense, 2024-03-09:
It's no problem.  You do not need to tag me if I have already commented on an issue as GitHub will automatically send new messages to me.  :)

The messages look quite serious.  The **Failed to set power state** errors indicate that the camera cannot be accessed.

Are you using the ROS1 **2.3.2** wrapper version or the ROS2 **4.54.1** wrapper, please?  If the ROS1 2.3.2 wrapper is being used then librealsense 2.50.0 or 2.51.1 and firmware driver **5.13.0.50** must be used.

Which ROS launch instruction are you using?

danielhertz, 2024-03-09:
well, i didnt get yet into the realsense ros wrapper coz i know its bassed on the sdk (that was my mistake before with the xioctl error or am i wrong? ). for now im so despparate to make the sdk works propparly even i can c in the viewer the rgb and the pointclouds. so we solved one issue and now theres another.
maybe its an hardware problem of msi laptops, what u think kind sir ? :)
btw, sorry about my lame english

MartyG-RealSense, 2024-03-09:
Your English is good, don't worry.

Are you accessing the camera with more than one program at the same time?  If you had depth and RGB enabled in the RealSense Viewer and then ran another RealSense program, the second program would not be able to use the depth and RGB streams.  This is because the first program that is run locks access to the streams so that a second program cannot use the streams until the first program stops using them.  For example:

Viewer uses depth and RGB first.  A program script run secondly afterwards cannot then use depth and RGB. Or

A program script uses depth and RGB first. If the Viewer is run secondly afterwards then it cannot use depth and RGB.

danielhertz, 2024-03-10:
this is while running the the viewer:

daniel@daniel-Pulse-GL66-11UGKV:~/dev/librealsense$ ps aux | grep realsense
daniel      4914  7.9  0.6 352548 102604 pts/0   Sl+  10:19   0:00 realsense-viewer
daniel      4932  0.0  0.0   9040   712 pts/1    S+   10:19   0:00 grep --color=auto realsense
daniel@daniel-Pulse-GL66-11UGKV:~/dev/librealsense$ 

it seem that there r no conflicts that running while the viewer works

danielhertz, 2024-03-10:
daniel@daniel-Pulse-GL66-11UGKV:~/dev/librealsense$ ps aux | grep realsense
daniel      4914  7.9  0.6 352548 102604 pts/0   Sl+  10:19   0:00 realsense-viewer
daniel      4932  0.0  0.0   9040   712 pts/1    S+   10:19   0:00 grep --color=auto realsense
daniel@daniel-Pulse-GL66-11UGKV:~/dev/librealsense$ 

this is while the viewer works, it seems that there r no conflicts that running while the viewer is in the bkround

danielhertz, 2024-03-10:
daniel@daniel-Pulse-GL66-11UGKV:/dev/librealsense$    ps aux | grep realsense
daniel 4914 7.9 0.6 352548 102604 pts/0 Sl+ 10:19 0:00 realsense-viewer
daniel 4932 0.0 0.0 9040 712 pts/1 S+ 10:19 0:00 grep --color=auto realsense

this is while the viewer works, it seems that there r no conflicts that running while the viewer is in the bkround

MartyG-RealSense, 2024-03-10:
What are you running in the foreground that causes the errors, please?

danielhertz, 2024-03-10:
nothing, just the browser and the terminal on an new ubuntu

MartyG-RealSense, 2024-03-10:
It is difficult to see where the problem is occurring that is causing these errors.  You built the SDK from source code with **-DFORCE_RSUSB_BACKEND=TRUE**.  This build method bypasses the kernel so there should not be kernel conflicts, and RSUSB = true does not care which Linux version is being used and so using Ubuntu 22.04 or 20.04 should not be a problem.

If realsense-viewer is working correctly and the errors are in the terminal, is it possible for you to just ignore the terminal errors?

danielhertz, 2024-03-10:
ok i moved forward to the ros wrapper(i upgraded again to ubuntu desktop 22.04, and DFORCE_RSUSB_BACKEND=TRUE to bypass the kernel) and its stuck in an warning when i simple launch:

daniel@daniel-Pulse-GL66-11UGKV:~/ros2_humble$  ros2 launch realsense2_camera rs_launch.py
[INFO] [launch]: All log files can be found below /home/daniel/.ros/log/2024-03-10-23-20-42-143538-daniel-Pulse-GL66-11UGKV-299212
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [299213]
[realsense2_camera_node-1] [INFO] [camera.camera]: RealSense ROS v4.54.1 (init() at ./src/realsense_node_factory.cpp:247)
[realsense2_camera_node-1] [INFO] [camera.camera]: Built with LibRealSense v2.54.1 (init() at ./src/realsense_node_factory.cpp:248)
[realsense2_camera_node-1] [INFO] [camera.camera]: Running with LibRealSense v2.54.1 (init() at ./src/realsense_node_factory.cpp:249)
[realsense2_camera_node-1] [INFO] [camera.camera]: Device with serial number 831612074153 was found.
[realsense2_camera_node-1]  (getDevice() at ./src/realsense_node_factory.cpp:103)
[realsense2_camera_node-1] [INFO] [camera.camera]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb4/4-4/4-4:1.0/video4linux/video2 was found. (getDevice() at ./src/realsense_node_factory.cpp:106)
[realsense2_camera_node-1] [INFO] [camera.camera]: Device with name Intel RealSense D435 was found. (getDevice() at ./src/realsense_node_factory.cpp:108)
[realsense2_camera_node-1] [INFO] [camera.camera]: Device with port number 4-4 was found. (getDevice() at ./src/realsense_node_factory.cpp:126)
[realsense2_camera_node-1] [INFO] [camera.camera]: Device USB type: 3.2 (getDevice() at ./src/realsense_node_factory.cpp:178)
[realsense2_camera_node-1] [INFO] [camera.camera]: getParameters... (getParameters() at ./src/parameters.cpp:23)
[realsense2_camera_node-1] [INFO] [camera.camera]: JSON file is not provided (setAvailableSensors() at ./src/rs_node_setup.cpp:92)
[realsense2_camera_node-1] [INFO] [camera.camera]: Device Name: Intel RealSense D435 (setAvailableSensors() at ./src/rs_node_setup.cpp:95)
[realsense2_camera_node-1] [INFO] [camera.camera]: Device Serial No: 831612074153 (setAvailableSensors() at ./src/rs_node_setup.cpp:98)
[realsense2_camera_node-1] [INFO] [camera.camera]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb4/4-4/4-4:1.0/video4linux/video2 (setAvailableSensors() at ./src/rs_node_setup.cpp:102)
[realsense2_camera_node-1] [INFO] [camera.camera]: Device FW version: 5.15.1 (setAvailableSensors() at ./src/rs_node_setup.cpp:105)
[realsense2_camera_node-1] [INFO] [camera.camera]: Device Product ID: 0x0B07 (setAvailableSensors() at ./src/rs_node_setup.cpp:108)
[realsense2_camera_node-1] [INFO] [camera.camera]: Sync Mode: Off (setAvailableSensors() at ./src/rs_node_setup.cpp:110)
[realsense2_camera_node-1] [INFO] [camera.camera]: Set ROS param depth_module.profile to default: 848x480x30 (operator()() at ./src/profile_manager.cpp:323)
[realsense2_camera_node-1] [INFO] [camera.camera]: Set ROS param rgb_camera.profile to default: 640x480x30 (operator()() at ./src/profile_manager.cpp:323)
[realsense2_camera_node-1] [INFO] [camera.camera]: Stopping Sensor: Depth Module (updateSensors() at ./src/rs_node_setup.cpp:329)
[realsense2_camera_node-1] [INFO] [camera.camera]: Starting Sensor: Depth Module (updateSensors() at ./src/rs_node_setup.cpp:351)
[realsense2_camera_node-1] [INFO] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 848, Height: 480, FPS: 30 (start() at ./src/ros_sensor.cpp:230)
[realsense2_camera_node-1] [INFO] [camera.camera]: Stopping Sensor: RGB Camera (updateSensors() at ./src/rs_node_setup.cpp:329)
[realsense2_camera_node-1] [INFO] [camera.camera]: Starting Sensor: RGB Camera (updateSensors() at ./src/rs_node_setup.cpp:351)
[realsense2_camera_node-1] [INFO] [camera.camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 640, Height: 480, FPS: 30 (start() at ./src/ros_sensor.cpp:230)
[realsense2_camera_node-1] [INFO] [camera.camera]: RealSense Node Is Up! (publishTopics() at ./src/base_realsense_node.cpp:171)
[realsense2_camera_node-1] **[WARN]** [camera.camera]:  (setBaseTime() at ./src/base_realsense_node.cpp:609)

^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[realsense2_camera_node-1] [INFO] [rclcpp]: signal_handler(signum=2) (signal_handler() at ./src/rclcpp/signal_handler.cpp:71)
[realsense2_camera_node-1] [INFO] [camera.camera]: Stop Sensor: Depth Module (stop() at ./src/ros_sensor.cpp:247)
[realsense2_camera_node-1] [INFO] [camera.camera]: Close Sensor.  (stop() at ./src/ros_sensor.cpp:258)
[realsense2_camera_node-1] [INFO] [camera.camera]: Close Sensor - Done.  (stop() at ./src/ros_sensor.cpp:267)
[realsense2_camera_node-1] [INFO] [camera.camera]: Stop Sensor: RGB Camera (stop() at ./src/ros_sensor.cpp:247)
[realsense2_camera_node-1] [INFO] [camera.camera]: Close Sensor.  (stop() at ./src/ros_sensor.cpp:258)
[realsense2_camera_node-1] [INFO] [camera.camera]: Close Sensor - Done.  (stop() at ./src/ros_sensor.cpp:267)
[INFO] [realsense2_camera_node-1]: process has finished cleanly [pid 299213]
daniel@daniel-Pulse-GL66-11UGKV:~/ros2_humble$

MartyG-RealSense, 2024-03-11:
If you are able to see streams even when warnings and errors are displayed, it possible for you to just ignore the terminal errors?

MartyG-RealSense, 2024-03-21:
Hi @danielhertz  Do you require further assistance with this case, please?  Thanks!

danielhertz, 2024-03-26:
well its still the same problem, the wrapper wont let me launch the camera and its not working with rviz2. i installed again the ubuntu with minimal that maybe the conflict with other programs will be solved and ill install again the sdk, lets c if it will work properlly

danielhertz, 2024-03-26:
the same :(

12:54:16 [Info]  - librealsense version: 2.54.2

12:54:16 [Error] /home/daniel/librealsense/src/libusb/handle-libusb.h:53 - failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
12:54:16 [Error] /home/daniel/librealsense/src/sensor.cpp:661 - acquire_power failed: failed to set power state
12:54:16 [Warn] /home/daniel/librealsense/src/sensor.cpp:1291 - Failed to add Backlight Compensation control for RGB Camera
12:54:16 [Error] /home/daniel/librealsense/src/libusb/handle-libusb.h:53 - failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
12:54:16 [Error] /home/daniel/librealsense/src/sensor.cpp:661 - acquire_power failed: failed to set power state
12:54:16 [Warn] /home/daniel/librealsense/src/sensor.cpp:1291 - Failed to add Brightness control for RGB Camera
12:54:16 [Error] /home/daniel/librealsense/src/libusb/handle-libusb.h:53 - failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
12:54:16 [Error] /home/daniel/librealsense/src/sensor.cpp:661 - acquire_power failed: failed to set power state
12:54:16 [Warn] /home/daniel/librealsense/src/sensor.cpp:1291 - Failed to add Contrast control for RGB Camera
12:54:16 [Error] /home/daniel/librealsense/src/libusb/handle-libusb.h:53 - failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
12:54:16 [Error] /home/daniel/librealsense/src/sensor.cpp:661 - acquire_power failed: failed to set power state
12:54:16 [Warn] /home/daniel/librealsense/src/sensor.cpp:1291 - Failed to add Exposure control for RGB Camera
12:54:16 [Error] /home/daniel/librealsense/src/libusb/handle-libusb.h:53 - failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
12:54:16 [Error] /home/daniel/librealsense/src/sensor.cpp:661 - acquire_power failed: failed to set power state
12:54:16 [Warn] /home/daniel/librealsense/src/sensor.cpp:1291 - Failed to add Gamma control for RGB Camera
12:54:16 [Error] /home/daniel/librealsense/src/libusb/handle-libusb.h:53 - failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
12:54:16 [Error] /home/daniel/librealsense/src/sensor.cpp:661 - acquire_power failed: failed to set power state
12:54:16 [Warn] /home/daniel/librealsense/src/sensor.cpp:1291 - Failed to add Hue control for RGB Camera
12:54:16 [Error] /home/daniel/librealsense/src/libusb/handle-libusb.h:53 - failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
12:54:16 [Error] /home/daniel/librealsense/src/sensor.cpp:661 - acquire_power failed: failed to set power state
12:54:16 [Warn] /home/daniel/librealsense/src/sensor.cpp:1291 - Failed to add Saturation control for RGB Camera
12:54:16 [Error] /home/daniel/librealsense/src/libusb/handle-libusb.h:53 - failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
12:54:16 [Error] /home/daniel/librealsense/src/sensor.cpp:661 - acquire_power failed: failed to set power state
12:54:16 [Warn] /home/daniel/librealsense/src/sensor.cpp:1291 - Failed to add Sharpness control for RGB Camera
12:54:16 [Error] /home/daniel/librealsense/src/libusb/handle-libusb.h:53 - failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
12:54:16 [Error] /home/daniel/librealsense/src/sensor.cpp:661 - acquire_power failed: failed to set power state
12:54:16 [Warn] /home/daniel/librealsense/src/sensor.cpp:1291 - Failed to add White Balance control for RGB Camera
12:54:16 [Error] /home/daniel/librealsense/src/libusb/handle-libusb.h:53 - failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
12:54:16 [Error] /home/daniel/librealsense/src/sensor.cpp:661 - acquire_power failed: failed to set power state
12:54:16 [Warn] /home/daniel/librealsense/src/sensor.cpp:1291 - Failed to add Enable Auto Exposure control for RGB Camera
12:54:16 [Error] /home/daniel/librealsense/src/libusb/handle-libusb.h:53 - failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
12:54:16 [Error] /home/daniel/librealsense/src/sensor.cpp:661 - acquire_power failed: failed to set power state
12:54:16 [Warn] /home/daniel/librealsense/src/sensor.cpp:1291 - Failed to add Enable Auto White Balance control for RGB Camera
12:54:16 [Info] /home/daniel/librealsense/src/rs.cpp:2697 - Framebuffer size changed to 1920 x 946
12:54:16 [Info] /home/daniel/librealsense/src/rs.cpp:2697 - Window size changed to 1920 x 946
12:54:16 [Info] /home/daniel/librealsense/src/rs.cpp:2697 - Scale Factor is now 1

MartyG-RealSense, 2024-03-26:
Please try unplugging the micro-sized end of the USB cable from the camera, reversing the connector's orientation and plugging it back into the camera (USB-C cables are two-way insertion at the micro-sized end).  Does doing so make any difference?

If that does not work, do you have another USB port to try the camera in?

Have you tried adding **initial_reset:=true** to your ros2 launch instruction to reset the camera at launch?

If that does not work, can you try launching with **ros2 run** instead of the launch file if you have not done so already, please?

`ros2 run realsense2_camera realsense2_camera_node`

danielhertz, 2024-03-27:
@MartyG-RealSense ok, i dived in and found out that in msi laptops boot, there is an option to get more options by pressing shift+ctrl+alt+f2 and the were opened lots of more options to change, the "usb support"(changed to initial) and "fast boot"(changed to disabled). the errors dissapeared like there never were but only when i open the stereo module an warning apears:
"11:04:46 [Warn] /home/daniel/librealsense/src/libusb/messenger-libusb.cpp:42 - control_transfer returned error, index: 768, error: Success, number: 0 " hope that warning isnt crusial and now ill go on for installing the wrapper again.
ill let u know if it worked.
btw, ty so so much

MartyG-RealSense, 2024-03-27:
You are very welcome, @danielhertz - thanks very much for the update with the detailed information about what worked for you!

It is possible for the ROS wrapper to work normally even if the **control_transfer returned** warning is generated in the terminal.  I look forward to your next report.  Good luck!

danielhertz, 2024-03-27:
well, im lucky, it didnt worked

daniel@daniel-Pulse-GL66-11UGKV:~$ ros2 run realsense2_camera realsense2_camera_node
[INFO] [1711564509.802620707] [camera.camera]: RealSense ROS v4.54.1
[INFO] [1711564509.802674605] [camera.camera]: Built with LibRealSense v2.54.2
[INFO] [1711564509.802682287] [camera.camera]: Running with LibRealSense v2.54.2
[INFO] [1711564509.940466560] [camera.camera]: Device with serial number 831612074153 was found.

[INFO] [1711564509.940523538] [camera.camera]: Device with physical ID 4-8-7 was found.
[INFO] [1711564509.940530362] [camera.camera]: Device with name Intel RealSense D435 was found.
[INFO] [1711564509.940652013] [camera.camera]: Device with port number 4-8 was found.
[INFO] [1711564509.940659388] [camera.camera]: Device USB type: 3.2
[INFO] [1711564509.941088074] [camera.camera]: getParameters...
[INFO] [1711564509.941242851] [camera.camera]: JSON file is not provided
[INFO] [1711564509.941252040] [camera.camera]: Device Name: Intel RealSense D435
[INFO] [1711564509.941257507] [camera.camera]: Device Serial No: 831612074153
[INFO] [1711564509.941263130] [camera.camera]: Device physical port: 4-8-7
[INFO] [1711564509.941268214] [camera.camera]: Device FW version: 5.15.1
[INFO] [1711564509.941273436] [camera.camera]: Device Product ID: 0x0B07
[INFO] [1711564509.941278042] [camera.camera]: Sync Mode: Off
[INFO] [1711564510.473288334] [camera.camera]: Stopping Sensor: Depth Module
[INFO] [1711564510.473615409] [camera.camera]: Stopping Sensor: RGB Camera
[INFO] [1711564510.500888120] [camera.camera]: Starting Sensor: Depth Module
 27/03 20:35:10,513 WARNING [136765071402560] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 27/03 20:35:10,524 WARNING [136765071402560] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
[INFO] [1711564510.534192428] [camera.camera]: Open profile: stream_type: Infra(1), Format: Y8, Width: 848, Height: 480, FPS: 30
[INFO] [1711564510.534247005] [camera.camera]: Open profile: stream_type: Infra(2), Format: Y8, Width: 848, Height: 480, FPS: 30
[INFO] [1711564510.534255064] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 848, Height: 480, FPS: 30
 27/03 20:35:10,535 WARNING [136765071402560] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
[INFO] [1711564510.545486309] [camera.camera]: Starting Sensor: RGB Camera
[INFO] [1711564510.553812424] [camera.camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 640, Height: 480, FPS: 30
[INFO] [1711564510.554793920] [camera.camera]: RealSense Node Is Up!
 27/03 20:35:10,734 WARNING [136765071402560] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 27/03 20:35:10,745 WARNING [136765071402560] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 27/03 20:35:10,755 WARNING [136765071402560] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 27/03 20:35:10,865 WARNING [136765071402560] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 27/03 20:35:10,875 WARNING [136765071402560] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
 27/03 20:35:10,886 WARNING [136765071402560] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0

MartyG-RealSense, 2024-03-27:
Have you checked whether the streams are being displayed in **RViz**, please?  Sometimes the streams will publish correctly in RViz even if `control_transfer returned` warnings are continuously generating in the log.

danielhertz, 2024-03-27:
[image]
[image]

MartyG-RealSense, 2024-03-28:
Are there still red errors in RViz if you change Fixed Frame from map to `camera_link`

closed 2024-04-10 by MartyG-RealSense