realsenseai/librealsense #12704: Questions about interference of realsense l515 in multiple systems.

Seunghui-Shin, 2024-02-26:
Hello! I'm studying about the interference of realsense l515 in multiple systems.

There are two realsense l515 cameras.

| Required Info                         |            f1182473 , f1182366                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   |  1.5.8.1| 
| Operating System & Version |   Ubuntu 20.04 | 
| Kernel Version (Linux Only)    |  5.15.0                                      | 
| Platform                                 | PC  |
| SDK Version                            |  2.54.2                           | 
| Language                            |  realsense-viewer                        | 
| Segment			|  others                       | 

When two cameras face each other, interference occurs as shown in the picture below.
The interference appears in the form of the RealSense l515 projecting IR.
However, as you can see in this picture, certain areas do not experience interference.

[image]

Also, if i turn off the two cameras and turn them on again, this area may shift or disappear.

[image]

1) Can I predict the area where interference does not occur?

I also have a question about the receiver.
When the emitter emits IR, how does the receiver process the reflected IR?
As shown in the picture below, when IR is emitted, a depth map is created within a certain area.

[image]

2) Then, does the receiver map the reflected IR to specific pixels?
3) Is the reason that interference occurs like an IR pattern because it is processed sequentially after receiving the IR?

Thank you!

MartyG-RealSense, 2024-02-26:
Hi @Seunghui-Shin  Intel's L515 multiple camera white-paper guide at the link below describes how interference between multiple L515s is expected when their fields of view overlap.  The paper describes how to synchronize the cameras to remove this interference.

https://dev.intelrealsense.com/docs/lidar-camera-l515-multi-camera-setup

In regard to your questions:

1.  Interference should not occur in areas where the field of view of the two cameras does not overlap, as shown by the image below extracted from the guide.

[image]

The PDF data sheet document for the L515 - linked to below - states on page 7: "The Intel RealSense LiDAR Camera L515 uses an IR laser, a MEMS, an IR photodiode, an RGB imager, a MEMS controller and a vison ASIC.  The MEMS is used
to scan the IR laser beam over the entire field-of-view (FOV)". 

"The L515 vision ASIC will process the data from the reflected beam captured by the photodiode and will output a depth point representing the accurate distance of a specific point in the scene from the camera.  Aggregation of the depth points will generate a point cloud depth data representing the full scene".

https://dev.intelrealsense.com/docs/lidar-camera-l515-datasheet

2.  This question "does the receiver map the reflected IR to specific pixels" is likely answered by the part of the above statement that states how the camera will "output a depth point representing the accurate distance of a specific point in the scene".

3.  Interference from IR between multiple cameras is likely to simply be because the L515 model is sensitive to IR light sources and is intended to be an indoor-only camera that is best used in locations with controlled lighting.

Seunghui-Shin, 2024-02-26:
The positions of the cameras in the two images I uploaded are the same, so the overlapping areas are the same.
But why do the areas without interference change?
This area changes every time i turn on the camera.

MartyG-RealSense, 2024-02-26:
Looking at how the cameras are shown as mostly black (no depth detail) but the arm behind them is okay, this may be because the cameras that are facing each other are positioned too closely together.  

RealSense cameras have a minimum depth sensing distance, and when the camera is closer to an object than the minimum distance then the depth detail of that object breaks up and then disappears from the image.  Because the arm behind the camera is further away, it should be above the minimum distance threshold and so be shown correctly on the image.  

If it is a problem related to minimum distance then you may be able to make the cameras appear on the image by going to **Stereo Module > Controls** in the RealSense Viewer and setting the **Min Distance** option to '0'.

[image]

Seunghui-Shin, 2024-02-27:
Thank you for your answer.
However, I am curious about why the areas where interference does not occur, as shown in the picture below, keep changing. When the cameras face each other in the same position, interference occurs in a wave pattern, but in the area marked with a red box, i.e., a certain area, interference does not occur.
Also, the location of this area changes every time i turn on the camera.
Can you tell me why?

[image]
[image]

MartyG-RealSense, 2024-02-27:
I have not seen this wave pattern before on an L515.  As both cameras are affected in the same way, it is unlikely to be a fault with the camera hardware.

Does changing the Visual Preset reduce the pattern at all?

Does the pattern occur if only one camera is turned on?

Does the pattern still occur if you reset the RealSense Viewer's settings using the instructions at the link below?

https://support.intelrealsense.com/hc/en-us/community/posts/360052013914/comments/1500000416522

Seunghui-Shin, 2024-02-28:
When only one camera is on, the pattern does not occur.
This pattern appears because interference occurs when the FoV of two or more cameras overlap.
I was curious as to why the area without interference keeps changing.
Thank you for your answer.

closed 2024-02-28 by Seunghui-Shin

MartyG-RealSense, 2024-02-28:
You are very welcome, @Seunghui-Shin