realsenseai/librealsense #12706: API Docs result in 404 error

kknives, 2024-02-26:
`librealsense` API was documented at https://intelrealsense.github.io/librealsense/doxygen. However, this site simply returns a 404 error. 
While this error is fixed, an alternative is to clone and build the docs using Doxygen locally or use the [ros mirror](https://docs.ros.org/en/humble/p/librealsense2).

MartyG-RealSense, 2024-02-26:
Thanks very much @kknives for the report.  My Intel RealSense colleagues are aware of this 404 issue.  

In addition to your own methods of access, you may find the menu-driven version of the API documentation at the link below to be an easy way to navigate the RealSense API.  It draws data directly from the official documentation pages and formats it into a user-friendly interface.

https://unanancyowen.github.io/librealsense2_apireference/classes.html

MartyG-RealSense, 2024-04-12:
Hi @jrecasens  I have responded at your other posting at https://github.com/IntelRealSense/librealsense/issues/12721#issuecomment-2051960544

MartyG-RealSense, 2024-05-30:
Hi everyone, the C++ API docs site has now been restored.  Thanks so much for your patience!

https://intelrealsense.github.io/librealsense/doxygen/

closed 2024-06-07 by kknives