realsenseai/librealsense #12710: Realsense SDK does not support Jetson platform with JetPack 6.0+

windyuan, 2024-02-26:
Realsense SDK does not support Jetson platform with JetPack 6.0+.
This is blocking Nvidia DeepStream 6.4+ to run Realsense camera on Jetson.

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400 } | 
| Firmware Version                   | 5.13.0.50 | 
| Operating System & Version |  Ubuntu 22.04  | 
| Kernel Version (Linux Only)    |  5.15.0-82-generic  aarch64    | 
| Platform                                 |  NVIDIA Jetson Orin  |
| SDK Version                            |  2.0+                        | 
| Language                            |  c++                          | 
| Segment			|  Mobile platforms                          | 

### Issue Description

NVIDIA DeepStream team have problem to run Realsense camera on Jetson Orin with Jetpack-6.0 release. The Ubuntu version is upgraded to 22.04 with this version.
https://developer.nvidia.com/embedded/jetpack-sdk-60dp

DeepStream SDK has a feature enabled Realsense camera(model: D435i) for previouse Jetpack version with Ubuntu 20.04/18.04. Now this feature is blocked since Realsense SDK doesn't support Jetson with Ubuntu 22.04.
https://docs.nvidia.com/metropolis/deepstream/dev-guide/text/DS_3D_Depth_Camera.html

We cannot find any support for Jetpack 6.0 with Ubuntu 22.04 from https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md.
Could you help us how to make realsense cameras running on Ubuntu 22.04 on Jetson arm64 platforms?
Thanks.

MartyG-RealSense, 2024-02-27:
Hi @windyuan  You should be able to use Jetson with librealsense without JetPack 6 support if the RealSense SDK is built from source code with CMake with the flag **-DFORCE_RSUSB_BACKEND=ON** included in the CMake build instruction.  When the SDK is built with RSUSB = on then it is not dependent on Linux versions or kernel versions and does not need to have a kernel patch script applied.

Another method for building the SDK from source with a kernel bypass is to use the **JetsonHacks** website's 'buildLibrealsense.sh' build script.

https://github.com/jetsonhacks/installRealSenseSDK?tab=readme-ov-file#buildlibrealsensesh

dmipx, 2024-03-04:
Hi.
You always can build LRS from sources on jetpack 6.0.
We are working on debian packages for ubuntu22/arm64.

windyuan, 2024-03-11:
Thank you for all your replies.
@MartyG-RealSense , we'll check the JetsonHacks.
 @dmipx, could you update this issue once ubuntu22/arm64 deb package is ready.
thanks.

MartyG-RealSense, 2024-03-27:
Hi @windyuan  Do you have an update about this case that you can provide, please?  Thanks!

dmipx, 2024-03-27:
Hi.
First of all - we did JP6 integration with our D457 mipi cameras.
Second - we found missing HID modules in JP6 kernel and you will need to recompile kernel with enabled modules so IMU accel/gyro willbe supported by LRS.
You can find all necessary information on how to get support for JP6 on [realsense_mipi_platform_driver](https://github.com/IntelRealSense/realsense_mipi_platform_driver/tree/dev)