realsenseai/librealsense #12711: Can't be able to access camera on docker with sdk

IamRam3, 2024-02-28:
I have installed realsense sdk on nvidia/cuda:12.3.1-runtime-ubuntu20.04 base image. Then I have ran the "realsense-viewer"
```
28/02 07:31:26,545 INFO [281473138989312] (backend-v4l2.cpp:657) Enumerating UVC video0 realpath=/sys/devices/platform/bus@0/3610000.usb/usb1/1-4/1-4.4/1-4.4:1.0/video4linux/video0
 28/02 07:31:26,546 INFO [281473138989312] (backend-v4l2.cpp:657) Enumerating UVC video1 realpath=/sys/devices/platform/bus@0/3610000.usb/usb1/1-4/1-4.4/1-4.4:1.0/video4linux/video1
 28/02 07:31:26,546 INFO [281473138989312] (backend-v4l2.cpp:657) Enumerating UVC video2 realpath=/sys/devices/platform/bus@0/3610000.usb/usb1/1-4/1-4.4/1-4.4:1.0/video4linux/video2
 28/02 07:31:26,546 INFO [281473138989312] (backend-v4l2.cpp:657) Enumerating UVC video3 realpath=/sys/devices/platform/bus@0/3610000.usb/usb1/1-4/1-4.4/1-4.4:1.0/video4linux/video3
 28/02 07:31:26,547 INFO [281473138989312] (backend-v4l2.cpp:657) Enumerating UVC video4 realpath=/sys/devices/platform/bus@0/3610000.usb/usb1/1-4/1-4.4/1-4.4:1.3/video4linux/video4
 28/02 07:31:26,547 INFO [281473138989312] (backend-v4l2.cpp:657) Enumerating UVC video5 realpath=/sys/devices/platform/bus@0/3610000.usb/usb1/1-4/1-4.4/1-4.4:1.3/video4linux/video5
 28/02 07:31:26,556 WARNING [281473138989312] (d400-factory.cpp:1195) DS5 group_devices is empty.
```

IamRam3, 2024-02-28:
And also given the appropriate permissions for the container like 
docker run --network host --privileged -v /dev:/dev -it [image_name]

MartyG-RealSense, 2024-02-28:
Hi @IamRam3  Does it make a difference if you launch Docker in sudo admin permissions mode with the commend **sudo docker run**

IamRam3, 2024-02-28:
No, Actually before launching the Docker I did run this command also "sudo usermod -aG docker $USER"

MartyG-RealSense, 2024-02-28:
And the camera is unable to be accessed if the Viewer is launched with sudo using **sudo realsense-viewer** ?

IamRam3, 2024-02-29:
Hi @MartyG-RealSense, the above method doesn't make any difference

MartyG-RealSense, 2024-02-29:
Does it make a difference if you add the commands below to your **docker run** instruction?

```
--device-cgroup-rule "c 81:* rmw" \
 --device-cgroup-rule "c 189:* rmw" \
```

Intel's librealsense Docker guide at the link below states that these commands "grant access for the docker to the USB and UVC resources of the host PC (needed to use RealSense devices)."

https://github.com/IntelRealSense/librealsense/tree/master/scripts/Docker

IamRam3, 2024-02-29:
No, it doesn't. The same error is keep surfacing.

IamRam3, 2024-02-29:
In local I can access camera, and lsusb output is :-
`Bus 002 Device 002: ID 0bda:0420 Realtek Semiconductor Corp. 4-Port USB 3.0 Hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 018: ID 30fa:0301  USB OPTICAL MOUSE 
Bus 001 Device 020: ID 8086:0b3a Intel Corp. Intel(R) RealSense(TM) Depth Camera 435i
Bus 001 Device 013: ID 046d:c31c Logitech, Inc. Keyboard K120
Bus 001 Device 003: ID 0bda:5420 Realtek Semiconductor Corp. 4-Port USB 2.0 Hub
Bus 001 Device 002: ID 13d3:3549 IMC Networks Bluetooth Radio
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`
In docker container the lsbusb output is :-
`Bus 002 Device 002: ID 0bda:0420 Realtek Semiconductor Corp. 4-Port USB 3.0 Hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 018: ID 30fa:0301 Generic 4-Port USB 2.0 Hub
Bus 001 Device 020: ID 8086:0b3a Intel Corp. 
Bus 001 Device 013: ID 046d:c31c Logitech, Inc. Keyboard K120
Bus 001 Device 003: ID 0bda:5420 Realtek Semiconductor Corp. 4-Port USB 2.0 Hub
Bus 001 Device 002: ID 13d3:3549 IMC Networks Bluetooth Radio
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
As we can see the Device 020(camera) is not getting recognized correctly. But I don't know why

MartyG-RealSense, 2024-02-29:
Some RealSense users who have not been able to detect the camera were able to do so after updating their camera's firmware driver.  Which librealsense SDK version and firmware driver version are you using, please?

dmipx, 2024-02-29:
> And also given the appropriate permissions for the container like docker run --network host --privileged -v /dev:/dev -it [image_name]

Hi. You need to export sysfs to docker.
`docker run --network host --privileged -v /dev:/dev -v /sys:/sys -it [image_name]`

IamRam3, 2024-03-01:
sry for the delay @MartyG-RealSense, here is my sdk and firmware driver version 
[image]
I have updated it. But doesn't work

IamRam3, 2024-03-01:
Hi @dmipx exporting sysfs doesn't working

closed 2024-03-16 by MartyG-RealSense