realsenseai/librealsense #12712: Hello! Where is the camera coordinate?

kevinDrawn, 2024-02-29:
Hello! @MartyG-RealSense 

I want to know where is the camera coordinate like below. 
[image]

[image]

Is it located in the lens?
If right, can you tell me what color it is?

kevinDrawn, 2024-02-29:
Thank you! 

Do you know exact dimensions of the depth camera origin and bolt hole?

[image]

MartyG-RealSense, 2024-02-29:
The D455's rear mounting holes are M4 size.  M4 holes have a diameter of 4 mm.

The distances between the D455's sensors are shown in the diagram below.  There is 37.5 mm between the centerlines of the left infrared sensor (depth origin) and the RGB sensor (RGB origin).

A CAD file model of D455 can be downloaded in a zipped folder by visiting the link below and clicking on the 'D400 Depth Cameras' link.

https://dev.intelrealsense.com/docs/stereo-depth-camera-d400#cad-files

kevinDrawn, 2024-02-29:
Thank you so much!

closed 2024-02-29 by kevinDrawn

MartyG-RealSense, 2024-02-29:
You are very welcome, @kevinDrawn - I'm pleased that I could help.  :)