realsenseai/librealsense #12717: D450 module not working with D4 vision board

ifnikran, 2024-03-01:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D450 | 
| Firmware Version                   | 2.54.2 | 
| Operating System & Version |   Linux (Ubuntu 20.04) | 
| Kernel Version (Linux Only)    |  (5.15.0-97-generic)                                         | 
| Platform                                 | Intel  |
| SDK Version                            |  {  }                          | 
| Language                            |  { }                          | 
| Segment			|  { }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
We recently purchased 5 units of D450 and D4 Vision Processor. The D450 connects with D3 Vision boards, but fails to be detected by D4 vision boards.

I have attached the pictures of the errors.

[image]
[image]

Please suggest how to get this to work. Or else if we need to just order a new set of boards. Thanks!

MartyG-RealSense, 2024-03-01:
Hi @ifnikran  The D450 depth camera module is only compatible with the **V3** version of the Vision Processor D4 Board.  The V4 version is for use with the D401 module (which is used inside the D405 camera model).

ifnikran, 2024-03-03:
> Hi @ifnikran The D450 depth camera module is only compatible with the **V3** version of the Vision Processor D4 Board. The V4 version is for use with the D401 module (which is used inside the D405 camera model).
> 
> I would recommend purchasing in multiples of 8 if possible with the RealSense Store's paired multipack at the link below. Purchasing the D450 and D4 V3 separately has a risk that an **Invalid SKU** error occurs when the two boards are connected together because of a mismatch between older and newer versions of D450 and D4 V3.
> 
> https://store.intelrealsense.com/buy-intel-realsense-depth-module-d450-and-d4-board-bundle-d.html

Thanks @MartyG-RealSense

closed 2024-03-03 by ifnikran