realsenseai/librealsense #12718: L515 imu problem !! help me !!!

qetuo105487900, 2024-03-01:
lun@lun:~$ rostopic list
/diagnostics
/rosout
/rosout_agg
/rs_l515/accel/imu_info
/rs_l515/align_to_color/parameter_descriptions
/rs_l515/align_to_color/parameter_updates
/rs_l515/aligned_depth_to_color/camera_info
/rs_l515/aligned_depth_to_color/image_raw
/rs_l515/aligned_depth_to_color/image_raw/compressed
/rs_l515/aligned_depth_to_color/image_raw/compressed/parameter_descriptions
/rs_l515/aligned_depth_to_color/image_raw/compressed/parameter_updates
/rs_l515/aligned_depth_to_color/image_raw/compressedDepth
/rs_l515/aligned_depth_to_color/image_raw/compressedDepth/parameter_descriptions
/rs_l515/aligned_depth_to_color/image_raw/compressedDepth/parameter_updates
/rs_l515/aligned_depth_to_color/image_raw/theora
/rs_l515/aligned_depth_to_color/image_raw/theora/parameter_descriptions
/rs_l515/aligned_depth_to_color/image_raw/theora/parameter_updates
/rs_l515/color/camera_info
/rs_l515/color/image_raw
/rs_l515/color/image_raw/compressed
/rs_l515/color/image_raw/compressed/parameter_descriptions
/rs_l515/color/image_raw/compressed/parameter_updates
/rs_l515/color/image_raw/compressedDepth
/rs_l515/color/image_raw/compressedDepth/parameter_descriptions
/rs_l515/color/image_raw/compressedDepth/parameter_updates
/rs_l515/color/image_raw/theora
/rs_l515/color/image_raw/theora/parameter_descriptions
/rs_l515/color/image_raw/theora/parameter_updates
/rs_l515/color/metadata
/rs_l515/depth/camera_info
/rs_l515/depth/image_rect_raw
/rs_l515/depth/image_rect_raw/compressed
/rs_l515/depth/image_rect_raw/compressed/parameter_descriptions
/rs_l515/depth/image_rect_raw/compressed/parameter_updates
/rs_l515/depth/image_rect_raw/compressedDepth
/rs_l515/depth/image_rect_raw/compressedDepth/parameter_descriptions
/rs_l515/depth/image_rect_raw/compressedDepth/parameter_updates
/rs_l515/depth/image_rect_raw/theora
/rs_l515/depth/image_rect_raw/theora/parameter_descriptions
/rs_l515/depth/image_rect_raw/theora/parameter_updates
/rs_l515/depth/metadata
/rs_l515/extrinsics/depth_to_color
/rs_l515/gyro/imu_info
/rs_l515/imu
/rs_l515/l500_depth_sensor/parameter_descriptions
/rs_l515/l500_depth_sensor/parameter_updates
/rs_l515/motion_module/parameter_descriptions
/rs_l515/motion_module/parameter_updates
/rs_l515/realsense2_camera_manager/bond
/rs_l515/rgb_camera/parameter_descriptions
/rs_l515/rgb_camera/parameter_updates
/tf
/tf_static

without /rs_l515/gyro/sample and /rs_l515/accel/sample what happened?

and as follow is my launch file 

<launch>
  <arg name="serial_no_camera1"    	default="f0190464"/> 	
  <arg name="camera1"              	default="rs_l515"/>		
  <arg name="tf_prefix_camera1"         default="$(arg camera1)"/>
  <arg name="reconnect_timeout"         default="6.0"/>
  <arg name="enable_confidence"   	default="false"/>
  <arg name="depth_fps"           	default="60"/>
  <arg name="gyro_fps"            	default="400"/>
  <arg name="accel_fps"           	default="200"/>
  <arg name="enable_gyro"         	default="true"/>
  <arg name="enable_accel"        	default="true"/>
  <arg name="align_depth"               default="true"/>
  <arg name="publish_tf"                default="true"/>
  <arg name="tf_publish_rate"           default="0"/>
  <arg name="initial_reset"             default="false"/>
  <arg name="unite_imu_method"          default="copy"/>

  <group ns="$(arg camera1)">
    <include file="$(find realsense2_camera)/launch/includes/nodelet.launch.xml">
      <arg name="serial_no"                value="$(arg serial_no_camera1)"/>
      <arg name="tf_prefix"         	   value="$(arg tf_prefix_camera1)"/>
      <arg name="reconnect_timeout"        value="$(arg reconnect_timeout)"/>
      <arg name="enable_confidence"        value="$(arg enable_confidence)"/>
      <arg name="align_depth"     	   value="$(arg align_depth)"/>
      <arg name="gyro_fps"     		   value="$(arg gyro_fps)"/>
      <arg name="accel_fps"     	   value="$(arg accel_fps)"/>
      <arg name="enable_gyro"              value="$(arg enable_gyro)"/>
      <arg name="enable_accel"             value="$(arg enable_accel)"/>
      <arg name="publish_tf"               value="$(arg publish_tf)"/>
      <arg name="tf_publish_rate"          value="$(arg tf_publish_rate)"/>
      <arg name="initial_reset"            value="$(arg initial_reset)"/>
      <arg name="unite_imu_method"         value="$(arg unite_imu_method)"/>
    </include>
  </group>

</launch>

qetuo105487900, 2024-03-01:
<launch>
  <arg name="serial_no_camera1"    	default="f0190464"/> 	
  <arg name="camera1"              	default="rs_l515"/>		
  <arg name="tf_prefix_camera1"         default="$(arg camera1)"/>
  <arg name="reconnect_timeout"         default="6.0"/>
  <arg name="enable_confidence"   	default="false"/>
  <arg name="depth_fps"           	default="60"/>
  <arg name="gyro_fps"            	default="400"/>
  <arg name="accel_fps"           	default="200"/>
  <arg name="enable_gyro"         	default="true"/>
  <arg name="enable_accel"        	default="true"/>
  <arg name="align_depth"               default="true"/>
  <arg name="publish_tf"                default="true"/>
  <arg name="tf_publish_rate"           default="0"/>
  <arg name="initial_reset"             default="false"/>
  <arg name="unite_imu_method"          default="copy"/>

  <group ns="$(arg camera1)">
    <include file="$(find realsense2_camera)/launch/includes/nodelet.launch.xml">
      <arg name="serial_no"                value="$(arg serial_no_camera1)"/>
      <arg name="tf_prefix"         	   value="$(arg tf_prefix_camera1)"/>
      <arg name="reconnect_timeout"        value="$(arg reconnect_timeout)"/>
      <arg name="enable_confidence"        value="$(arg enable_confidence)"/>
      <arg name="align_depth"     	   value="$(arg align_depth)"/>
      <arg name="gyro_fps"     		   value="$(arg gyro_fps)"/>
      <arg name="accel_fps"     	   value="$(arg accel_fps)"/>
      <arg name="enable_gyro"              value="$(arg enable_gyro)"/>
      <arg name="enable_accel"             value="$(arg enable_accel)"/>
      <arg name="publish_tf"               value="$(arg publish_tf)"/>
      <arg name="tf_publish_rate"          value="$(arg tf_publish_rate)"/>
      <arg name="initial_reset"            value="$(arg initial_reset)"/>
      <arg name="unite_imu_method"         value="$(arg unite_imu_method)"/>
    </include>
  </group>

</launch>

qetuo105487900, 2024-03-01:
but i use       `<arg name="unite_imu_method"         value=" "/>`
the topic list as follow :
```
lun@lun:~$ rostopic list
/diagnostics
/rosout
/rosout_agg
/rs_l515/accel/imu_info
/rs_l515/accel/metadata
/rs_l515/accel/sample
/rs_l515/align_to_color/parameter_descriptions
/rs_l515/align_to_color/parameter_updates
/rs_l515/aligned_depth_to_color/camera_info
/rs_l515/aligned_depth_to_color/image_raw
/rs_l515/aligned_depth_to_color/image_raw/compressed
/rs_l515/aligned_depth_to_color/image_raw/compressed/parameter_descriptions
/rs_l515/aligned_depth_to_color/image_raw/compressed/parameter_updates
/rs_l515/aligned_depth_to_color/image_raw/compressedDepth
/rs_l515/aligned_depth_to_color/image_raw/compressedDepth/parameter_descriptions
/rs_l515/aligned_depth_to_color/image_raw/compressedDepth/parameter_updates
/rs_l515/aligned_depth_to_color/image_raw/theora
/rs_l515/aligned_depth_to_color/image_raw/theora/parameter_descriptions
/rs_l515/aligned_depth_to_color/image_raw/theora/parameter_updates
/rs_l515/color/camera_info
/rs_l515/color/image_raw
/rs_l515/color/image_raw/compressed
/rs_l515/color/image_raw/compressed/parameter_descriptions
/rs_l515/color/image_raw/compressed/parameter_updates
/rs_l515/color/image_raw/compressedDepth
/rs_l515/color/image_raw/compressedDepth/parameter_descriptions
/rs_l515/color/image_raw/compressedDepth/parameter_updates
/rs_l515/color/image_raw/theora
/rs_l515/color/image_raw/theora/parameter_descriptions
/rs_l515/color/image_raw/theora/parameter_updates
/rs_l515/color/metadata
/rs_l515/depth/camera_info
/rs_l515/depth/image_rect_raw
/rs_l515/depth/image_rect_raw/compressed
/rs_l515/depth/image_rect_raw/compressed/parameter_descriptions
/rs_l515/depth/image_rect_raw/compressed/parameter_updates
/rs_l515/depth/image_rect_raw/compressedDepth
/rs_l515/depth/image_rect_raw/compressedDepth/parameter_descriptions
/rs_l515/depth/image_rect_raw/compressedDepth/parameter_updates
/rs_l515/depth/image_rect_raw/theora
/rs_l515/depth/image_rect_raw/theora/parameter_descriptions
/rs_l515/depth/image_rect_raw/theora/parameter_updates
/rs_l515/depth/metadata
/rs_l515/extrinsics/depth_to_color
/rs_l515/gyro/imu_info
/rs_l515/gyro/metadata
/rs_l515/gyro/sample
/rs_l515/l500_depth_sensor/parameter_descriptions
/rs_l515/l500_depth_sensor/parameter_updates
/rs_l515/motion_module/parameter_descriptions
/rs_l515/motion_module/parameter_updates
/rs_l515/realsense2_camera_manager/bond
/rs_l515/rgb_camera/parameter_descriptions
/rs_l515/rgb_camera/parameter_updates
/tf
/tf_static
```

MartyG-RealSense, 2024-03-01:
Hi @qetuo105487900  In the RealSense ROS wrapper the IMU topics (gyro and accel) are disabled by default.  They can be enabled by adding the following commands to the launch instruction.

`enable_gyro:=true enable_accel:=true`

Adding the unite_imu_method command will combine the gyro and accel topics together into a single /imu topic. 

**ROS1**
`unite_imu_method:=linear_interpolation`

**ROS2**
`unite_imu_method:=2`

closed 2025-01-06 by MartyG-RealSense