realsenseai/librealsense #12719: Improving precision of rs2_deproject_pixel_to_point

shrutichakraborty, 2024-03-02:
---------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D435i } | 
| Firmware Version                   | () | 
| Operating System & Version |   {Linux (Ubuntu 22.04)   | 
| Kernel Version (Linux Only)    |                          |                | 
| Language                            |  {pythony }                          | 
| Segment			|  {Robot }                          | 

### Issue Description
Hi, I am using 2D pixel to 3D conversion (similar to the code in this issue : (https://github.com/IntelRealSense/librealsense/issues/1413) and I would like to know what is the accuracy of the rs2_deproject_pixel_to_point function and how much error is expected? I have been trying to access the doculmentation of the api but the website seems to be unavailable. Can someone provide the mathematical backgroud behind this function? 

I am mainly only interested in converting pixel coordinates to the camera frame, and i need high submilimeter precision. For the moment, my code is similar to the issue in #1413 and when I test it the deprojected point has a about 20 mm offset in x and y (in meters) which is quite high. I am using aligned depth and colour frames.

Can anyone suggest methods/best practises to improve the accuracy/precision of the conversion or alternate more precise methods used to convert pixel to camera frame coordinates? 

Thanks!

MartyG-RealSense, 2024-03-02:
Hi @shrutichakraborty  The two main methods of converting 2D to 3D are **rs2_deproject_pixel_to_point** or **pc.calculate**.  When depth-color alignment is used with rs2_deproject_pixel_to_point there tends to be higher error because of inaccuracy introduced by the alignment process.  The pc.calculate method though can use an instruction called **map_to** to map the depth and color together.  The two methods are compared at https://github.com/IntelRealSense/librealsense/issues/4315

The only RealSense camera model that is officially rated for sub-millimeter detection accuracy though is the **D405**.

https://store.intelrealsense.com/buy-intel-realsense-depth-camera-d405.html

That model is not equipped with an RGB sensor.  Instead, the RGB image is provided by the depth sensor.  The D405 is designed to provide high accuracy, high quality images at close range (its ideal depth sensing range is between 7 cm and 50 cm distance from the canera).

Although the official API documentation is experiencing a 404 error at the time of writing this, you can access the official docs through an alternative link: https://unanancyowen.github.io/librealsense2_apireference/classes.html

shrutichakraborty, 2024-03-03:
> Hi @shrutichakraborty The two main methods of converting 2D to 3D are **rs2_deproject_pixel_to_point** or **pc.calculate**. When depth-color alignment is used with rs2_deproject_pixel_to_point there tends to be higher error because of inaccuracy introduced by the alignment process, so pc.calculate is slightly more accurate.
> 
> The pc.calculate method can use an instruction called **map_to** to map the depth and color together instead of using align_to.
> 
> The two methods are compared at #4315
> 
> The SDK's **opencv_pointcloud_viewer.py** Python pointcloud example demonstrates using map_to and pc.calculate.
> 
> https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/opencv_pointcloud_viewer.py#L305-L306
> 
> The only RealSense camera model that is officially rated for sub-millimeter detection accuracy is the **D405**.
> 
> https://store.intelrealsense.com/buy-intel-realsense-depth-camera-d405.html
> 
> That model is not equipped with an RGB sensor. Instead, the RGB image is provided by the depth sensor. The D405 is designed to provide high accuracy, high quality images at close range (its ideal depth sensing range is between 7 cm and 50 cm distance from the canera).
> 
> Although the official API documentation is experiencing a 404 error at the time of writing this, you can access the official docs through an alternative link: https://unanancyowen.github.io/librealsense2_apireference/classes.html

Thanks a lot for your response @MartyG-RealSense, I see I am experiencing the same issues as mentioned in the comments in the issues you linked, and my deprojected point is sightly shifted in x and y. At the current moment gettinga nother camera is not a solution for me, although I might consider it on a later stage. 
I still have a few questions : 
1. In my case I know the height above between the camera and the surface where I am detecting objects. Would using rs2_deproject_pixel_to_point with the known height between the surface and camera help improve the conversion? 
2. Do you know how the is the offset in the point computed from rs2_deproject_pixel_to_point without depth to colour alignment? 
3. Have there been any tests done on how much the offset is in the 3D point calculated from rs2_deproject_pixel_to_point? Is it a constant repeatable offset that could perhaps be integrated in the code?
4. Can you point me to some code examples to understand how to find the 3D point for a pixel (px,py)  using the pointcloud? Having an accurate X,Y coordinate is more important to me than depth accuracy.

Btw, In my case I have re-done the intrinsic camera calibration and I am using these intrinsics. 

Thanks a lot!

MartyG-RealSense, 2024-03-04:
I studied your questions carefully.  Thanks very much for your patience!

If the intrinsics are correct and XY still has an offset then I do not believe that there is much else that can be done in an rs2_deproject_pixel_to_point script to correct it.

 There isn't a consistent, predictable offset value.  It will vary from project to project.

A RealSense user at https://github.com/IntelRealSense/librealsense/issues/9749#issuecomment-919922308 who also had an offset problem hard-coded a correction mechanism into their script.

For your particular project, it may be better to use the instruction **rs2_project_color_pixel_to_depth_pixel**.  This converts a 2D color pixel to a 3D depth pixel.  It converts a single pixel coordinate and so does not require use of depth-color alignment.  An example Python script that demonstrates use of this instruction can be found at https://github.com/IntelRealSense/librealsense/issues/5603#issuecomment-574019008

shrutic12, 2024-03-04:
> I studied your questions carefully. Thanks very much for your patience!
> 
> If the intrinsics are correct and XY still has an offset then I do not believe that there is much else that can be done in an rs2_deproject_pixel_to_point script to correct it.
> 
> There isn't a consistent, predictable offset value. It will vary from project to project.
> 
> A RealSense user at [#9749 (comment)](https://github.com/IntelRealSense/librealsense/issues/9749#issuecomment-919922308) who also had an offset problem hard-coded a correction mechanism into their script.
> 
> For your particular project, it may be better to use the instruction **rs2_project_color_pixel_to_depth_pixel**. This converts a 2D color pixel to a 3D depth pixel. It converts a single pixel coordinate and so does not require use of depth-color alignment. An example Python script that demonstrates use of this instruction can be found at [#5603 (comment)](https://github.com/IntelRealSense/librealsense/issues/5603#issuecomment-574019008)

Thanks! I am trying the rs2_project_color_pixel_to_depth_pixel function. I was wondering about the depth_scale parameter, how can I get it? in the code they use `depth_scale = profile.get_device().first_depth_sensor().get_depth_scale()` but in my case I have `cnf = pipeline.get_start(config)` and `profile = cfg.get_stream(rs.stream.color)` so I am not sure I can find the depth scale from that..

shrutichakraborty, 2024-03-04:
Hello again @MartyG-RealSense, could you also point me to any code examples on how to map a pixel to a point in the pointcloud? Thanks!

shrutichakraborty, 2024-03-04:
> > I studied your questions carefully. Thanks very much for your patience!
> > If the intrinsics are correct and XY still has an offset then I do not believe that there is much else that can be done in an rs2_deproject_pixel_to_point script to correct it.
> > There isn't a consistent, predictable offset value. It will vary from project to project.
> > A RealSense user at [#9749 (comment)](https://github.com/IntelRealSense/librealsense/issues/9749#issuecomment-919922308) who also had an offset problem hard-coded a correction mechanism into their script.
> > For your particular project, it may be better to use the instruction **rs2_project_color_pixel_to_depth_pixel**. This converts a 2D color pixel to a 3D depth pixel. It converts a single pixel coordinate and so does not require use of depth-color alignment. An example Python script that demonstrates use of this instruction can be found at [#5603 (comment)](https://github.com/IntelRealSense/librealsense/issues/5603#issuecomment-574019008)
> 
> Thanks! I am trying the rs2_project_color_pixel_to_depth_pixel function. I was wondering about the depth_scale parameter, how can I get it? in the code they use `depth_scale = profile.get_device().first_depth_sensor().get_depth_scale()` but in my case I have `cnf = pipeline.get_start(config)` and `profile = cfg.get_stream(rs.stream.color)` so I am not sure I can find the depth scale from that..
> 
> Also I am trying to understand what is the purpose of using the ``depth_min`and `depth_max` parameters. Is this something to set manually? How accurate does this need to be and what does it signify ?

Thanks I tried the rs2_project_color_pixel_to_depth_pixel, and I understand its giving us the pixel expressed in the depth frame? After that I need to reuse rs2_deproject function to get the point in the camera frame?

MartyG-RealSense, 2024-03-04:
For transforming a 2D pixel coordinate to a 3D world coordinate (the camera frame) with rs2_deproject_pixel_to_point and printing it, the script at https://github.com/IntelRealSense/librealsense/issues/1904 may be a helpful reference.

shrutichakraborty, 2024-03-04:
> For transforming a 2D pixel coordinate to a 3D world coordinate (the camera frame) with rs2_deproject_pixel_to_point and printing it, the script at #1904 may be a helpful reference.

sorry for the confusion, I am looking for any code examples on how to map a pixel to a point in the pointcloud? Thanks!

MartyG-RealSense, 2024-03-04:
Something like https://github.com/IntelRealSense/librealsense/issues/9340 ?

shrutichakraborty, 2024-03-04:
> Something like #9340 ?

Hi so I am using the formula `pcd[y * IMAGE_WIDTH + x]` from #9340 my question is here what is pcd? Is it `verts` from the code below?
 ```
points = pc.calculate(depth_frame)
        pc.map_to(mapped_frame)

        # Pointcloud data to arrays
        v, t = points.get_vertices(), points.get_texture_coordinates()
        verts = np.asanyarray(v).view(np.float32).reshape(-1, 3)  # xyz
        texcoords = np.asanyarray(t).view(np.float32).reshape(-1, 2)  # uv
```

MartyG-RealSense, 2024-03-04:
A PCD file is also known as **Point Cloud Data** and as the name suggests it is a file format for storing pointcloud data.

https://pointclouds.org/documentation/tutorials/pcd_file_format.html

In the case of the script at https://github.com/IntelRealSense/librealsense/issues/9340 though, it looks as though the program is importing a different kind of pointcloud data file called a **.ply** and then storing the contents in a variable called **loaded_pcd** (loaded point cloud data).

shrutichakraborty, 2024-03-04:
> A PCD file is also known as **Point Cloud Data** and as the name suggests it is a file format for storing pointcloud data.
>
> https://pointclouds.org/documentation/tutorials/pcd_file_format.html
>
> In [...]

I do not want to save and load the pointcloud. I want to get the 3D point in calera frame corresponding to a pixel in real-time from the camera. Currently this is the code I have : 

```
while go_on:
            frames_nb +=1
            # Wait for a coherent pair of frames: depth and color
            frames = pipeline.wait_for_frames()
            aligned_frames = align.process(frames)

            depth_frame = aligned_frames.get_depth_frame()
            color_frame = aligned_frames.get_color_frame()
            if not depth_frame or not color_frame:
                continue

            # Convert images to numpy arrays
            depth_image = np.asanyarray(depth_frame.get_data())
            color_image = np.asanyarray(color_frame.get_data())

            # Apply colormap on depth image (image must be converted to 8-bit per pixel first)
            depth_colormap = cv2.applyColorMap(cv2.convertScaleAbs(depth_image, alpha=0.03), cv2.COLORMAP_JET)

            depth_colormap_dim = depth_colormap.shape
            color_colormap_dim = color_image.shape
            w = color_frame.get_width()
            h = color_frame.get_height()
            points = pc.calculate(depth_frame)
            pc.map_to(color_frame)
            v,t = points.get_vertices(), points.get_texture_coordinates()
            verts = np.asanyarray(v).view(np.float32).reshape(-1,3)
            print("pointcloud shape is : ")
            print(verts.shape)          
            d.getDetection(color_image)
            centres = []
            if len(d.bboxes)>0:
                b = np.array(d.bboxes)
                for bounding_box,id,co_id in zip(b,d.labels,d.conf):
                        start_point = (int(bounding_box[0]), int(bounding_box[1]))
                        end_point = (int(bounding_box[2]), int(bounding_box[3]))
                        centre  = [(bounding_box[0]+bounding_box[2])/2, (bounding_box[1]+bounding_box[3])/2]
                        index = int(centre[1]*w+centre[0])
                        t = verts[index,:]
                        print(index)
                        print("coordinates in pointcloud =", t)
                        t = transform(list(t), transform_matrix )
                        print("coordinates in tcp frame =", t)
                        ### From TCP to Base Frame 
                        transform_m = getTransform()
                        t = transform(list(t), transform_m)
                        print("coordinates in base frame =", t)
```

I am using depth aligned to colour frame. However the results are very off so please let me know if the method is correct?

MartyG-RealSense, 2024-03-04:
**pc.map_to** is usually placed before the pc.calculate instruction, though in your script map_to is placed after.  Does it make a difference if you swap the lines around?
```
pc.map_to(color_frame)
points = pc.calculate(depth_frame)
```

shrutichakraborty, 2024-03-11:
Hello again, unfortunately I am sill getting approximately 6 to 9 cm error in position between real and computed from the methods. Is this normal? Is there any chance this could be due to a faulty camera... A strange thing I have noticed is that although my depth value seems correct, I have errors in the x and y coordinates. I was of the opinion that the error in computation of the x and y coordinates stems from incorrect depth.. any ideas what the issue could be that leads to more error in x and y values? 
Additionally, I have been looking at the intrinsic calibration parameters and they do not have the scaling parameters (scaling pixel to meters), is this normal or could this be an issue? 

Thanks a lot!

MartyG-RealSense, 2024-03-12:
The **pix** method that you use to calculate the depth in meters is not often used.  I would recommend using the **get_distance()** instructon instead if possible to retrieve and print a depth value in meters, like in the Python script at https://github.com/IntelRealSense/librealsense/issues/9939

A 6 to 9 cm error in XY would not be normal, unless there was a problem with the intrinsic values.  Do you have access to the **realsense-viewer** tool?  If you do, please try resetting the camera's calibration to its factory-new default settings in realsense-viewer with the instructions at https://github.com/IntelRealSense/librealsense/issues/10182#issuecomment-1019854487 if you have not done so already.

shrutichakraborty, 2024-03-12:
> **get_distance()**

Hi, can you link an issue with a complete reference to using `get_distance()` with colour aligned depth frames? I get an error `AttributeError: 'pyrealsense2.pyrealsense2.frame' object has no attribute 'get_distance`

MartyG-RealSense, 2024-03-12:
The complete script at the link below uses aligned depth-color and get_distance.

https://github.com/soarwing52/RealsensePython/blob/master/separate%20functions/measure_new.py

MartyG-RealSense, 2024-03-12:
As **measure_new.py** measures the distance between two points instead of from the camera to a point, the script at https://github.com/IntelRealSense/librealsense/issues/6749#issuecomment-654185205 may be more relevant to your project.

MartyG-RealSense, 2024-03-12:
1-3 cm error is what may be seen in XY coordinates that are towards the sides of the image, with accuracy being best at the center of the image.

Does XY accuracy improve if you set the depth resolution to **1280x720**, as accuracy increases as resolution is increased.

shrutichakraborty, 2024-03-12:
> 1-3 cm error is what may be seen in XY coordinates that are towards the sides of the image, with accuracy being best at the center of the image.
> 
> Does XY accuracy improve if you set the depth resolution to **1280x720**, as accuracy increases as resolution is increased.

Hi for some reason usinng `config.enable_stream(rs.stream.depth, 1280,720,rs.format.z16,30)` causes a RuntimeError at `pipeline.start(config)`, am I making a mistake?

MartyG-RealSense, 2024-03-12:
A RuntimeException may occur with that resolution if the camera was being detected as being on a USB 2.1 connection instead of USB 3.2, as 1280x720 depth is only supported at 6 FPS instead of 30 on USB 2.1.

closed 2024-03-27 by MartyG-RealSense