realsenseai/librealsense #12721: Missing API documentation

brunomcebola, 2024-03-03:
### Issue Description

Hi everyone, 

I'm currently developing my master thesis on computer vision and I'm using d435 cameras to capture my scene. 

I've been using the following links to access the documentation:
https://intelrealsense.github.io/librealsense/python_docs/_generated/pyrealsense2.config.html
https://intelrealsense.github.io/librealsense/doxygen/namespacers2.html

However, as of last month they have stopped working, and instead return a 404 not found github page. Does any one know where I can access the docs? Or by any chance have a local copy?

Best regards,
Bruno Cebola

MartyG-RealSense, 2024-03-03:
Hi @brunomcebola  Alternative working locations for these documentations can be found at the links below.

**C++ API docs**
https://unanancyowen.github.io/librealsense2_apireference/classes.html

**pyrealsense2 API docs**
https://web.archive.org/web/20230328071532/https://intelrealsense.github.io/librealsense/python_docs/_generated/pyrealsense2.html

brunomcebola, 2024-03-04:
Hi @MartyG-RealSense. 

Thanks for the links. I've already checked them both and all seems to be working fine. 

Best regards!

closed 2024-03-04 by brunomcebola

MartyG-RealSense, 2024-04-12:
Hi @jrecasens  The issue has been highlighted to the appropriate department of the Intel RealSense team.  I do not have any further information available at the time of writing this, unfortunately.  I sincerely apologize for the inconvenience.

There have not been many new RS2 instructions added to the SDK since the Wayback edition of the pyrealsense2 documentation was created in 2023, so you are not currently missing much detail by using the Wayback version.

MartyG-RealSense, 2024-05-25:
Hi @jrecasens  There is no information available at this time, unfortunately.  I have requested an update after receiving your inquiry.

MartyG-RealSense, 2024-05-30:
Hi @brunomcebola  The C++ API docs site has now been restored.  Work is continuing on restoration of the pyrealsense2 site.  Thanks so much for your patience!

MartyG-RealSense, 2024-07-18:
Hi everyone, the pyrealsense2 docs site is now restored.

https://intelrealsense.github.io/librealsense/python_docs/

brunomcebola, 2024-08-30:
Hi @MartyG-RealSense, Thanks for the heads-up!!!