realsenseai/librealsense #12727: Crashes in Mac M3 Pro Sonoma when connected to camera

philchao, 2024-03-05:
| | |
|---|---|
|**librealsense**|2.54.2 RELEASE|
|**OS**|Mac OS|

Please provide a description of the problem

I am able to start the Intel RealSense Viewer v2.54.2 without the camera connected. But once I connect the camera the application automatically shuts down:

Process:               realsense-viewer [6158]
Path:                  /opt/homebrew/*/realsense-viewer
Identifier:            realsense-viewer
Version:               ???
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2024-03-05 14:34:00.6287 +1300
OS Version:            macOS 14.2.1 (23C71)
Report Version:        12
Anonymous UUID:        9E225096-2F73-84E7-D358-A6214E7D6214

Time Awake Since Boot: 11000 seconds

System Integrity Protection: enabled

Crashed Thread:        1

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000028
Exception Codes:       0x0000000000000001, 0x0000000000000028

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [6158]

VM Region Info: 0x28 is not in any region.  Bytes before following region: 68719476696
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      commpage (reserved)        1000000000-7000000000   [384.0G] ---/--- SM=NUL  ...(unallocated)

Thread 0::  Dispatch queue: com.apple.main-thread

philchao, 2024-03-05:
05/03 14:53:54,186 ERROR [0x1709cb000] (handle-libusb.h:127) failed to claim usb interface: 0, error: RS2_USB_STATUS_ACCESS
 05/03 14:53:54,187 ERROR [0x1de44d000] (sensor.cpp:661) acquire_power failed: failed to set power state
 05/03 14:53:54,188 WARNING [0x1de44d000] (rs.cpp:312) null pointer passed for argument "device"
 05/03 14:53:54,188 WARNING [0x1de44d000] (rs.cpp:2700) Couldn't refresh devices - failed to set power state
 05/03 14:53:56,178 ERROR [0x16f3c7000] (context-libusb.cpp:16) libusb_init failed
zsh: segmentation fault  /opt/homebrew/Cellar/librealsense/2.54.2/bin/realsense-viewer

MartyG-RealSense, 2024-03-05:
Hi @philchao  Intel has not provided official support for use of RealSense on Mac computers since 2017's MacOS High Sierra.  Whilst the RealSense SDK has been able to work on Macs in subsequent years despite not being officially supported, problems have occurred more frequently since MacOS Monterey onwards and the introduction of the Apple Silicon Mac models (M1-M3).  The community of RealSense Mac users is therefore usually the provider of solutions for issues by sharing knowledge with each other.

The discussion at https://github.com/IntelRealSense/librealsense/issues/12307 is likely to be relevant to your particular issue with **Failed to set power state** errors, especially the final comment in that discussion at https://github.com/IntelRealSense/librealsense/issues/12307#issuecomment-1812688156

philchao, 2024-03-06:
@MartyG-RealSense Thank you for your prompt feedback. I am just wondering if is there some other way to change the output-input of the RealSense camera? For instance, I tried to connect this Apple USB-C Digital AV Multiport Adapter (https://www.apple.com/nz/shop/product/MUF82ZA/A/usb-c-digital-av-multiport-adapter) to the camera and provided charge for the camera through the adapter's USB-C charge input and output from the adapter's HDMI to the computer but it does not seem to register the device. Is it possible to output the RealSense camera to HDMI whilst providing it with power through a USB-C? Thanks

MartyG-RealSense, 2024-03-07:
I have never seen a case where RealSense has been made to work with HDMI, unfortunately.

There have been a couple of past cases where a camera was able to be detected when attached to an adaptor cable on a Mac, but in most cases it is not successful.

MartyG-RealSense, 2024-03-13:
Hi @philchao  Bearing in mind the advice in the comment above, do you require further assistance with this case please?  Thanks!

philchao, 2024-03-13:
Thanks @MartyG-RealSense. Unfortunately, I still was not able to get RealSense Viewer to work with my M3 Mac on Sonoma. Not sure if @cansik has been able to with his pyrealsense2

MartyG-RealSense, 2024-03-21:
Hi @cansik  Do you have any advice that you can offer @philchao regarding the above comment, please?  Thanks so much for your continued support for RealSense on Mac!

cansik, 2024-03-21:
@MartyG-RealSense @philchao Not really, it's a matter of trial and error. The second attempt to start the relasense-viewer (installed with [brew](https://formulae.brew.sh/formula/librealsense)) worked for me. The camera was already connected. However, only the depth camera produced an image, not the RGB camera.

[image]

In my opinion, RealSense is no longer usable in production for MacOS users (which is a great pity). Today I would recommend the cameras from [Luxonis](https://shop.luxonis.com/collections/oak-cameras-1), as they have an [API](https://docs.luxonis.com/projects/api/en/latest/) that is supported on all major platforms.

MartyG-RealSense, 2024-03-27:
Hi @philchao  Do you require further assistance with this case, please?  Thanks!

philchao, 2024-03-28:
Hi @MartyG-RealSense, yes, I would like the Intel Real Sense Viewer to work with Mac Sonoma, but it does not seem possible at this stage.

MartyG-RealSense, 2024-03-28:
One Mac user took the approach of using the camera with the **Asahi Linux** OS on their Apple Silicon machine instead of MacOS and wrote an installation guide.

https://github.com/HasanTheSyrian/librealsense-hasan/blob/master/doc/installation_asahi.md

MartyG-RealSense, 2024-04-03:
Hi @philchao  Is the Asahi Linux solution suggested in the comment above of use to you or do you still require a MacOS Sonoma solution, please?

Niranjini-Kathiravan, 2024-04-20:
Hi, I'm facing the same issue. My mac M2 (MacOS sonoma) is able to detect the camera but when I open the realsense-viewer its crashing and it's not able to open. Is there any solution I could get for this issue.

Jialus, 2024-04-25:
Hi, I'm getting the same issue on Mac M3.

I also tried to use Parallels to run realsense-viewer on ARM windows. In Device Manager, I could see the depth camera. But the viewer isn't able to detect it completely. Only 2D works, and it seems not actually found a RealSens Camera.

[image]

[image]

franckOL, 2024-07-04:
> @MartyG-RealSense @philchao Not really, it's a matter of trial and error. The second attempt to start the relasense-viewer (installed with [brew](https://formulae.brew.sh/formula/librealsense)) [...]

Can you explain how you are able to start realsense viewer ?
I install every whith brew for sonoma M3, but not working

michelgokan, 2024-08-22:
I have a very similar issue:

```
$ sudo realsense-viewer
GLFW Driver Error: Cocoa: Regular windows do not have icons on macOS
 22/08 18:16:01,479 INFO [0x2028e0f40] (context.cpp:116) ... 1-2-2
 22/08 18:16:01,479 INFO [0x2028e0f40] (context.cpp:129) Found 1 RealSense devices (0xff requested & 0xff from device-mask in settings)
[1]    42953 segmentation fault  sudo realsense-viewer
```

It's been reported here as well: https://support.intelrealsense.com/hc/en-us/community/posts/4548413451539-Activating-the-realsense-D435-Depth-Camera-in-MacOs

Any solution?

philchao, 2024-08-22:
I’ve found recently that mine seems to work with Parallels now on a windows virtual environment

michelgokan, 2024-08-22:
yes, that's exactly what I did but it's not very efficient :-(

MartyG-RealSense, 2024-09-03:
Hi everyone, 

The current latest 2.56.1 version of the librealsense SDK contains a fix for SDK compilation on MacOS Apple Silicon 'M' processors and also an update for MacOS 14 (Sonoma). 

These changes are confirmed in 2.56.1's release notes.

[image]

2.56.1 is currently an 'interim beta' release of the SDK which can only be installed by source code compilation. 

https://github.com/IntelRealSense/librealsense/releases/tag/v2.56.1

alifeinbinary, 2024-09-04:
@MartyG-RealSense 
I've just built from tag v2.56.1 on an Intel MacPro and encountered the same crash. 

My MacOS crash report is saying that the crash is occurring when libusb_get_device_list is called, same as with 2.55.1

I built with flags `cmake .. -DBUILD_EXAMPLES=true -DBUILD_WITH_OPENMP=false -DHWM_OVER_XU=false` as recommended in the docs. 

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               realsense-viewer [23430]
Path:                  /Users/USER/*/realsense-viewer
Identifier:            realsense-viewer
Version:               ???
Code Type:             X86-64 (Native)
Parent Process:        zsh [908]
Responsible:           iTerm2 [676]
User ID:               501

Date/Time:             2024-09-03 17:24:57.1403 -0700
OS Version:            macOS 14.6.1 (23G93)
Report Version:        12
Bridge OS Version:     8.6 (21P6074)
Anonymous UUID:        F2A3B944-89E3-69DF-85E9-E5875F22E685

Time Awake Since Boot: 49000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000028
Exception Codes:       0x0000000000000001, 0x0000000000000028

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [23430]

VM Region Info: 0x28 is not in any region.  Bytes before following region: 4475903960
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      10ac8e000-10af1a000    [ 2608K] r-x/r-x SM=COW  /Users/USER/*/realsense-viewer

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libsystem_pthread.dylib       	    0x7ff8032ca3e4 pthread_mutex_lock + 4
1   libusb-1.0.0.dylib            	       0x108cc6976 libusb_get_device_list + 261
2   librealsense2.2.56.1.dylib    	       0x10a27ad1e librealsense::platform::usb_context::usb_context() + 894
3   librealsense2.2.56.1.dylib    	       0x10a288dc4 librealsense::platform::usb_enumerator::query_devices_info() + 84
4   librealsense2.2.56.1.dylib    	       0x10a293d4e librealsense::platform::query_uvc_devices_info() + 46
5   librealsense2.2.56.1.dylib    	       0x10a2a910e librealsense::platform::rs_backend::query_uvc_devices() const + 14
6   librealsense2.2.56.1.dylib    	       0x10a2d5292 librealsense::backend_device_factory::query_devices(unsigned int) const + 162
7   librealsense2.2.56.1.dylib    	       0x10a2dbe74 librealsense::context::query_devices(int) const + 260
8   librealsense2.2.56.1.dylib    	       0x10a313569 rs2_query_devices_ex + 57
9   realsense-viewer              	       0x1086fee39 rs2::context::query_devices(int) const + 41
10  realsense-viewer              	       0x1086fe9fa rs2::ux_window::imgui_config_push() + 842
11  realsense-viewer              	       0x1086ff294 rs2::ux_window::operator bool() + 852
12  realsense-viewer              	       0x10883e97c main + 3148
13  dyld                          	    0x7ff802f43345 start + 1909
```

My terminal reports the following at the time of crash

```
GLFW Driver Error: Cocoa: Regular windows do not have icons on macOS
 03/09 17:24:56,750 INFO [0x7ff846812dc0] (context.cpp:119) ... 3-4-4
 03/09 17:24:56,750 INFO [0x7ff846812dc0] (context.cpp:132) Found 1 RealSense devices (0xff requested & 0xff from device-mask in settings)
[1]    23430 segmentation fault  ./realsense-viewer
```

When typing `lsusb` command into the terminal I get 
```Bus 003 Device 004: ID 8086:0ad3 Intel Corporation Intel(R) RealSense(TM) Depth Camera 415  Serial: 211123022034```

`libusb_get_device_list()` gets called in `src/libusb/context-libusb.cpp` but really it looks like all apps and examples are failing due to the calling of `ImGui_ImplGlfw_Init(app, false)`, the ImGui library intializition, which you can find in a number of files. The rest is a bit beyond me as I don't code in C++ very often. 

Hope that helps!

MartyG-RealSense, 2024-09-04:
Hi @alifeinbinary  Thanks very much for your feedback about LibUSB.  Further information regarding the Mac support updates in 2.56.1, including the LibUSB issue, can be found at https://github.com/IntelRealSense/librealsense/issues/9916#issuecomment-2326392774

I will pass your information to my Intel RealSense colleagues.  Thanks again!

closed 2026-10-25 by MartyG-RealSense