realsenseai/librealsense #12730: How to align rgb-d photos in L515

OneShz, 2024-03-05:
I want to have a colorful pointcloud like picture1 (which is collected by L515 in ros in pointcloud),but I got a 2 dimension point in picture2(align by using rgb photo and depth photo) .Here are my codes. 
[image] --figure 1
[image] --figure 2

import open3d as o3d
import numpy as np
from matplotlib import pyplot as plt

color_raw = o3d.io.read_image("K:/Image-Capture-With-RealSense-master/2024_01_09_16_16_55/Images/166.png")
depth_raw = o3d.io.read_image("K:/Image-Capture-With-RealSense-master/2024_01_09_16_16_55/DepthImages/166.png")
rgbd_image = o3d.geometry.RGBDImage.create_from_color_and_depth(color_raw, depth_raw, convert_rgb_to_intensity=False)
print(rgbd_image)

plt.subplot(1, 2, 1)
plt.title('Redwood grayscale image')
plt.imshow(rgbd_image.color)
plt.subplot(1, 2, 2)
plt.title('Redwood depth image')
plt.imshow(rgbd_image.depth)
plt.show()

inter = o3d.camera.PinholeCameraIntrinsic()
inter.set_intrinsics(640, 480, 599.6552734375, 599.66796875, 325.10821533203125, 235.8369598388672)
pcd = o3d.geometry.PointCloud().create_from_rgbd_image(
    rgbd_image, inter)

# Flip it, otherwise the pointcloud will be upside down
pcd.transform([[0.9999945163726807, -0.003304983489215374, -0.00013662033597938716, 0.0006620920030400157],
               [0.003301972057670355, 0.999830961227417, -0.018086116760969162, 0.01393149420619011],
               [0.00019637156219687313, 0.01808556728065014, 0.9998364448547363, -0.0035399049520492554],
               [0, 0, 0, 1]])
o3d.visualization.draw_geometries([pcd])

MartyG-RealSense, 2024-03-05:
Hi @OneShz  I note that we have previously discussed this subject at https://github.com/IntelRealSense/librealsense/issues/12625 and https://github.com/IntelRealSense/librealsense/issues/12629

May I first confirm with you that the color and depth PNG images that you are attempting to align are the same resolution size?

OneShz, 2024-03-06:
> Hi @OneShz I note that we have previously discussed this subject at #12625 and #12629
> 
> May I first confirm with you that the color and depth PNG images that you are attempting to align are the same resolution size?

Thank you for your help,yes, The size of both of them are 640*480. Here are the code abot collection rgb and depth image.

import argparse
import datetime
import json
import os

import cv2
import numpy as np
import pyrealsense2 as rs

def parse_opt():
    parser = argparse.ArgumentParser()
    parser.add_argument("--path", type=str, default='', help="images save path")
    parser.add_argument("--mode", type=int, default=0, help="0(auto) or 1(manual)")
    parser.add_argument("--image_format", type=int, default=0, help="option: 0->jpg 1->png")
    parser.add_argument("--image_width", type=int, default=1280, help="width of the image, recommended 1280 or 640")
    parser.add_argument("--image_height", type=int, default=720, help="height of the image, recommended 720 or 480")
    parser.add_argument("--fps", type=int, default=30, help="frame rate of shooting")
    opt = parser.parse_args()
    return opt

def get_aligned_images(dirname, aligned_frames, depth_scale):
    aligned_depth_frame = aligned_frames.get_depth_frame()
    color_frame = aligned_frames.get_color_frame()
    intr = color_frame.profile.as_video_stream_profile().intrinsics
    camera_parameters = {'fx': intr.fx, 'fy': intr.fy,
                         'ppx': intr.ppx, 'ppy': intr.ppy,
                         'height': intr.height, 'width': intr.width,
                         'depth_scale': profile.get_device().first_depth_sensor().get_depth_scale()
                         }
    with open(os.path.join(dirname, 'intrinsics.json'), 'w') as fp:
        json.dump(camera_parameters, fp)
    color_image = np.asanyarray(color_frame.get_data(), dtype=np.uint8)
    depth_image = np.asanyarray(aligned_depth_frame.get_data(), dtype=np.float32)
    mi_d = np.min(depth_image[depth_image > 0])
    ma_d = np.max(depth_image)
    depth = (255 * (depth_image - mi_d) / (ma_d - mi_d + 1e-8)).astype(np.uint8)
    depth_image_color = cv2.applyColorMap(depth, cv2.COLORMAP_JET)
    depth_image = np.asanyarray(aligned_depth_frame.get_data(), dtype=np.float32) * depth_scale * 1000
    return color_image, depth_image, depth_image_color

if __name__ == "__main__":
    opt = parse_opt()
    pipeline = rs.pipeline()
    config = rs.config()
    config.enable_stream(rs.stream.depth, opt.image_width, opt.image_height, rs.format.z16, opt.fps)
    config.enable_stream(rs.stream.color, opt.image_width, opt.image_height, rs.format.bgr8, opt.fps)
    profile = pipeline.start(config)
    depth_sensor = profile.get_device().first_depth_sensor()
    depth_scale = depth_sensor.get_depth_scale()
    align_to = rs.stream.color
    align = rs.align(align_to)
    now = datetime.datetime.now()
    if os.path.exists(os.path.join(opt.path, 'images')):
        dirname = opt.path
        if len(os.listdir(os.path.join(opt.path, 'images'))):
            li = sorted(os.listdir(os.path.join(opt.path, 'images')), key=lambda x: eval(x.split('.')[0]))
            n = eval(li[-1].split('.')[0])
        else:
            n = 0
    elif opt.path == '':
        n = 0
        dirname = os.path.join(opt.path, now.strftime("%Y_%m_%d_%H_%M_%S"))
    else:
        n = 0
        dirname = os.path.join(opt.path)
    color_dir = os.path.join(dirname, 'images')
    depth_dir = os.path.join(dirname, 'DepthImages')
    depth_color_dir = os.path.join(dirname, 'DepthColorImages')
    depth_npy_dir = os.path.join(dirname, 'DepthNpy')
    if not os.path.exists(dirname):
        os.mkdir(dirname)
        os.mkdir(color_dir)
        os.mkdir(depth_dir)
        os.mkdir(depth_color_dir)
        os.mkdir(depth_npy_dir)
    flag = 0
    image_formats = ['.jpg', '.png']
    while True:
        frames = pipeline.wait_for_frames()
        aligned_frames = align.process(frames)
        try:
            rgb, depth, depth_rgb = get_aligned_images(dirname, aligned_frames, depth_scale)
            cv2.imshow('RGB image', rgb)
            key = cv2.waitKey(1)
            if key == ord('q') or key == ord('Q'):
                pipeline.stop()
                break
            elif opt.mode:
                if key == ord('s') or key == ord('S'):
                    n = n + 1
                    cv2.imwrite(os.path.join(color_dir, str(n) + image_formats[opt.image_format]), rgb)
                    cv2.imwrite(os.path.join(depth_dir, str(n) + image_formats[opt.image_format]), depth)
                    cv2.imwrite(os.path.join(depth_color_dir, str(n) + image_formats[opt.image_format]), depth_rgb)
                    np.save(os.path.join(depth_npy_dir, str(n)), depth)
                    print('{}{} is saved!'.format(n, image_formats[opt.image_format]))
            else:
                if key == ord('s') or key == ord('S'):
                    flag = 1
                if key == ord('w') or key == ord('W'):
                    flag = 0
                if flag:
                    n = n + 1
                    cv2.imwrite(os.path.join(color_dir, str(n) + image_formats[opt.image_format]), rgb)
                    cv2.imwrite(os.path.join(depth_dir, str(n) + image_formats[opt.image_format]), depth)
                    cv2.imwrite(os.path.join(depth_color_dir, str(n) + image_formats[opt.image_format]), depth_rgb)
                    np.save(os.path.join(depth_npy_dir, str(n)), depth)
                    print('{}{} is saved!'.format(n, image_formats[opt.image_format]))
        except:
            pass

OneShz, 2024-03-06:
In this process I got the intrinsics.json like this 
{"fx": 599.6552734375, "fy": 599.66796875, "ppx": 325.10821533203125, "ppy": 235.8369598388672, "height": 480, "width": 640, "depth_scale": 0.0002500000118743628}

and the following are my RGB，depth image and depth color image.
[image]
[image]
[image]

OneShz, 2024-03-06:
> I researched your problem carefully but I cannot find any past RealSense / Open3D cases where a depth-color aligned image looks as good as your Figure 1 pointcloud from ROS. Your Figure 2 image may unfortunately be the best that can be achieved by using Open3D and aligning PNG images instead of using live camera data.

Thank you for your help,so what can I do to align rgb and depth just like in ros?Did you mean record a bag and extracted them just as the pointcloud but how can i change my launch file and how to align  them ?

OneShz, 2024-03-06:
> I researched your problem carefully but I cannot find any past RealSense / Open3D cases where a depth-color aligned image looks as good as your Figure 1 pointcloud from ROS. Your Figure 2 image may [...]
Do the pointcloud  extracted in ros also use the rgb and depth image to generate ?

MartyG-RealSense, 2024-03-06:
https://github.com/isl-org/Open3D/issues/473#issuecomment-508139946 has a Python / Open3D pointcloud script with depth-color alignment.

OneShz, 2024-03-07:
> [isl-org/Open3D#473 (comment)](https://github.com/isl-org/Open3D/issues/473#issuecomment-508139946) has a Python / Open3D pointcloud script with depth-color alignment.

Thank you for your help ,I 'll try .

OneShz, 2024-03-07:
> [isl-org/Open3D#473 (comment)](https://github.com/isl-org/Open3D/issues/473#issuecomment-508139946) has a Python / Open3D pointcloud script with depth-color alignment.

Thank you for your help ,I'll try.

closed 2024-03-21 by MartyG-RealSense