realsenseai/librealsense #12733: RealSense D435 camera getting highest point's distance

erayak, 2024-03-07:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                         |  D435 | 
| Firmware Version                     |  5.15.1  | 
| Operating System & Version  |  Ubuntu  22.04 LTS | 
| Kernel Version (Linux Only)    |  6.5.0-14-generic  | 
| Platform                                    |  PC  |
| SDK Version                             |  2.54.2.5684  | 
| Language                                 |  Python  | 
| Segment			          |  Others   | 

Hi, 

Is there a function in the RealSense camera SDK to determine the distance from the camera to the highest point of an object using bounding box coordinates (x1, y1, x2, y2)? If there isn't a function how can we achieve that?

Thanks.

MartyG-RealSense, 2024-03-07:
Hi @erayak  If you know the distance from the camera to the highest point and you also know the angle then you may be able to calculate the height using **trigonometry** mathematics, as described at https://github.com/IntelRealSense/librealsense/issues/9297

erayak, 2024-03-08:
Hi @MartyG-RealSense,

We don't know the highest point of the object we detected. Actually we are trying to do is finding the highest point of the detected object and get the distance from camera of that highest point. We think about the searching the highest point all through our coordinates with using get_distance function in the SDK. But that would slow the program so much. Is there any way other way to achieve that?

Also, the get_distance function uses angle information to calculate the distance, right? If not, it will calculate the distance larger or smaller than it actually is when the object moves away from 90 degrees below the camera, for example when the camera is on top and the object is in a plane below the camera.

erayak, 2024-03-11:
Hi @MartyG-RealSense,

How do you think we can proceed for a solution? We are trying to use the function `depth_frame.get_data()` to achieve our goal, can we give specific coordinates as parameters?

MartyG-RealSense, 2024-03-17:
Apologies for the delay in responding further.

If the background could be excluded from the image so that only the object is rendered on it, perhaps by using a post-processing **Threshold** filter, then the highest point could be the depth value on the object that is furthest from the camera.

MartyG-RealSense, 2024-03-23:
Hi @erayak  Do you require further assistance with this case, please?  Thanks!

erayak, 2024-03-28:
Hi @MartyG-RealSense thanks for your support

MartyG-RealSense, 2024-03-28:
You are very welcome.  As you do not require further assistance, I will close this case.  Thanks again!

closed 2024-03-28 by MartyG-RealSense