realsenseai/librealsense #12734: Is it possible to run on chip self calibration with advanced settings such as disparity shift?

mattiasbax, 2024-03-07:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415                                                       | 
| Firmware Version                   | 5.15.1.0                                                   | 
| Operating System & Version |  Linux                                                      |  
| Kernel Version (Linux Only)    |  6.6.12                                                    | 
| Platform                                 | Intel NUC                                                |
| SDK Version                            |  2.40.0                                                    | 
| Language                            |  C++                                                          | 
| Segment			|  Other                                                              | 

### Issue Description
Hello,

I want to be able to monitor health and recalibrate our cameras in a production environment. However, I am using the camera to monitor close up objects and thus use disparity shift to see objects in the range of interest.

When I try to run on chip self calibration using the Realsense-Viewer it looks like settings such as disparity shift is being reset. This then leads to the self calibration failing due to not having enough defined depth points (because the "wall" is too close to the camera).

When trying the self calibration programmatically via the SDK I cannot set the disparity shift prior to calling the self calibration. This will lead to errors such as "HW not ready" or similar. If I don't set the disparity shift via the SDK the self calibration will throw exception due to timeout (which I guess is because it's not converging due to the too close "wall" it is looking at).

My question is, should it be possible to use more advanced settings prior to self calibration, or will it always fall back to a certain preset of settings? 

Thanks!

MartyG-RealSense, 2024-03-07:
Hi @mattiasbax  Instead of performing a calibration, you could try resetting the camera's calibration to its factory-new default settings using the instructions at https://github.com/IntelRealSense/librealsense/issues/10182#issuecomment-1019854487 to see whether this provides satisfactory results.

If you need to monitor objects at close range then the **D405** camera model may work better for you.  It provides high quality, high accuracy close-range images at a depth sensing range of 7 cm to 50 cm without having to use disparity shift.

https://www.intelrealsense.com/depth-camera-d405/

If the wall that you are directing the camera at is white then the white-paper guide for self-calibration informs that the D415 model specifically has difficulty with this kind of wall.  It recommends against using D415 for white-wall calibration, but if you need to do so then it suggests selecting the **High Accuracy** visual preset mode and setting the self-calibration's Speed to the **white wall** mode.  

With these settings, it may be possible to avoid using disparity shift, though the camera will have to be positioned further from the wall so that its distance is above the D415's minimum depth sensing distance.

https://dev.intelrealsense.com/docs/self-calibration-for-depth-cameras

mattiasbax, 2024-03-07:
Thank you @MartyG-RealSense for the quick response.

Changing to a different camera model is unfortunately not possible currently.

If i understand you correctly it **should** be possible to run the self calibration with advanced settings such as disparity shift turned on?

mattiasbax, 2024-03-11:
I got self calibration for close up targets to work using disparity shift by updating to the latest SDK version.

You can close this issue, thank you @MartyG-RealSense !

MartyG-RealSense, 2024-03-17:
You are very welcome, @mattiasbax - I'm pleased to hear that you achieved a solution regarding disparity shift.  Thanks very much for the update.  As you suggested, I will close this case.  Thanks again!

closed 2024-03-17 by MartyG-RealSense