realsenseai/librealsense #12735: Jetson Nano Orin error only when sourcing the ros2 in the terminal

mohamed1915, 2024-03-07:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 5.15.1 | 
| Operating System & Version |  ubuntu 22.04 | 
| Kernel Version (Linux Only)    |  5.15.122-tegra    | 
| Platform                                 |  NVIDIA Jetson Nano Orin  |
| SDK Version                            |  v2.54.2  | 
| Language                            |  C++ | 
| Segment			|   Robot  | 

### Issue Description
I can access the camera's data (RGB, stereo and IMU) through realsense-viewer and through C++ code as long as I don't source my ros2 iron code, if I did I get the error :
14:50:32 [Warn] ./src/ds/d400/d400-factory.cpp:1195 - DS5 group_devices is empty.
And even though I can still see it in lsusb, It doesn't even show on the realsense-viewer nor does it show on the c++ code. 
Do you have an idea on how to debug this or figure out where the issue was, I used : 
printenv > env_before.txt
source /opt/ros/<ros2_distro>/setup.bash
printenv > env_after.txt
diff env_before.txt env_after.txt

but didn't learn much from it.
Thanks,

mohamed1915, 2024-03-07:
Thank you for your quick response. I tried both, Ros2 wrapper of the Realsense and a simple code that I wrote only using realsense as a C++ library, and both give the no device detected. However I tried to work around this by building a simple publisher of the data yet I get the same error, and while debugging I found out that just sourcing ROS2 makes the Realsense-viewer not detect the camera.

martinerk0, 2024-03-11:
Isn't it similar to this? https://github.com/IntelRealSense/realsense-ros/issues/2973
That when you install realsense(sdk) library manually, and then install realsense-ros wrapper, it also installs another realsense(sdk) with apt and there are conflicts?
What is output of: 
```
dpkg -l | grep realsense
```
?

mohamed1915, 2024-03-11:
```
ii  ros-iron-librealsense2                          2.54.1-2jammy.20240207.114722               arm64        Library for controlling and capturing data from the Intel(R) RealSense(TM) D400 devices.
ii  ros-iron-realsense2-camera                      4.54.1-2jammy.20240213.170727               arm64        RealSense camera package allowing access to Intel SR300 and D400 3D cameras
ii  ros-iron-realsense2-camera-dbgsym               4.54.1-2jammy.20240213.170727               arm64        debug symbols for ros-iron-realsense2-camera
ii  ros-iron-realsense2-camera-msgs                 4.54.1-2jammy.20240213.160829               arm64        RealSense camera_msgs package containing realsense camera messages definitions
ii  ros-iron-realsense2-camera-msgs-dbgsym          4.54.1-2jammy.20240213.160829               arm64        debug symbols for ros-iron-realsense2-camera-msgs
ii  ros-iron-realsense2-description                 4.54.1-2jammy.20240213.164734               arm64        RealSense description package for Intel 3D D400 cameras
```
Here is the output of 
`dpkg -l | grep realsense`

martinerk0, 2024-03-11:
Try `sudo apt remove ros-iron-librealsense2` and see if it works (you can always reinstall it)

mohamed1915, 2024-03-11:
Yep, it works, Thank you so much @martinerk0 !

mohamed1915, 2024-03-26:
Thank you guys, the  C++ works for publishing the data, However I don't have a running python wrapper of the realsense (sdk) library, So I can't access the default ROS2 packages existing in the library. Is there a file I need to source or a step I have missed to get the wrapper to run on my working library?

MartyG-RealSense, 2024-03-26:
Hi @mohamed1915  You could try installing the Python wrapper with the command **pip install pyrealsense2** as this install method has had support for Arm device installation since SDK 2.54.1.

mohamed1915, 2024-03-26:
`pip install pyrealsense2 ` doesn't work, that's why I install the ROS2 version previously, here is the error
`ERROR: Could not find a version that satisfies the requirement pyrealsense2 (from versions: none)
ERROR: No matching distribution found for pyrealsense2
`

MartyG-RealSense, 2024-03-26:
The Arm version of the pip install file has the limitation that it will only work with Python 3.7, 3.8 and 3.9.

mohamed1915, 2024-03-26:
Yeah and I have 3.10 running in Ubuntu 22, is there a way to install the wrapper that I assume exists in the library that I already have installed ? or maybe I just need to source a certain .so file ? or add a path to the python path ?

MartyG-RealSense, 2024-03-26:
The wrapper can be installed from source code for use with newer Python versions not currently supported by a pip install.  The wrapper can either be built from source at the same time as installing librealsense or separately afterwards by using the instructions at the link below.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python#building-from-source

closed 2024-04-04 by mohamed1915

MartyG-RealSense, 2024-04-04:
You are very welcome, @mohamed1915 - I'm pleased to hear that a source code installation worked for you.  Thanks very much for the update!