realsenseai/librealsense #12737: position of bounding box from image doesn't match with point cloud

nokey93, 2024-03-08:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 5.15.1 | 
| Operating System & Version |   Win 10  | 
| Kernel Version (Linux Only)    |  N/A                                       | 
| Platform                                 | PC |
| SDK Version                            | 2.54.2.5684                           | 
| Language                            |  python                          | 
| Segment			|  others                           | 

Hi everyone!
I'm trying to find the position of the object by using yolov5 and match the bounding box with point cloud. 
But when i try to make it match with the point cloud, the bounding box is smaller to compare with the position of the object. For example is the pictures I took from my process.
[image]
[image]

Is the point cloud bigger than the image I use for find the objects? 
Is there a function to help me align the bouding box from image to the point cloud?
Thank you in advance!

MartyG-RealSense, 2024-03-09:
Hi @nokey93  On the D435i camera model, the field of view size of its RGB color sensor is smaller than the field of view size of the depth sensor, so the RGB sensor cannot see as much of the observed scene as the depth sensor can.  

The differences can be compensated for by using a depth-color alignment instruction such as `align_to` to map the depth points onto the color points.  This is depth to color alignment.

nokey93, 2024-03-10:
Hi @MartyG-RealSense i did align the depth and RGB image and the second picture is the point cloud, which i made from RGB and depth images.
The Bounding box to recognize objects in Point cloud is smaller than from normal image.
So it's actually, how I can align the Bounding box from normal image to the point cloud (sorry for the confused from my post).
Like I understand, the function align_to just can align RGB with Depth, not with Point cloud. If it's false, please correct me!

MartyG-RealSense, 2024-03-17:
Hi @nokey93  An approach that a RealSense user took at https://github.com/IntelRealSense/librealsense/issues/6267#issuecomment-618261685 was to use Python and Open3D to perform depth-color alignment with align_to, then generate a pointcloud with the **pc.calculate()** instruction and finally draw a bounding box on the RGB image.

nokey93, 2024-03-21:
@MartyG-RealSense thank you for your suggestion!
I try to use this code to create point cloud 
https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/export_ply_example.py

There is other problem, from above I have a ring light to make it easier to see the object in the box, but the point cloud is kinda messy and not in the coordinate origin (I use Meshlab to check the point cloud).
[image]

when I turn the light off, then the point cloud is kinda ok, still not in the middle of the coordinate origin, but i can't see the object in the picture i took from box
[image]

And btw, the point cloud is a little bit wider than the resolution I set up, which is 640x480
[image]

Is there anyway to improve this situation, when I want to keep the light on to see the object in box?
I guess the wrong position of bounding box in point cloud, because the 3D point cloud is not the same coordination with the 2D, but I did try to align the depth and color image in the script realsense_depth
`import pyrealsense2 as rs
import numpy as np

class DepthCamera:
    def __init__(self):
        # Configure depth and color streams
        self.pipeline = rs.pipeline()
        config = rs.config()

        # Preferred settings for the streams
        config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
        config.enable_stream(rs.stream.color, 640, 480, rs.format.rgb8, 30)

        # Start the pipeline and keep the profile as an attribute
        self.profile = self.pipeline.start(config)
        
        # Initialize filters
        self.filters = [rs.disparity_transform(),
                        rs.spatial_filter(),
                        rs.temporal_filter(),
                        rs.disparity_transform(False)]
        
        # Alignment from depth to color
        self.align = rs.align(rs.stream.color)

    def get_frame(self):
        frames = self.pipeline.wait_for_frames()
        
        # Get depth frame before alignment to apply filters
        depth_frame = frames.get_depth_frame()
        if not depth_frame:
            return False, None, None
        
        # Apply filters
        for f in self.filters:
            depth_frame = f.process(depth_frame)
        
        # Now align the frames
        aligned_frames = self.align.process(frames)
        depth_frame = aligned_frames.get_depth_frame()  # Re-fetch the aligned depth frame
        color_frame = aligned_frames.get_color_frame()

        if not color_frame:
            return False, None, None

        # Convert images to numpy arrays
        depth_image = np.asanyarray(depth_frame.get_data())
        color_image = np.asanyarray(color_frame.get_data())
        return True, depth_image, color_image

    def release(self):
        self.pipeline.stop()`

MartyG-RealSense, 2024-03-21:
If you are exporting a .ply file from RealSense and importing it into MeshLab then a guide at the link below provides useful advice if you have not seen it already.

https://www.andreasjakl.com/capturing-3d-point-cloud-intel-realsense-converting-mesh-meshlab/

Does the depth image quality improve with the ring light on if you move the light a little further away or turn the light's intensity down if it has such a control button?  

If not, in the Viewer you could also try changing the **Emitter Enabled** option's drop-down menu from 'Laser' to 'Off' to test whether the emitter, in combination with the ring light's output, is making the intensity of light cast on the object surfaces too great and making the surface harder to read by blinding the camera.

I also recommend trying to change the 'Preset' option from its default of "Custom" to the setting "Medium Density" to see whether the depth quality improves.

MartyG-RealSense, 2024-03-22:
If you want to export a point cloud .ply with Python that has color data in it then that will be a problem, as color export to .ply typically only works with C++ code except for an export_to_ply() script at https://github.com/IntelRealSense/librealsense/issues/6194#issuecomment-608371293 which is the only code example that has been shown to perform color export.

closed 2025-01-06 by MartyG-RealSense