realsenseai/librealsense #12738: Intelrealsense D435i  No Camera feed

Syedifthikar, 2024-03-09:
| | |
|---|---|
|**librealsense**|2.54.2 RELEASE|
|**OS**|Linux|

[image]

During the execution of a code with the camera, I encountered an error and subsequently checked the camera feed, only to find that it was not displaying. However, the error message indicated that the RealSense camera was detected.

please find image attached

MartyG-RealSense, 2024-03-09:
Hi @Syedifthikar  I note that you are using a Raspberry Pi.

Does the camera appear in the Viewer if you unplug it from the USB port, wait a couple of seconds and then re-insert it?

Syedifthikar, 2024-03-10:
Dear IntelRealSense/librealsense,

Same issue  after following the above.
I have attached the images for the same.

MartyG-RealSense, 2024-03-16:
Hi @Syedifthikar  Apologies for the delay in responding further.  Your images did not attach as images cannot be sent to this support forum by email.  They have to be pasted into the comment writing box on this web-page.

closed 2024-03-29 by MartyG-RealSense