realsenseai/librealsense #12740: Cannot detect d435i device on Ubuntu 22.04

Norweig1an, 2024-03-10:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435i                                                     | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Linux Ubuntu 22.04                              | 
| Kernel Version (Linux Only)    |  6.2.0-32-generic                                   | 
| Platform                                 | PC                                                          |
| SDK Version                            |   legacy / 2.54.2                                    | 
| Language                               | python3.10.10                                        | 
| Segment		          	|  Robot                                                     | 

### Issue Description
The PC could recognized the device cause after `lsusb | grep 8086`, I got output `Bus 002 Device 002: ID 8086:0b3a Intel Corp. Intel(R) RealSense(TM) Depth Camera 435i`.
However,
[image]

MartyG-RealSense, 2024-03-11:
Hi @Norweig1an  Which command are you entering that causes the list in the image above to be generated, please?

Norweig1an, 2024-03-11:
If I understand it clearly, I just command `realsense-viewer` and the output generated.

MartyG-RealSense, 2024-03-17:
Apologies for the delay in responding further.

In the log image that you supplied, it looks as though the launch of realsense-viewer is having problems with rendering the graphics.  

Which method did you use to install the librealsense SDK on your computer, please?

You could also try updating the video drivers on your computer to see whether a missing or outdated video driver is causing the graphics not to be generated.

Another cause of such a problem could be if a computer is in a 'headless' configuration (a computer that does not have a display monitor screen attached).

MartyG-RealSense, 2024-03-23:
Hi @Norweig1an  Do you require further assistance with this case, please?  Thanks!

Norweig1an, 2024-03-23:
Thanks for your help, I found that the driver of realsense conflict with the driver of nvidia graphics card, and after I downgrade the version of Nvidia driver it solved.

closed 2024-03-23 by Norweig1an