realsenseai/librealsense #12746: Offline retrieval of (complete) pointcloud data for a depth-align RGB and depth capture using Intel RealSense D435

monajalal, 2024-03-12:
I have done a depth-align capture that includes depth and RGB frames and I would like to know if I can retrieve the associated pointcloud for each frame? Is there any script you could please provide for this task?

I don't know if it matters but I did my capture with ROS Noetic with a command like

`$ rosbag record -O mona_noetic.bag /camera/aligned_depth_to_color/image_raw /camera/color/image_raw /camera/depth/camera_info /camera/color/camera_info /camera/extrinsics/depth_to_color`

That said, I am also interested to know if I could obtain depth-aligned point cloud associated with the corresponding RGB+depth frame bundles using ROS Noetic if I do another capture?

* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

MartyG-RealSense, 2024-03-12:
Hi @monajalal  With the `rosbag record` instruction for aligned topics that you have used, Intel's ROS1 D435i guide for generating a pointcloud, saving the aligned topics to bag and playing back the bag comes to mind.

https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i
 
In that guide, the recorded bag was played back to display the pointcloud in RViz by inputting the following list of commands.

```
roscore >/dev/null 2>&1 &
rosparam set use_sim_time true
rosbag play my_bagfile_1.bag --clock
roslaunch realsense2_camera opensource_tracking.launch offline:=true
```

As these instructions are designed for a RealSense camera equipped with an IMU such as D435i and D455, and you have an IMU-less D435, you may need to use a different launch file such as **rs_camera.launch**.  I hope that the above information will be helpful to you though.

MartyG-RealSense, 2024-03-21:
Hi @monajalal  Do you require further assistance with this case, please? Thanks!

monajalal, 2024-03-22:
I am slow on responses on my end. I will get back to you when I get a chance. Thanks a lot for providing a clarification.

closed 2024-03-26 by monajalal