realsenseai/librealsense #12747: Getting  module rs_connect error when i run my and code ,depth disttance also not displaying in realsense viewer

Syedifthikar, 2024-03-13:
[image]

| | |
|---|---|
|**librealsense**|2.54.2 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense D435I|
|**Serial Number**|233522071788|
|**Firmware Version**|5.15.1|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|D400|
|**Asic Serial Number**|221123063471|
|**Firmware Update Id**|221123063471|

I am getting a 'module rs_connect' error when I run my code, and the depth distance is also not displaying in the RealSense viewer. I have tried to do calibration, but it fails. Please help me with this issue

MartyG-RealSense, 2024-03-13:
Hi @Syedifthikar  Which model of Raspberry Pi board are you using, please?  Boards earlier than Pi 4, such as Pi 3, only had a USB 2.1 port instead of USB 3.  Your program is apparently loading settings from a json camera configuration file called **d4xx-default.json**, including stream resolution.

 https://github.com/thien94/vision_to_mavros/blob/master/scripts/d4xx_to_mavlink.py#L394

The resolution in that json is 848x480 and the FPS is 30, but this resolution is only available at 6 FPS on a D435i when the USB connection is 2.1.  So if you are using a pre-Pi4 board with a USB 2.1 port or you have a Pi 4 with a USB 3 port but the camera is being mis-detected as USB 2.1, then you would get the **RunTimeError: Couldn't resolve requests** error in your image.

The error means that the camera was unable to provide the resolution and / or FPS that had been requested, because the json file was setting the FPS to 30 but only 6 was supported for the requested resolution on USB 2.1.

 You could try editing **d4xx-default.json** to change `stream_width` to 640 and` stream_height` to 480, as 640x480 at 30 FPS is supported on both USB 2.1 and USB 3.2.

MartyG-RealSense, 2024-03-21:
You are very welcome.  Thanks very much for the update.  As you do not require further assistance, I will close this case.  Thanks again!

closed 2024-03-21 by MartyG-RealSense