realsenseai/librealsense #12749: RaspberryPi RSTP rs-server Multicast

JuanDYB, 2024-03-13:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.14.00.00 | 
| Operating System & Version |   Raspberry Pi4B / Win10  | 
| Kernel Version (Linux Only)    |  6.1.0-rpi7-rpiv8                                         | 
| Platform                                 | PC/Raspberry Pi  |
| SDK Version                            |  2.53.1                           | 
| Language                            |  C#                           | 
| Segment			|                            | 

### Issue Description
I want to access IntelRealSense D415 camera over ethernet. For that I'm using a Raspberry pi and I have run the `libuvc_installation.sh` script with a `Cmake` params modification. I have added `-DBUILD_NETWORK_DEVICE=ON`.

With this modification I can run `rs-server` service on RaspberryPi, and I'm able to connect and see streaming in a Windows computer using `C#` wrapper over Ethernet.

```
var context = new Intel.RealSense.Context();
var device = new NetDevice(ipAddress);
device.AddTo(context);
```

The problem is that I cannot connect any other clients to RaspberryPi, because if there is any other client connected it fails when another come. It isn't a multicast behaviour.

How can I support multiple connections to raspberrypi?

MartyG-RealSense, 2024-03-14:
Hi @JuanDYB  Have you attempted to open more than one terminal window like in the link below in order to have multiple SSH connections to the same Raspberry Pi, please?

https://raspberrypi.stackexchange.com/questions/87387/multiple-simultaneous-ssh-connections

JuanDYB, 2024-03-14:
Hi @MartyG-RealSense 
I can open multiple ssh without problems. My problem is with `rs-server`. With ssh or SFTP I don't have any problem.

MartyG-RealSense, 2024-03-17:
If the problem is with the RTSP protocol then there has been a case at https://github.com/IntelRealSense/realsense-ros/issues/1808 where a RealSense user modified the ethernet tool for use with ROS networking protocols instead.
The official documentation for the rs-server tool suggests in its **Further Research** section that it might be modifiable for multi-client broadcasting if it used the **RTP** protocol.

https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras#62-further-research

In the references section at the bottom of the doc, it provides a link to a PDF paper for further reading about RTP and RTSP protocols.

https://www.cse.wustl.edu/~jain/books/ftp/rtp.pdf

The rs-server networking tool is no longer supported by Intel and has been removed from the librealsense SDK, so will not be developed further by Intel.  The introduction of a new networking interface is planned in an upcoming SDK version.  The rs-server tool could though potentially be customized by RealSense users by downloading the source code of an SDK version that supports it (2.53.1 or older) and going to the rs-server section of the source code folder at **librealsense > tools > rs-server**

[image]

closed 2024-03-29 by MartyG-RealSense