realsenseai/librealsense #12751: D405 Not working on realsense-viewer

vilakshan2996, 2024-03-14:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D400 } | 
| Firmware Version                   | (5.15.1) | 
| Operating System & Version |   Win 11  | 
| Platform                                 | PC |
| SDK Version                            |  { 2.0 }                          | 
| Language                            |  python                         | 
| Segment			|  Robot}                          | 

### Issue Description
The Intel RealSense Viewer is giving particular error , when i add the depth camera as a source.
```
Invalid Value in rs2_get_option_range(options:0000025B8F3C4F20,
option:Depth Units, min:2.8026e-45, max:1, step:1, def:0):
hwmon command 0x2c( 9 1 0 0 ) failed (response -9= No expected user action)
```

```
Invalid Value in rs2_get_option(options:0000025B8F3C4F20, option:Depth
Units):
hwmon command 0x2c( 9 0 0 0 ) failed (response -9= No expected user action)
```

$ rs-enumerate-devices: 

Device info:
    Name                          :     Intel RealSense D400
    Serial Number                 :     ffffffffffff
    Firmware Version              :     5.15.1
    Recommended Firmware Version  :     5.15.1
    Physical Port                 :     \\?\usb#vid_8086&pid_0ad1&mi_00#6&32ab0d3e&0&0000#{e5323777-f976-4f5b-9b55-b94699c46e44}\global
    Debug Op Code                 :     15
    Advanced Mode                 :     YES
    Product Id                    :     0AD1
    Camera Locked                 :     YES
    Usb Type Descriptor           :     3.2
    Product Line                  :     D400
    Asic Serial Number            :     128223070408
    Firmware Update Id            :     128223070408

Stream Profiles supported by Stereo Module
 Supported modes:
    stream       resolution      fps       format
    Infrared      1280x720      @ 30Hz     UYVY
    Infrared      1280x720      @ 30Hz     BGRA8
    Infrared      1280x720      @ 30Hz     RGBA8
    Infrared      1280x720      @ 30Hz     BGR8
    Infrared      1280x720      @ 30Hz     RGB8
    Infrared      1280x720      @ 15Hz     UYVY
    Infrared      1280x720      @ 15Hz     BGRA8
    Infrared      1280x720      @ 15Hz     RGBA8
    Infrared      1280x720      @ 15Hz     BGR8
    Infrared      1280x720      @ 15Hz     RGB8
    Infrared      1280x720      @ 6Hz      UYVY
    Infrared      1280x720      @ 6Hz      BGRA8
    Infrared      1280x720      @ 6Hz      RGBA8
    Infrared      1280x720      @ 6Hz      BGR8
    Infrared      1280x720      @ 6Hz      RGB8
    Infrared      848x480       @ 90Hz     UYVY
    Infrared      848x480       @ 90Hz     BGRA8
    Infrared      848x480       @ 90Hz     RGBA8
    Infrared      848x480       @ 90Hz     BGR8
    Infrared      848x480       @ 90Hz     RGB8
    Infrared      848x480       @ 60Hz     UYVY
    Infrared      848x480       @ 60Hz     BGRA8
    Infrared      848x480       @ 60Hz     RGBA8
    Infrared      848x480       @ 60Hz     BGR8
[... 196 lines cut ...]
    Depth         848x480       @ 30Hz     Z16
    Depth         848x480       @ 15Hz     Z16
    Depth         848x480       @ 6Hz      Z16
    Depth         848x100       @ 100Hz    Z16
    Depth         640x480       @ 90Hz     Z16
    Depth         640x480       @ 60Hz     Z16
    Depth         640x480       @ 30Hz     Z16
    Depth         640x480       @ 15Hz     Z16
    Depth         640x480       @ 6Hz      Z16
    Depth         640x360       @ 90Hz     Z16
    Depth         640x360       @ 60Hz     Z16
    Depth         640x360       @ 30Hz     Z16
    Depth         640x360       @ 15Hz     Z16
    Depth         640x360       @ 6Hz      Z16
    Depth         480x270       @ 90Hz     Z16
    Depth         480x270       @ 60Hz     Z16
    Depth         480x270       @ 30Hz     Z16
    Depth         480x270       @ 15Hz     Z16
    Depth         480x270       @ 6Hz      Z16
    Depth         424x240       @ 90Hz     Z16
    Depth         424x240       @ 60Hz     Z16
    Depth         424x240       @ 30Hz     Z16
    Depth         424x240       @ 15Hz     Z16
    Depth         424x240       @ 6Hz      Z16
    Depth         256x144       @ 90Hz     Z16

MartyG-RealSense, 2024-03-14:
Hi @vilakshan2996  I note that you list your camera's serial number as **ffffffffffff**.  This indicates that there is a problem.  In an example of this issue at https://github.com/IntelRealSense/librealsense/issues/7337#issuecomment-693120103 the RealSense user in that case identified a bent USB cable connector as the cause of their particular serial number problem.

You also state that you "add the depth camera as a source".  Do you mean that you are using the **Add Source** button in the RealSense Viewer to list the camera in the options side-panel?  If you are then this would also indicate a problem, as the camera should appear automatically in the panel when plugged in. Failure to do so could also be a USB related issue.

vilakshan2996, 2024-03-14:
Hi @MartyG-RealSense , It seems that the RealSense viewer is automatically recognizing the device, which suggests that the issue is not with the USB connection itself. After I removed and re-added the device as a source, the connectivity was confirmed to be fine. However, the error message appears immediately upon selecting the device or opening the application.

Additionally, there are no bent USB connectors in my setup.

[image]

MartyG-RealSense, 2024-03-14:
Thanks very much for the information.  The camera name is showing as **D400**, which is incorrect.  It should be displaying as D405.

Are you able to access the 'Calibration Data' menu option under the **More** option in the side-panel?  If you can, please try using the instructions at https://github.com/IntelRealSense/librealsense/issues/10182#issuecomment-1019854487 to reset the calibration data of your camera to its factory-new default settings to see whether the camera can then be identified correctly as D405.

vilakshan2996, 2024-03-15:
Even after selecting the factory reset option, table doesn’t update. Additionally, the write button remains unclickable and disabled even following a factory reset.

[image]

When I initiated calibration using the Dynamic Calibrator, I encountered these error messages.

[image]

MartyG-RealSense, 2024-03-17:
Hi @vilakshan2996  The depth stream has to be started by left-clicking the red icon beside 'Stereo Module' before opening the Camera Calibration window in order for the Write Table button to become clickable.  

Even if the calibration information does not update when clicking Restore Factory, writing a calibration to the camera could repair the camera's internal 'calibration table' if it is corrupted.  A corrupted calibration table can cause the camera to not function correctly.

vilakshan2996, 2024-03-17:
I am getting an error when I start streaming the data (the button is not turning on).
```
Timestamp: 1710688306753.557617
Severity: Error
Description: Invalid Value in rs2_open_multiple(sensor:000002925C254940, profiles:000002925C3C5098, count:1):
hwmon command 0x2c( 9 0 0 0 ) failed (response -9= No expected user action)

```
[image]

MartyG-RealSense, 2024-03-19:
I have consulted with my colleagues.  Please send me an email at the address below so that we can discuss the next step in assisting with your issue.

martyx.grover@intel.com

MartyG-RealSense, 2024-05-18:
Hi @vilakshan2996  Can this GitHub case be closed now?  Thanks!

closed 2024-05-26 by MartyG-RealSense