realsenseai/librealsense #12752: D415 plug/unplug crashed

LLY13, 2024-03-14:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415| 
| Firmware Version                   |   05.11.01.100    | 
| Operating System & Version |   win10 | 
| Platform                                 | PC |
| SDK Version                            |  v2.51.1   (because we use 32-bit and this is the last version including 32-bit realsense2.dll)    | 
| Language                            |  java invoke c++              | 
| Segment			|  Robot    | 

### Issue Description
i use a usb and a 10-meter fiber connected to computer.  when i unplug and replug , sometimes it crashes.  Most time crashed in wait_for_frames. Need to know how to slove this problem.   And when i use Intel realsense viewer.exe to repeat the same actions, most time it can work. 

[image]

[hs_err_pid19272.log](https://github.com/IntelRealSense/librealsense/files/14599101/hs_err_pid19272.log)
[hs_err_pid12592.log](https://github.com/IntelRealSense/librealsense/files/14599102/hs_err_pid12592.log)
[hs_err_pid23804.log](https://github.com/IntelRealSense/librealsense/files/14599103/hs_err_pid23804.log)
[hs_err_pid24796.log](https://github.com/IntelRealSense/librealsense/files/14599100/hs_err_pid24796.log)

MartyG-RealSense, 2024-03-14:
Hi @LLY13  I would first recommend updating your camera firmware driver from 5.11.1.100 (which is 5 years old at the time of writing this) to version **5.13.0.50** (the recommended firmware for SDK 2.51.1).  Using the recommended firmware for a particular SDK version can help to avoid errors and you also get the benefit of bug-fixes and improvements accumulated over time (such as better USB stability when plugging-unplugging).

https://dev.intelrealsense.com/docs/firmware-releases#d400-series-firmware-downloads

LLY13, 2024-03-15:
> Hi @LLY13 I would first recommend updating your camera firmware driver from 5.11.1.100 (which is 5 years old at the time of writing this) to version **5.13.0.50** (the recommended firmware for SDK [...]

I've updated firmware driver to 5.13.0.50. I tested unplug/replug and it still stuck in wait_for_frames sometimes when i unplug it. No idea why it happened.  I posted code and can u check what i can do?

GetFrames() is the main function invoked by java every 10ms after initing.  Most time it stuck or crashed in wait for frame

`
av_CameraStatus CameraRealSense::GetFrames()
	{
		std::cout << "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx" << std::endl;
		//if (!isOpen) return RS_NoDev;	
		if (!isOpen) {
			Status = AV_CameraStatus_Onlineing;
			//if (!isInit)CameraStatus = AV_CameraStatus_NoInit;
			if (!OpenCmd)Status = AV_CameraStatus_isClose;
			else {
				Sleep(500);
				CameraRealSense::getInstance()->GetDevCount();
				std::cout << "CameraRealSense::GetDevCount()=" << CameraRealSense::getInstance()->DevCount << std::endl;
				if (CameraRealSense::getInstance()->DevCount <= 0)
					Status = AV_CameraStatus_NoDev;
				else {
					std::cout << "dev count" << CameraRealSense::getInstance()->DevCount << std::endl;
					CameraRealSense::getInstance()->OpenDev();
				}
			}

			return Status;
		}

		try {
			std::cout << "qqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqq" <<  std::endl;
			auto frames = pipe.wait_for_frames(5000);
			std::cout << "wwwwwwwwwwwwwwwwwwwwwwwwwwwwwww" << std::endl;
			while (pipe.poll_for_frames(&frames)) {};
			//std::lock_guard<std::mutex> lock(rfs.Cameramutex1);
			//
			if (!frames || frames.size() < 3)return AV_CameraStatus_NoFrame;
			fir = frames.get_infrared_frame();
			if (!fir)  return AV_CameraStatus_NoIrFrame;
			if (fir.get_data_size() < rfs.Ir.Size.width*rfs.Ir.Size.height)return AV_CameraStatus_NoIrFrame;
			fcolor = frames.get_color_frame();
			if (!fcolor)  return AV_CameraStatus_NoColorFrame;
			int n = (int)(fcolor.get_data_size() / 3);
			int n1 = rfs.Color.Size.width*rfs.Color.Size.height;
			if (n < (n1))return AV_CameraStatus_NoColorFrame;
			fdepth = frames.get_depth_frame();
			if (!fdepth)  return AV_CameraStatus_NoDepthFrame;
			n = (int)(fdepth.get_data_size() / 3);
			n1 = rfs.Depth.Size.width*rfs.Depth.Size.height;
			//rs2::video_stream_profile depth_profile = profile.get_stream(RS2_STREAM_DEPTH).as<rs2::video_stream_profile>();
			//n1 = depth_profile.height()*depth_profile.width();
			//if (n<(n1))return AV_CameraStatus_NoDepthFrame;
			//深度图滤波--------------------------------------------------------
			// 抽取滤波器
			//fdepth = dec_filter.process(fdepth);
			//
			//fdepth = depth2disparity.process(fdepth);
			// 空间滤波器
			//if (isOpenTemporal_filter)
			//fdepth.frame = spat_filter.process(fdepth.frame);
			// 时域滤波器
			if (Temporal_filter_enabled)
				fdepth = temp_filter.process(fdepth);
			//fdepth = hole_filter.process(fdepth);
			// If we are in disparity domain, switch back to depth
			//fdepth = disparity2depth.process(fdepth);			
			rfsUpData();
			std::cout << "sssssssssssssssssssssssssss" << std::endl;
			ErrCount = 0;
[... 115 lines cut ...]
				//rs2::pipeline p;
				//pipe = p;
				//AR::debug() << "profile = pipe.start(cfg)";
				AR::debug() << "Color,width:"<< rfs.Color.Size.width<<" height:"<< rfs.Color.Size.height<<"framerate:"<< rfs.Color.Framerate;
				AR::debug() << "Depth,width:" << rfs.Depth.Size.width << " height:" << rfs.Depth.Size.height << "framerate:" << rfs.Depth.Framerate;
				AR::debug() << "Ir,width:" << rfs.Ir.Size.width << " height:" << rfs.Ir.Size.height << "framerate:" << rfs.Ir.Framerate;
				try 
				{ 
					pipe.get_active_profile(); 
				}
				catch (std::exception& e) {
					profile = pipe.start(cfg);
					//Set_Para();
					AR::debug() << "Camera Open OK ";
				}
				AR::debug() << "nnnnnnnnnnnnnnnnnnnnnnnnnnnnnn";
				isOpen = TRUE;
				return AV_CameraStatus_OK;
			}
			catch (std::exception &e)
			{
				AR::debug() << "Camera Open1 try " << e.what();
				DoClose1();
				return AV_CameraStatus_OnlineErr;
			}`

MartyG-RealSense, 2024-03-17:
My knowledge of Java programming is admittedly limited but I will do my best to be of assistance.  

As your fiber cable connection mostly works with the RealSense Viewer tool, it seems unlikely that the cable is the problem.

Does it make a difference if you remove the 5000 from the brackets of wait_for_frames please?

I note that after the wait_for_frames instruction, you have a **poll_for frames** instruction.

`while (pipe.poll_for_frames(&frames)) {};`

If your application is a single-camera one then it is recommended to use wait_for_frames.  The wait_for_frames instruction waits for a complete frame to be delivered before providing it, whilst poll_for_frames (which is suited for multiple-camera applications) does not wait and supplies new frames immediately even if there is a chance that the frames are incomplete.  It can also be necessary to manually control when the CPU is put to sleep and for how long when using poll_for_frames, otherwise the CPU's usage could max out at 100%.  More information about this can be found at https://github.com/IntelRealSense/librealsense/issues/2422#issuecomment-423254709

LLY13, 2024-03-18:
tried removing 5000 from wait_for_frames and removing pipe.poll_for_frames from code. still sometimes stuck in wait_for_frames or java just quit the program when I unplug.

MartyG-RealSense, 2024-03-18:
Instead of unplugging the camera to quit the program, are you able to exit out of the Java program with Ctrl-C instead?

LLY13, 2024-03-19:
No. How can this help me

MartyG-RealSense, 2024-03-19:
I wondered whether exiting the application with Ctrl-C, if possible to do so, might provide a cleaner way of shutting down the program than unplugging the camera.

MartyG-RealSense, 2024-03-27:
Hi @LLY13  Do you require further assistance with this case, please?  Thanks!

LLY13, 2024-03-28:
No thanks.  Could u give me a demo of simplest unplug/replug code  I can test the problem is code or other last

MartyG-RealSense, 2024-03-28:
USB connection and disconnection events can be listened to in C++ with the **set_devices_changed_callback** instruction, which is discussed with a code example at https://github.com/IntelRealSense/librealsense/issues/9287

LLY13, 2024-03-29:
One question. Now when i unplug the usb,  after pipe.wait_for_frames(1000)   throws exception for 3 times. I'll call pipe.stop(). Is this a correct way or should i do it? if not, what code i should have after that

MartyG-RealSense, 2024-03-29:
Yes, that would be a reasonable way to approach dealing with an exception.

closed 2024-04-26 by MartyG-RealSense