realsenseai/librealsense #12757: how to get rgb map in matlab?

pgswag, 2024-03-15:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Operating System & Version |  win10| 
| Platform                                 | PC |
| Language                            |  MATLAB 

### Issue Description
1.How to get an RGB map in matlab (not a color depth map)?(as shown below)
[image]
2.How to get infrared right and left map in matlab?(as shown below)
[image]

MartyG-RealSense, 2024-03-15:
Hi @pgswag  The link below has a RealSense MATLAB wrapper script for obtaining the left and right infrared streams.  you will need to edit the two cfg lines to give each infrared stream an index number ('1' for left and '2' for right) and change the Y16 format to Y8 if you do not want Y16 format infrared.

https://community.intel.com/t5/Items-with-no-label/How-to-enable-both-Y16-unrectified-streams-from-D435-in-SDK2-0/td-p/618843

In regard to RGB, I am not aware of an RGB-only MATLAB wrapper script but you might be able to strip down a depth-RGB alignment script at https://github.com/IntelRealSense/librealsense/issues/4245 to only display the RGB stream without alignment.

There is also a script reference at https://github.com/IntelRealSense/librealsense/issues/7102#issuecomment-673623848 for accessing the RGB sensor only.

pgswag, 2024-03-15:
I have another question: if 640*480 is selected for both the depth map and the rgb map, is the position of the object in the image equivalent when the rgb map and the depth map are not registered?
For example, if the box in the scene presents the range of the picture in the depth map: the upper left point (50,60), the lower right point (70,80), then does the box present the picture in the rgb map in the same position, the upper left point (50,60), the lower right point (70,80)?

MartyG-RealSense, 2024-03-15:
The position of the RGB sensor is physically offset horizontally from the position of the depth origin (the left infrared sensor), so the two images will not correspond to each other exactly unless depth-color alignment (also known as registration) is performed.

Also, the D435 camera model's RGB sensor has a smaller field of view than the depth sensor, so the outer edges of an aligned image may be cut off.  This is because the RGB sensor's smaller view means that it cannot see as much of the observed scene as the depth sensor can and so there is not colour data in those edge areas to match to depth data.

pgswag, 2024-03-16:
Thank you so much for your assistance.

closed 2024-03-17 by pgswag