realsenseai/librealsense #12758: ModuleNotFoundError: No module named 'pyrealsense2' on running python script with Raspberry Pi 4B

Saurabh-Sac, 2024-03-16:
| Camera Model                       | Intel RealSense depth camera D435i | 
| Operating System & Version | Raspberry Pi OS 64-bit "Buster" |                          
| Platform                                 | Raspberry Pi 4B, 8GB, (SD card 32GB)  |
| Language                               | python 3.7 |                       
| RealSense-Viewer                  | 2.54.2 |    

Hi,
I am trying to run a python script to get the depth data for object detection task with YOLO v8 using Raspberry Pi 4B and Intel RealSense depth camera D435i. 

**The realsense-viewer is working perfectly fine.** 

**But when I am running a python script, I am getting an error:  ModuleNotFoundError: No module named 'pyrealsense2'**

For installation, I have followed the following repository: 
[https://github.com/datasith/Ai_Demos_RPi/wiki/Raspberry-Pi-4-and-Intel-RealSense-D435]

For Raspberry Pi OS 64-bit "Buster", I used this link: 
[https://support.pishop.us/article/137-official-links-to-raspberry-pi-os-buster]

For given error, I checked the following links to resolve it but unfortunately it is still unresolved:
1. [https://github.com/IntelRealSense/librealsense/issues/12404]
2. [https://github.com/IntelRealSense/librealsense/issues/9865]
3. [https://support.intelrealsense.com/hc/en-us/community/posts/4402953910291-Help-Error-no-module-named-pyrealsense2]
4. [https://support.intelrealsense.com/hc/en-us/community/posts/360048396854-Raspberry-Pi-ModuleNotFoundError-No-module-named-pyrealsense2]

**Actually I am not getting any .so file starting with  'pyrealsense2'.**

**I checked the folder librealsense > build > wrappers > python and other folders also.** 

_I tried with Raspberry Pi OS 64-bit "Buster" and Raspberry Pi OS 32-bit "Buster" both but getting same error._ 

Kindly suggest me where is the problem??

_Or, If you can tell what is the content of those .so file then I can create those files inside the same folder of my python script._

Thanks in advance!!
Saurabh

MartyG-RealSense, 2024-03-16:
Hi @Saurabh-Sac  Please first try replacing **import pyrealsense2 as rs** in your Python script with the instruction below to see whether it enables the module to be found.

`import pyrealsense2.pyrealsense2 as rs`

If that does not work, you could try building the pyrealsense2 wrapper from a package instead of source code using the command **pip install pyrealsense2** as pip install has had an Arm package for pyrealsense2 since SDK 2.54.1.

Saurabh-Sac, 2024-03-16:
Hello @MartyG-RealSense,

I tried with both OS: Raspberry Pi OS 64-bit "Buster" and Raspberry Pi OS 32-bit "Buster" and python 3.7

The command **import pyrealsense2.pyrealsense2 as rs** is not working and getting the same error (ModuleNotFoundError: No module named 'pyrealsense2')

And the command **pip install pyrealsense2**, I am  getting error of **"Could not find a version that satisfies the requirement pyrealsense2 (from versions: ) 
No matching distribution found for pyrealsense2"**

And when I tried **pip install pyrealsense** this gave the error as shown in the attached screen shot.

[image]

MartyG-RealSense, 2024-03-17:
If you have the folder **librealsense > build > wrappers > python** then this suggests that you were able to successfully build the pyrealsense2 wrapper from source code.

The .so files are not needed though if you instead set the **PYTHONPATH** with the command below in the terminal if you have not done so already.

`export PYTHONPATH=$PYTHONPATH:/usr/local/lib`

Saurabh-Sac, 2024-03-18:
I tried this and still it is giving same error.

**Should I go for Ubuntu OS? If YES, Kindly let me know the compatible Ubuntu and PYTHON version for Raspberry Pi 4B and Intel RealSense depth camera D435i.**

**_And please send me all the related links for installation._** 

I am using Raspberry Pi Imager v1.8.5

TheSav1101, 2024-03-18:
Hello, I am currently having the same issue with RealSense lidar camera L515 and a raspberry pi 5, I also tried:
`python3 -m pip install pyrealsense2-aarch64 --break-system-packages`
and compiling from source as suggested in :
[https://github.com/IntelRealSense/librealsense/blob/development/doc/installation_raspbian.md](url)
but no luck with that either even though I can see in the output of make:
`[ 65%] Built target pyrealsense2`

Any idea on how to fix this?

MartyG-RealSense, 2024-03-18:
@Saurabh-Sac  https://github.com/IntelRealSense/librealsense/issues/4375 has information about installing both the SDK and pyrealsense2 wrapper, although it is complicated.

An Ubuntu installation on Pi is likely to be easier than a Raspberry PI OS installation, though still potentially difficult. An Ubuntu install method that has a good history of success on Pi is the **libuvc backend** procedure at the link below.  However, the build script at this link does not include the Python wrapper flags in its CMake build instruction, so the script would have to bed edited to include them.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

https://github.com/IntelRealSense/librealsense/blob/master/scripts/libuvc_installation.sh

******

@TheSav1101  We have only had a couple of cases of RealSense users using Pi 5 at the time of writing this and both of them had difficulties so it is hard to say whether or not a build failure is due to a difference between Pi 4 and Pi 5, unfortunately. 

It is possible (though often difficult) to install librealsense successfully on a Pi (the libuvc method has the best chance) but adding the Python wrapper further increases the difficulty of achieving a successful build.

TheSav1101, 2024-03-21:
@MartyG-RealSense The thing is that the c++ library actually works, I can take all the data I need with that, but to me it seems strange that even if the python wrapper compiles with no error it is not installed correctly :)

MartyG-RealSense, 2024-03-21:
@TheSav1101  Raspberry Pi installation can often provide challenges for RealSense software unfortunately, whether it is the main SDK or one of its wrappers such as pyrealsense2 or ROS.

MartyG-RealSense, 2024-03-27:
Hi @Saurabh-Sac and @TheSav1101  Do either of you require further assistance with this case, please?  Thanks!

TheSav1101, 2024-03-27:
Hello @MartyG-RealSense thank you for the help, but I think I will try and use the ros2 wrapper (should less of a hustle in my case) :)

MartyG-RealSense, 2024-03-27:
Thanks very much @TheSav1101 for the update about your situation!

Saurabh-Sac, 2024-03-27:
Hello @MartyG-RealSense,

_Now I am able to run successfully with 64-bit "Buster"._ 

**But the processing of Intel RealSense depth camera D435i is very slow. I am using it for gesture recognition and object detection with MediaPipe and YOLOv8.** 

_How can we speed up the processing of depth camera?_

MartyG-RealSense, 2024-03-27:
Hi @Saurabh-Sac  Another RealSense user with YOLOv8 also reported slow performance at https://github.com/IntelRealSense/librealsense/issues/12136#issuecomment-1708943802 using a different type of single-board computer.

The kind of processing-intensive recognition and detection activities that are being performed are probably primarily responsible for the slowdown.  

As mentioned at https://github.com/IntelRealSense/librealsense/issues/12136#issuecomment-1709591007 you may be able to achieve some extra performance if librealsense was built from source code with support for use of multiple CPU cores enabled.  If you already have a working librealsense installation with Buster then it may not be worth the risk of installing librealsense again though in case it becomes broken.

TheSav1101, 2024-03-27:
By the way @MartyG-RealSense I think that I might have figured out the problem I had.
Do you know if there is an "upper limit" to the python version when building from source? Because I was using python3.11, but maybe I could get better results using an older version...
I am posting this in case somebody has my same issue :smile:

MartyG-RealSense, 2024-03-27:
@TheSav1101  I am not aware of an upper limit when building the pyrealsense2 wrapper from source.

erayak, 2024-03-28:
> Hello @MartyG-RealSense,
>
> _Now I am able to run successfully with 64-bit "Buster"._
>
> **But the processing of Intel RealSense depth camera D435i is very slow. I am using it for gesture recognition [...]

Hi @Saurabh-Sac 

We have the same problem. How did you proceed in solving this problem? Can you explain it detailly. 

Thanks

MartyG-RealSense, 2024-04-03:
@TheSav1101  Do you require further assistance with this case, please?  Thanks!

Hi @erayak  Were you able to achieve a solution?

TheSav1101, 2024-04-03:
Thank you @MartyG-RealSense but I resolved all my issues :)

Saurabh-Sac, 2024-04-04:
> > Hello @MartyG-RealSense,
> > _Now I am able to run successfully with 64-bit "Buster"._
> > **But the processing of Intel RealSense depth camera D435i is very slow. I am using it for gesture recognition and object detection with MediaPipe and YOLOv8.**
> > _How can we speed up the processing of depth camera?_
> 
> Hi @Saurabh-Sac
> 
> We have the same problem with raspbian we couldn't install pyrealsense2. How did you proceed in solving this problem? Can you explain it detailly.
> 
> Thanks

Hello @erayak 

Actually It was just by trial and error. It was not straight forward. 
I got one path related error and one cmake (package missing) error during pyrealsense2 build.

For installation, I have followed the following repository:
[https://github.com/datasith/Ai_Demos_RPi/wiki/Raspberry-Pi-4-and-Intel-RealSense-D435]

For Raspberry Pi OS 64-bit "Buster", I used this link:
[https://support.pishop.us/article/137-official-links-to-raspberry-pi-os-buster]

Saurabh-Sac, 2024-04-04:
Hello @MartyG-RealSense 

Can I use laptop (in place of Raspberry Pi) to process the Intel RealSense depth camera D435i data (means object position (x, y, z)) and Raspberry Pi to only control (send PWM signal for servo motors) a robotic arm to pick an object using its inverse kinematics **simultaneously** ?

MartyG-RealSense, 2024-04-04:
@Saurabh-Sac   Something like the RealSense inverse-kinematics robot arm project at https://github.com/silvery107/silvery-armlab-f22 ?

TheSav1101, 2024-04-04:
> Hi
> 
> yes...something like this. I want to pick the different objects and for object detection I am using YOLO algorithm.
> 
> Actually I want to run.... 
> 1. RealSense object detection data .py file using laptop **(not Raspberry Pi)**  [due to slow processing speed of depth camera with Raspberry Pi]
> and 
> 2. After getting that data (object position), I want to use Raspberry Pi to run the servo motors of robotic arm.
> 
> Can we do these two things simultaneously? 

I belive you can do all of this with ROS @

Saurabh-Sac, 2024-04-04:
> > Hi
> > yes...something like this. I want to pick the different objects and for object detection I am using YOLO algorithm.
> > Actually I want to run....
> > 
> > 1. RealSense object detection data .py file using laptop **(not Raspberry Pi)**  [due to slow processing speed of depth camera with Raspberry Pi]
> >    and
> > 2. After getting that data (object position), I want to use Raspberry Pi to run the servo motors of robotic arm.
> > 
> > Can we do these two things simultaneously?
> 
> I belive you can do all of this with ROS @Saurabh-Sac

Hello @TheSav1101 
But I want to directly implement on hardware setup for robotic arm (without any simulator like ROS & Gazebo)

TheSav1101, 2024-04-04:
I mean, you can use ros to implement the lan communication between the two devices using messages and have the raspberry node handle the movement with your own code without using simulations, If I understood correctly that's exactly what you need, isn't it?

Saurabh-Sac, 2024-04-05:
yes...Okay.

Is there any other interface or platform for communication? (Like MATLAB tool or other protocol)

MartyG-RealSense, 2024-04-15:
Hi @Saurabh-Sac  Have you been able to solve your issue or do you require further assistance, please?  Thanks!

Saurabh-Sac, 2024-04-16:
Hi @MartyG-RealSense I shifted to Jetson Nano now. After resolving many issues it is working fine now.

**I believe there should be a single line command to install pyrealsense2 for single-board computer.** 

**_Because the provided installation document generate issues in any kind of single-board computer whether it is Arduino, Raspberry Pi or Jetson._** 

Thanks!!

MartyG-RealSense, 2024-04-17:
The **pip install pyrealsense2** single line installation command covers most situations for x86/x64 and Arm architecture computers, though it admittedly does not work for every single-board computer (such as Raspberry Pi) and it also takes time to add support to the pip packages for the newest Python 3 versions.  For situations where pip install does not work or a recent Python version is being used, source-code install can be used.

MartyG-RealSense, 2024-04-28:
Hi @Saurabh-Sac  Do you require further assistance with this case, please?  Thanks!

Saurabh-Sac, 2024-04-28:
Hello @MartyG-RealSense We can close this issue.
Thanks everyone...!!