realsenseai/librealsense #12761: I would like to record a single frame of the bag file

mengxinglvhappy, 2024-03-18:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)? 
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   WIN11 | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  C++                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

I want to use the D415 camera to record a bag file that only contains a single frame of video. Is there any relevant C++ script available that can help me achieve this?

MartyG-RealSense, 2024-03-18:
Hi @mengxinglvhappy   There is an SDK instruction called **save_single_frameset** that saves a single frame to a bag file.  There are only Python examples of its use available though.  However, it seems as though it would be straightforward to convert to C++.  Here is Python example code:

```
frameset = rs.save_single_frameset()
frames = pipeline.wait_for_frames()
frameset.process(frames)
```

rs.save_single_frameset() will likely be `rs2.save_single_frameset() `in C++.

mengxinglvhappy, 2024-05-22:
import pyrealsense2 as rs
import numpy as np
import cv2

pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, 1280, 720, rs.format.z16, 15)
config.enable_stream(rs.stream.color, 1920, 1080, rs.format.rgb8, 15)
profile = pipeline.start(config)

filt = rs.save_single_frameset()

for x in range(100):
    pipeline.wait_for_frames()

frame = pipeline.wait_for_frames()
filt.process(frame)

Hello I have found this script that can handle individual frames. How can I save a frame as a bag file?

MartyG-RealSense, 2024-05-22:
The **save_single_frameset()** instruction that you already have in your script saves a single frame as a bag file.

closed 2024-05-24 by mengxinglvhappy