realsenseai/librealsense #12770: Compilation Error on Ubuntu 20.04 with OpenSSL 1.1.1f During Kernel Module Configuration for Librealsense

angeelalg, 2024-03-19:
Hello,

I've encountered a compilation error while setting up Librealsense on Ubuntu 20.04 LTS, specifically during the kernel module configuration step when running the patch-realsense-ubuntu-lts-hwe.sh script. My system is using OpenSSL version 1.1.1f, and I've confirmed that libssl-dev is up to date.

**Environment:**

OS: Ubuntu 20.04 LTS
OpenSSL Version: 1.1.1f
Librealsense Version: Attempting to compile from source (I have tried master, v2.54.2 and v2.53.1)
Kernel Version: 5.15.0-53-generic

**Error Message:**
```
$ ./scripts/patch-realsense-ubuntu-lts-hwe.sh
Reading package lists... Done
Building dependency tree       
Reading state information... Done
bc is already the newest version (1.07.1-2build1).
build-essential is already the newest version (12.8ubuntu1.1).
git is already the newest version (1:2.25.1-1ubuntu3.11).
linux-headers-generic is already the newest version (5.4.0.174.172).
0 upgraded, 0 newly installed, 0 to remove and 837 not upgraded.
Package required libusb-1.0-0-dev:  - found
Package required libssl-dev:  - found
Create patches workspace in ubuntu-focal folder

Package required libelf-dev:  - found
Package required elfutils:  - found
Package required bison:  - found
Package required flex:  - found
Package required dwarves:  - found
Initialized empty Git repository in /home/watchbot/dev_ws/src/watchbot2/third-party/librealsense/ubuntu-focal/.git/
Fetching Ubuntu LTS tag Ubuntu-hwe-5.15-5.15.0-53.59_20.04.1  to the local kernel sources folder
remote: Enumerating objects: 78716, done.
remote: Counting objects: 100% (78716/78716), done.
remote: Compressing objects: 100% (74469/74469), done.
remote: Total 78716 (delta 8938), reused 27649 (delta 3406)
Receiving objects: 100% (78716/78716), 212.97 MiB | 3.61 MiB/s, done.
Resolving deltas: 100% (8938/8938), done.
From https://git.launchpad.net/~ubuntu-kernel/ubuntu/+source/linux/+git/focal
 * [new tag]             Ubuntu-hwe-5.15-5.15.0-53.59_20.04.1 -> Ubuntu-hwe-5.15-5.15.0-53.59_20.04.1
Switching to LTS tag Ubuntu-hwe-5.15-5.15.0-53.59_20.04.1
Updating files: 100% (74205/74205), done.
Note: switching to 'Ubuntu-hwe-5.15-5.15.0-53.59_20.04.1'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at 0b869e67a UBUNTU: Ubuntu-hwe-5.15-5.15.0-53.59~20.04.1
Applying patches for focal-hwe-5.15 line
Applying realsense-uvc patch
patching file drivers/media/usb/uvc/uvc_driver.c
patching file drivers/media/usb/uvc/uvcvideo.h
patching file drivers/media/v4l2-core/v4l2-ioctl.c
patching file include/uapi/linux/videodev2.h
Applying realsense-metadata patch
patching file drivers/media/usb/uvc/uvc_driver.c
Hunk #1 succeeded at 3221 (offset 57 lines).
patching file drivers/media/usb/uvc/uvcvideo.h
Hunk #1 succeeded at 221 (offset 34 lines).
Applying realsense-hid patch
patching file drivers/iio/accel/hid-sensor-accel-3d.c
patching file drivers/iio/gyro/hid-sensor-gyro-3d.c
Applying realsense-powerlinefrequency-fix patch
patching file drivers/media/usb/uvc/uvc_ctrl.c
Hunk #1 succeeded at 361 with fuzz 1 (offset 4 lines).
Increase UVC_URBs in uvcvideo
patching file drivers/media/usb/uvc/uvcvideo.h
Hunk #1 succeeded at 217 (offset 44 lines).
Prepare kernel modules configuration
/home/watchbot/dev_ws/src/watchbot2/third-party/librealsense/ubuntu-focal
2713$
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/kconfig/conf.o
  HOSTCC  scripts/kconfig/confdata.o
  HOSTCC  scripts/kconfig/expr.o
  LEX     scripts/kconfig/lexer.lex.c
  YACC    scripts/kconfig/parser.tab.[ch]
  HOSTCC  scripts/kconfig/lexer.lex.o
  HOSTCC  scripts/kconfig/menu.o
  HOSTCC  scripts/kconfig/parser.tab.o
  HOSTCC  scripts/kconfig/preprocess.o
  HOSTCC  scripts/kconfig/symbol.o
  HOSTCC  scripts/kconfig/util.o
  HOSTLD  scripts/kconfig/conf
#
# configuration written to .config
#
  SYSHDR  arch/x86/include/generated/uapi/asm/unistd_32.h
  SYSHDR  arch/x86/include/generated/uapi/asm/unistd_64.h
  SYSHDR  arch/x86/include/generated/uapi/asm/unistd_x32.h
  SYSTBL  arch/x86/include/generated/asm/syscalls_32.h
  SYSHDR  arch/x86/include/generated/asm/unistd_32_ia32.h
  SYSHDR  arch/x86/include/generated/asm/unistd_64_x32.h
  SYSTBL  arch/x86/include/generated/asm/syscalls_64.h
  SYSTBL  arch/x86/include/generated/asm/syscalls_x32.h
  HYPERCALLS arch/x86/include/generated/asm/xen-hypercalls.h
  HOSTCC  arch/x86/tools/relocs_32.o
  HOSTCC  arch/x86/tools/relocs_64.o
  HOSTCC  arch/x86/tools/relocs_common.o
  HOSTLD  arch/x86/tools/relocs
  HOSTCC  scripts/genksyms/genksyms.o
  YACC    scripts/genksyms/parse.tab.[ch]
  HOSTCC  scripts/genksyms/parse.tab.o
  LEX     scripts/genksyms/lex.lex.c
  HOSTCC  scripts/genksyms/lex.lex.o
  HOSTLD  scripts/genksyms/genksyms
  HOSTCC  scripts/selinux/genheaders/genheaders
  HOSTCC  scripts/selinux/mdp/mdp
  HOSTCC  scripts/bin2c
  HOSTCC  scripts/kallsyms
  HOSTCC  scripts/sorttable
  HOSTCC  scripts/asn1_compiler
  HOSTCC  scripts/sign-file

scripts/sign-file.c: In function ‘read_private_key’:
scripts/sign-file.c:144:7: warning: implicit declaration of function ‘ENGINE_by_id’ [-Wimplicit-function-declaration]
  144 |   e = ENGINE_by_id("pkcs11");
      |       ^~~~~~~~~~~~
scripts/sign-file.c:144:5: warning: assignment to ‘ENGINE *’ {aka ‘struct engine_st *’} from ‘int’ makes pointer from integer without a cast [-Wint-conversion]
  144 |   e = ENGINE_by_id("pkcs11");
      |     ^
scripts/sign-file.c:146:7: warning: implicit declaration of function ‘ENGINE_init’; did you mean ‘ENGINE_new’? [-Wimplicit-function-declaration]
  146 |   if (ENGINE_init(e))
      |       ^~~~~~~~~~~
      |       ENGINE_new
scripts/sign-file.c:151:9: warning: implicit declaration of function ‘ENGINE_ctrl_cmd_string’ [-Wimplicit-function-declaration]
  151 |    ERR(!ENGINE_ctrl_cmd_string(e, "PIN", key_pass, 0),
      |         ^~~~~~~~~~~~~~~~~~~~~~
scripts/sign-file.c:107:18: note: in definition of macro ‘ERR’
  107 |   bool __cond = (cond);   \
      |                  ^~~~
scripts/sign-file.c:153:17: warning: implicit declaration of function ‘ENGINE_load_private_key’ [-Wimplicit-function-declaration]
  153 |   private_key = ENGINE_load_private_key(e, private_key_name,
      |                 ^~~~~~~~~~~~~~~~~~~~~~~
scripts/sign-file.c:153:15: warning: assignment to ‘EVP_PKEY *’ {aka ‘struct evp_pkey_st *’} from ‘int’ makes pointer from integer without a cast [-Wint-conversion]
  153 |   private_key = ENGINE_load_private_key(e, private_key_name,
      |               ^
/usr/bin/ld: /tmp/cc5tShlf.o: in function `main':
sign-file.c:(.text.startup+0x55): undefined reference to `OpenSSL_add_all_algorithms'
/usr/bin/ld: sign-file.c:(.text.startup+0x5a): undefined reference to `ERR_load_crypto_strings'
/usr/bin/ld: sign-file.c:(.text.startup+0x1df): undefined reference to `BIO_reset'
/usr/bin/ld: sign-file.c:(.text.startup+0x50a): undefined reference to `BIO_should_retry'
/usr/bin/ld: sign-file.c:(.text.startup+0x70d): undefined reference to `BIO_reset'
/usr/bin/ld: sign-file.c:(.text.startup+0x77b): undefined reference to `OpenSSL_add_all_digests'
collect2: error: ld returned 1 exit status
make[2]: *** [scripts/Makefile.host:95: scripts/sign-file] Error 1
make[1]: *** [Makefile:1237: scripts] Error 2
make: *** [Makefile:350: __build_one_by_one] Error 2
```

**Steps to Reproduce:** Follow the [installation guide](https://github.com/IntelRealSense/librealsense/blob/v2.54.2/doc/installation.md) 
Clone the Librealsense repository.
Navigate to the /librealsense directory.
Run ./scripts/patch-realsense-ubuntu-lts-hwe.sh.
I have already tried updating my system and the libssl-dev package, verifying the path of the OpenSSL libraries, and checking the compilation environment for potential issues, but the error persists.

**Expected Behavior:**
The script should complete without errors, allowing me to proceed with the Librealsense installation and kernel module configuration.

**Actual Behavior:**
Compilation fails with the above-mentioned error related to OpenSSL functions not being found or properly linked.

Could you please provide any guidance on how to resolve this issue or suggest any additional steps I might take to troubleshoot further? Any help or pointers would be greatly appreciated.

Thank you for your time and assistance.

MartyG-RealSense, 2024-03-19:
Hi @angeelalg  If the patch script is failing because openssl is a dependency of it then you can avoid using the patch script by installing the librealsense SDK by a method that **bypasses** the kernel so that you do not need to apply a patch.  A kernel-bypassed build of the SDK is not dependent on kernel versions or Linux versions.  

This can be done with the **libuvc backend** install method using a build script at the link below.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

angeelalg, 2024-03-19:
Again, with that method, more "undefined reference" errors appear:
```
[ 86%] Building CXX object tools/depth-quality/CMakeFiles/rs-depth-quality.dir/__/__/common/option-model.cpp.o
[ 87%] Building CXX object tools/depth-quality/CMakeFiles/rs-depth-quality.dir/__/__/common/device-model.cpp.o
[ 88%] Building CXX object tools/realsense-viewer/CMakeFiles/realsense-viewer.dir/__/__/common/subdevice-model.cpp.o
[ 88%] Building CXX object tools/depth-quality/CMakeFiles/rs-depth-quality.dir/__/__/common/subdevice-model.cpp.o
[ 88%] Building CXX object tools/realsense-viewer/CMakeFiles/realsense-viewer.dir/__/__/common/processing-block-model.cpp.o
[ 88%] Building CXX object tools/realsense-viewer/CMakeFiles/realsense-viewer.dir/__/__/common/stream-model.cpp.o
[ 88%] Building CXX object tools/depth-quality/CMakeFiles/rs-depth-quality.dir/__/__/common/processing-block-model.cpp.o
[ 88%] Building CXX object tools/depth-quality/CMakeFiles/rs-depth-quality.dir/__/__/common/stream-model.cpp.o
[ 88%] Building CXX object tools/realsense-viewer/CMakeFiles/realsense-viewer.dir/__/__/common/post-processing-filters.cpp.o
[ 89%] Building CXX object tools/realsense-viewer/CMakeFiles/realsense-viewer.dir/__/__/common/sw-update/http-downloader.cpp.o
[ 89%] Building CXX object tools/realsense-viewer/CMakeFiles/realsense-viewer.dir/__/__/common/sw-update/dev-updates-profile.cpp.o
[ 90%] Building CXX object tools/depth-quality/CMakeFiles/rs-depth-quality.dir/__/__/common/post-processing-filters.cpp.o
[ 90%] Building CXX object tools/realsense-viewer/CMakeFiles/realsense-viewer.dir/__/__/common/sw-update/versions-db-manager.cpp.o
[ 90%] Building CXX object tools/depth-quality/CMakeFiles/rs-depth-quality.dir/__/__/common/sw-update/http-downloader.cpp.o
[ 91%] Building CXX object tools/realsense-viewer/CMakeFiles/realsense-viewer.dir/__/__/common/reflectivity/reflectivity.cpp.o
[ 91%] Building CXX object tools/depth-quality/CMakeFiles/rs-depth-quality.dir/__/__/common/sw-update/dev-updates-profile.cpp.o
[ 91%] Building CXX object tools/realsense-viewer/CMakeFiles/realsense-viewer.dir/__/__/common/utilities/imgui/wrap.cpp.o
[ 91%] Building CXX object tools/realsense-viewer/CMakeFiles/realsense-viewer.dir/realsense-viewer.cpp.o
[ 92%] Building CXX object tools/depth-quality/CMakeFiles/rs-depth-quality.dir/__/__/common/sw-update/versions-db-manager.cpp.o
[ 92%] Building CXX object tools/depth-quality/CMakeFiles/rs-depth-quality.dir/__/__/common/reflectivity/reflectivity.cpp.o
[ 93%] Building CXX object tools/realsense-viewer/CMakeFiles/realsense-viewer.dir/__/__/third-party/imgui/imgui.cpp.o
[ 93%] Building CXX object tools/depth-quality/CMakeFiles/rs-depth-quality.dir/__/__/common/utilities/imgui/wrap.cpp.o
[ 93%] Building CXX object tools/depth-quality/CMakeFiles/rs-depth-quality.dir/rs-depth-quality.cpp.o
[ 93%] Building CXX object tools/realsense-viewer/CMakeFiles/realsense-viewer.dir/__/__/third-party/imgui/imgui_draw.cpp.o
[ 94%] Building CXX object tools/depth-quality/CMakeFiles/rs-depth-quality.dir/depth-quality-model.cpp.o
[ 94%] Building CXX object tools/realsense-viewer/CMakeFiles/realsense-viewer.dir/__/__/third-party/imgui/imgui_impl_glfw.cpp.o
[ 94%] Building C object tools/realsense-viewer/CMakeFiles/realsense-viewer.dir/__/__/third-party/glad/glad.c.o
[ 95%] Building C object tools/realsense-viewer/CMakeFiles/realsense-viewer.dir/__/__/third-party/tinyfiledialogs/tinyfiledialogs.c.o
[ 95%] Linking CXX executable ../../release/realsense-viewer
/usr/bin/ld: ../../libcurl/libcurl_install/lib/libcurl.a(openssl.c.o): in function `ossl_sha256sum':
openssl.c:(.text+0xa1): undefined reference to `EVP_MD_CTX_create'
/usr/bin/ld: openssl.c:(.text+0xdf): undefined reference to `EVP_MD_CTX_destroy'
/usr/bin/ld: ../../libcurl/libcurl_install/lib/libcurl.a(openssl.c.o): in function `ossl_init':
openssl.c:(.text+0xe4c): undefined reference to `SSL_get_ex_new_index'
/usr/bin/ld: openssl.c:(.text+0xe74): undefined reference to `SSL_get_ex_new_index'
/usr/bin/ld: openssl.c:(.text+0xe9c): undefined reference to `SSL_get_ex_new_index'
/usr/bin/ld: ../../libcurl/libcurl_install/lib/libcurl.a(openssl.c.o): in function `ossl_new_session_cb':
openssl.c:(.text+0xfa4): undefined reference to `SSL_get_ex_new_index'
/usr/bin/ld: openssl.c:(.text+0xfcc): undefined reference to `SSL_get_ex_new_index'
/usr/bin/ld: ../../libcurl/libcurl_install/lib/libcurl.a(openssl.c.o):openssl.c:(.text+0xfec): more undefined references to `SSL_get_ex_new_index' follow
/usr/bin/ld: ../../libcurl/libcurl_install/lib/libcurl.a(openssl.c.o): in function `servercert':
openssl.c:(.text+0x1fc8): undefined reference to `BIO_get_mem_ptr'
/usr/bin/ld: openssl.c:(.text+0x204f): undefined reference to `BIO_get_mem_data'
/usr/bin/ld: openssl.c:(.text+0x2073): undefined reference to `BIO_reset'
/usr/bin/ld: openssl.c:(.text+0x2092): undefined reference to `BIO_get_mem_data'
/usr/bin/ld: openssl.c:(.text+0x20b6): undefined reference to `BIO_reset'
/usr/bin/ld: openssl.c:(.text+0x218d): undefined reference to `sk_num'
/usr/bin/ld: openssl.c:(.text+0x21ec): undefined reference to `sk_value'
/usr/bin/ld: openssl.c:(.text+0x2288): undefined reference to `BIO_get_mem_ptr'
/usr/bin/ld: openssl.c:(.text+0x256b): undefined reference to `sk_num'
/usr/bin/ld: openssl.c:(.text+0x25f4): undefined reference to `sk_value'
/usr/bin/ld: openssl.c:(.text+0x262d): undefined reference to `BIO_get_mem_data'
/usr/bin/ld: openssl.c:(.text+0x264e): undefined reference to `BIO_reset'
[... cut ...]
collect2: error: ld returned 1 exit status
make[2]: *** [tools/depth-quality/CMakeFiles/rs-depth-quality.dir/build.make:605: release/rs-depth-quality] Error 1
make[1]: *** [CMakeFiles/Makefile2:1826: tools/depth-quality/CMakeFiles/rs-depth-quality.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```

It's strange, in the past I had the SDK installed on this machine by [Using pre-build packages](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md). But now I need an installation from source.

MartyG-RealSense, 2024-03-19:
Did you remove the previous package-based librealsense installation from the computer before performing a source code build, please?  Having both a package and source code build of the SDK on the same computer has the potential to cause conflicts between them.

You can uninstall a package build with the instruction below.

`dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge`

To uninstall a source code build, use the instruction below in your source code folder's **build** directory to uninstall the source code SDK build and clean up CMake.

`sudo make uninstall && make clean`

angeelalg, 2024-03-19:
Hi @MartyG-RealSense,

Yes, prior to my attempt to install source I completely removed the previous installation, I have checked it again. 

I also tried to disable secure boot in the bios, as I remember that during my package-based installation I had to confirm a MOK pass after reboot. No success. 

:(

MartyG-RealSense, 2024-03-19:
Do you have the option to wipe the computer and install everything from the beginning again, including Ubuntu?  Doing so can sometimes suddenly cause unresolvable problems to disappear because hidden 'tangles' on the computer resulting from past installs are cleared away by the wipe.

angeelalg, 2024-03-20:
I just tried to install the source installation on another new device completely clean and the problem described in the original post persists.

MartyG-RealSense, 2024-03-20:
In https://github.com/IntelRealSense/librealsense/issues/12770#issuecomment-2007040829 you said that you had succeeded with package builds in the past but now needed a source code build, and it is the source code builds that are failing.  Is this because you need to install something not included in the packages, such as the Python wrapper?

angeelalg, 2024-03-21:
I need the SDK installation to be stable and time-invariant in order to deploy on a very reliable system. I need the source installation.

angeelalg, 2024-03-25:
I need a **librealsense installation from source**. If someone can help me with the described _openssl_ problem I will be very grateful.

Thank you very much.

MartyG-RealSense, 2024-04-03:
Hi @angeelalg  Were you able to find a solution or do you require further assistance, please?  Thanks!

closed 2024-04-10 by MartyG-RealSense