realsenseai/librealsense #12771: Hi! Is there a way to know the size of a pixel (dimensions) in real life

laracb01, 2024-03-19:
-------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435, L515 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Windows 11  | 
| Kernel Version (Linux Only)    |                                         | 
| Platform                                 | VisualStudio Code  |
| SDK Version                            |  2.0                          | 
| Language                            |  VisualStudio Code Python                         | 
| Segment			|                   | 

### Issue Description

I'm utilizing both my IntelReal Sense D435 and IntelReal Sense L515 for object avoidance. I'm curious if there's a method to determine the real-life dimensions represented by each pixel, essentially finding out their actual size.

MartyG-RealSense, 2024-03-19:
Hi @laracb01  The closest to what you have in mind may be to measure the distance between two points in the real-world scene (instead of measuring from the camera to the point).  The link below has a Python script for doing so.

https://github.com/soarwing52/RealsensePython/blob/master/separate%20functions/measure_new.py

laracb01, 2024-03-19:
I already have a script that calculates the distance from the camera to the pixels I select and, on the IntelReal Sense app, you are able to extract the distance between two point. However, I'm looking to establish a direct correspondence between the pixel coordinates I obtain when selecting in the image and their corresponding 'area' in the real world.

MartyG-RealSense, 2024-03-19:
You can obtain pixel width X and height Y **intrinsic** values for the image which change based on the resolution that is selected (e.g 1280x720).   I am not aware though of an equivalent representation for the amount of space in the real-world scene that a pixel represents.  Usually, real-world elements are measured from point-to-point instead of per-pixel.

closed 2024-04-07 by MartyG-RealSense