realsenseai/librealsense #12781: Segmentation Fault on RegisteredLoggers

ikilinc103, 2024-03-21:
Camera Model : DR35i
Operating System : Ubuntu 20.04
Kernel Version : 5.15.0-101-generic
Platform : PC
SDK Version : librealsense-2.54.2
Hardware : NUC12WSKI7 Intel
Language : C++
Segment : AR

Although we use the library effectively, we rarely receive the following error during camera is operational.

GDB full stack trace of crash:

Program terminated with signal SIGSEGV, Segmentation fault.
#0  __GI___pthread_mutex_lock (mutex=0x8) at ../nptl/pthread_mutex_lock.c:67
        type = <optimized out>
        __PRETTY_FUNCTION__ = "__pthread_mutex_lock"
        id = <optimized out>
#1  0x00007faf4af8729d in el::base::RegisteredLoggers::get(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool) ()
    at ../libs/librealsense2.so.2.54
#2  0x00007faf4af8773d in el::base::Writer::initializeLogger(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool, bool) () at ../libs/librealsense2.so.2.54
#3  0x00007faf4af87b05 in el::base::Writer::construct(int, char const*, ...) ()
    at ../libs/librealsense2.so.2.54
#4  0x00007faf4af2b5ab in librealsense::hid_sensor::start(std::shared_ptr<rs2_frame_callback>)::{lambda(librealsense::platform::sensor_data const&)#1}::operator()(librealsense::platform::sensor_data const&) ()
    at ../libs/librealsense2.so.2.54
#5  0x00007faf4ae8bb14 in librealsense::platform::iio_hid_sensor::start_capture(std::function<void (librealsense::platform::sensor_data const&)>)::{lambda()#1}::operator()() const [clone .isra.0] () at ../libs/librealsense2.so.2.54
#6  0x00007faf469d3de4 in  () at /lib/x86_64-linux-gnu/libstdc++.so.6
#7  0x00007faf46776609 in start_thread (arg=<optimized out>)
    at pthread_create.c:477
        ret = <optimized out>
--Type <RET> for more, q to quit, c to continue without paging--
        pd = <optimized out>
        unwind_buf = 
              {cancel_jmp_buf = {{jmp_buf = {140388958385920, 2618322333070841624, 140729070220030, 140729070220031, 140729070220032, 140388958345344, -2662870144842672360, -2663691554726947048}, mask_was_saved = 0}}, priv = {pad = {0x0, 0x0, 0x0, 0x0}, data = {prev = 0x0, cleanup = 0x0, canceltype = 0}}}
        not_first_call = 0
#8  0x00007faf4669b133 in clone ()
    at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95

MartyG-RealSense, 2024-03-21:
Hi @ikilinc103  Thanks very much for your issues here and https://github.com/IntelRealSense/librealsense/issues/12782 and https://github.com/IntelRealSense/librealsense/issues/12783

As all three cases likely have a common cause, I will handle it on this one page.

**Program terminated with signal SIGSEGV, Segmentation fault** errors are very difficult to solve, unfortunately.  Considering the rarity at which your particular error occurs and the contents of the three cases that you kindly reported, I would like to first test whether the problem is triggered by **logging**.

Which method did you use to install the librealsense SDK on your computer, please?

If you built it from source code with CMake, would it be possible to build librealsense again but this time set the build flag **-DBUILD_EASYLOGGINGPP=FALSE** to disable the easylogging system.  This will help to determine whether or not easylogging is the cause of the problem.

ikilinc103, 2024-03-22:
Hi MartyG,
Thank you for your interest and quick response. I built the library again by disabling logging as you recommended. I think that removing logging will resolve the issue 12781. But Issue 12782 and 1783 may be not related with the logging library since call stack is different. After performing detailed tests, I will share here whether the error has been resolved or not.

MartyG-RealSense, 2024-03-22:
Thanks very much for the information.  I look forward to your next report.  Good luck!

closed 2024-04-26 by MartyG-RealSense