realsenseai/librealsense #12784: Is there a way to obtain the position change of the d435i camera

zxc0210, 2024-03-21:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i } | 
| Operating System & Version |   Win 10  | 
| Platform                                 | PC   |
| SDK Version                            |  2                          | 
| Language                            |  {C++}                          | 

### Issue Description
I am using a d435i camera，and I want  to do point cloud register. However, when I moved the camera, I found that the point cloud I obtained would overlap in cloudcompare.  I want to know if there is a way to obtain the position change of the camera，so that I can restore the position of the point cloud in the world coordinate system.

zxc0210, 2024-03-21:
Thanks a lot for your swify reply, I 'll try **rs-kinfu** later. 
And I have another question. I tried to integrate the accelerometer data of **imu** to obtain the movement, but I got wrong result. I found when the camera is stationary, the value of the accelerometer is not zero. Or do you know any method to eliminate the impact of gravity?

MartyG-RealSense, 2024-03-21:
This is because the IMU component of RealSense cameras does not have internal hardware calibration.  The non-zero Y-accel value when the camera is stationary can be corrected by using an instruction called **RS2_OPTION_ENABLE_MOTION_CORRECTION** that can perform correction on inaccurate raw IMU data to produce better "final" IMU values.  An example of implementing this instruction in C++ is at https://github.com/IntelRealSense/librealsense/issues/5176

That script sets RS2_OPTION_ENABLE_MOTION_CORRECTION to '0' (correction disabled), so change '0' to '1' in the script to enable correction.

closed 2024-04-07 by MartyG-RealSense