realsenseai/librealsense #12786: rs-examples

kreddyw, 2024-03-21:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 1.5.8.1 | 
| Operating System & Version |  Windows 10  | 
| Platform                                 | PC |
| SDK Version                            | Latest | 
| Language                            |  C++                          | 

### Issue Description
Hello, I have been trying to run the function dist_3d() from rs-measure in a separate project, but the issue I get is that it returns 0. It otherwise gives me the distance value perfectly when I run rs-measure.cpp. What exact config do I have to set to get it running?

MartyG-RealSense, 2024-03-21:
Hi @kreddyw  Becuase of the way that the official RealSense SDK examples are built, copying the code directly into your own project is not likely to work.  You should ensure that your project is equipped with the necessary linkages to the librealsense library.

If you are using Visual Studio to code in C++ then the link below provides a link to a tutorial guide for easily setting up these library linkages in your project using three **.props** property sheet files provided by the RealSense SDK when you install the full SDK on Windows with the **Intel.RealSense.SDK-WIN10** installer file. 

https://support.intelrealsense.com/hc/en-us/community/posts/4412175411859/comments/4412182682387

kreddyw, 2024-03-21:
The second project is basically running on rs-dnn on the same vs solution. Putting the function in the same project solution does that.

MartyG-RealSense, 2024-03-22:
As you mention rs-dnn, do you mean that you are combining the code of rs-measure and rs-dnn into a single program?  Apologies if I have misunderstood.

kreddyw, 2024-03-23:
I solved the issue by adding in the line ``rs2::config cfg;
    cfg.enable_stream(RS2_STREAM_DEPTH);``

closed 2024-03-23 by kreddyw

MartyG-RealSense, 2024-03-23:
Thanks very much for sharing your solution.  I'm pleased to hear that you succeeded. Thanks for the update!