realsenseai/librealsense #12790: issue with the D435IF depth image

Nimaro76, 2024-03-24:
Hello, I have a D435IF cmera that is giving us a distorted depth image. we tried on-chip calibration and gold reset and tare calibration but this distortion has not been fixed to reach the second image, could you please let me know hat could be the issue ? 
[image]
[image]
best regards

MartyG-RealSense, 2024-03-24:
Hi @Nimaro76 

If a gold reset did not correct the depth image then you could try another method of resetting the camera to its factory-new default calibration, using the RealSense Viewer with instructions at https://github.com/IntelRealSense/librealsense/issues/10182#issuecomment-1019854487

If that does not work then you can try resetting the Viewer to its default settings instead of the camera with instructions at the link below.

https://support.intelrealsense.com/hc/en-us/community/posts/360052013914/comments/1500000416522

I also note that you are using camera firmware driver 5.13,0.55 in your camera.  This firmware is designed for Viewer / SDK versions 2.50.0 and 2.51.1, and using it with 2.54.2 could cause errors.  For 2.54.2 it is recommended that the firmware is updated to **5.15.1.0**, as suggested by the 'Firmware Update Recommended' pop-up box in the top corner of your Viewer.

MartyG-RealSense, 2024-04-03:
Hi @Nimaro76  Do you require further assistance with this case, please?  Thanks!

Nimaro76, 2024-04-05:
not for now, thank you.

best, Nima

MartyG-RealSense, 2024-04-05:
You are very welcome.  As you do not require further assistance, I will close this case.  Thanks again!

closed 2024-04-05 by MartyG-RealSense