realsenseai/librealsense #12795: IMU Not Working

suptimq, 2024-03-25:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 5.15.1 | 
| Operating System & Version |   Win10  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.54.2                          | 
| Language                            | python                          | 
| Segment			|  Robot                        | 

### Issue Description

Hey I can run this example just fine https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/opencv_viewer_example.py

However, when I added two lines of code to enable the IMU the program starts to complain.

```python
config.enable_stream(rs.stream.accel, rs.format.motion_xyz32f)
config.enable_stream(rs.stream.gyro, rs.format.motion_xyz32f)
```

```python
Traceback (most recent call last):
  File ".\examples\opencv_viewer_example.py", line 42, in <module>
    pipeline.start(config)
RuntimeError: Couldn't resolve requests
```

suptimq, 2024-03-25:
Hey it is working! Thanks for the quick response.

I have been using the same `fps` for both `accel` and `gyro` and never got them worked. I wondered why a different fps is required.

MartyG-RealSense, 2024-03-25:
RealSense cameras equipped with an IMU and manufactured after mid 2022 support a pairing of 100 Accel / 200 Gyro or 200 Accel / 400 Gyro.

suptimq, 2024-03-25:
I see. That is very helpful! Where can I find that information?

MartyG-RealSense, 2024-03-25:
It is on page 103 of the data sheet document for the 400 Series cameras, which can be downloaded as a PDF at the link below.

https://dev.intelrealsense.com/docs/intel-realsense-d400-series-product-family-datasheet

suptimq, 2024-03-25:
Thank you! I will close this issue for now.

closed 2024-03-25 by suptimq