realsenseai/librealsense #12796: Realsense D450 without stainless steel enclosure

Hrithik1049, 2024-03-26:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / Qualcomm/etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
We want a Realsense Camera without a stainless steel enclosure to reduce the weight of the Camera. Can this be done for a large quantity? And will it affect the performance? I've included a pdf for you to look over. 
[D455 Depth Camera Requirement.pdf](https://github.com/IntelRealSense/librealsense/files/14753036/D455.Depth.Camera.Requirement.pdf)

MartyG-RealSense, 2024-03-26:
Hi @Hrithik1049  I will consult with my Intel RealSense colleagues about your questions.

MartyG-RealSense, 2024-03-28:
My colleagues informed me that the metal is not an enclosure but is instead a stiffener, which is key to maintaining calibration and depth performance.  

The D450 module would therefore not be supplied without this stiffener.

MartyG-RealSense, 2024-04-03:
Hi @Hrithik1049  Bearing in mind the advice in the comment above, do you require further assistance with this case please?  Thanks!

Hrithik1049, 2024-04-04:
Hi @MartyG-RealSense I wanted to ask whether the material on the casing is changeable to another lighter material.

MartyG-RealSense, 2024-04-04:
Are you referring to the silver plastic outer casing with black front glass of a D455 camera, please?  If so, is it the black glass material that you want to change?

Hrithik1049, 2024-04-04:
No Im referring to Metal casing (stiffner) of D450 camera which you earlier mentioned. I wanted to know whether it possible to change the material of this stiffner.

MartyG-RealSense, 2024-04-04:
I will consult again with my colleagues.  Thanks very much for your patience.

MartyG-RealSense, 2024-04-04:
Can you send an email to me at the address below please so that I have your email address.  Then I can provide it to one of my colleagues who can email you about your material query.

lesreaper, 2024-04-05:
We're looking at these also. Are [these modules](https://store.intelrealsense.com/buy-intel-realsense-depth-module-d450-and-d4-board-bundle-d.html) what you're referring to @Hrithik1049 ?

MartyG-RealSense, 2024-04-05:
Hi @lesreaper  The D450 camera module and Vision Processor D4 V3 are the two circuit boards contained in a D455 camera.   Purchasing the multipack of D450 and D4 V3 that you quoted helps to ensure that you have boards that will correctly match with each other.

You would additionally need an **interposer** to connect the two boards together as these are not supplied in the d450 / D4 V3 multipack.  The interposer can also be found in the official RealSense Store site, individually or in a multipack, by scrolling down the front page to its **Accessories** section.

https://store.intelrealsense.com/

Hrithik1049, 2024-04-05:
@lesreaper No we already have those.

MartyG-RealSense, 2024-04-19:
Hi @Hrithik1049  As you were put in contact by email with my Intel RealSense colleagues, can this GitHub support issue be closed or do you have further questions, please?  Thanks!

closed 2024-04-28 by MartyG-RealSense