realsenseai/librealsense #12803: RuntimeError: Frame didn't arrive within 5000

CBTezcan, 2024-03-27:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |   D4150   | 
| Firmware Version                   |   librealsense version: 2.54.1 | 
| Operating System & Version |   Linux Ubuntu 20.04  | 
| Kernel Version (Linux Only)    |   5.15.0-101-generic
                        | 
| Platform                                 |PC  |
| SDK Version                            | 2                          | 
| Language                            |  Python                         | 

////////////////////////////////////////error//////////////////////////////////////////
REALSENSE ERROR
Traceback (most recent call last):
  File "/home/tezcan/workspaces/realsense/librealsense_v2/realsense_task/test_realsense_v1.py", line 28, in <module>
    frames = pipe.wait_for_frames()
RuntimeError: Frame didn't arrive within 5000
///////////////////////////////////////error/////////////////////////////////////////////

//////////////////////////////////////code//////////////////////////////////////////////
import pyrealsense2 as rs   
import numpy as np
import cv2 as cv

pipe = rs.pipeline()
cfg = rs.config()
ctx = rs.context()

cfg.enable_stream(rs.stream.color, 640,480, rs.format.bgr8, 15)
cfg.enable_stream(rs.stream.depth, 640,480, rs.format.z16, 15)

pipe.start(cfg)

while True:

    frames = pipe.wait_for_frames()
    if frames:

        depth_frame = frames.get_depth_frame()
        color_frame = frames.get_color_frame()

        depth_image = np.asanyarray(depth_frame.get_data())
        color_image = np.asanyarray(color_frame.get_data())

        cv.imshow("rgb", color_image)
        cv.imshow("depth", depth_image)

        if cv.waitKey(1) == ord("q"):
            break
        
pipe.stop()
/////////////////////////////////////////code//////////////////////////////////////////////

MartyG-RealSense, 2024-03-28:
Hi @CBTezcan  Which camera firmware driver version is installed in your camera, please?  The recommended firmware version to use with SDK 2.54.1 is **5.15.0.2**

If you do not know your current firmware version then you can check it using the **rs-enumerate-devices** tool with the command below.

`rs-enumerate-devices -s`

CBTezcan, 2024-03-28:
Hi @MartyG-RealSense, used 5.15.0.1 firmware verison

MartyG-RealSense, 2024-03-28:
I note that you are using kernel **5.15.0-101-generic**.  The kernel version and the firmware driver version are different things, in case you are thinking of the kernel version as being the firmware version.

CBTezcan, 2024-03-29:
[image]

@MartyG-RealSense

MartyG-RealSense, 2024-03-29:
Thanks very much for the image.  Your firmware driver version is 5.15.1.0.  The recommended firmware for SDK version 2.54.1 is 5.15.0.2.  I am not convinced though that it would make enough of a difference to cause the **RuntimeError: Frame didn't arrive within 5000** error.

Does the error still occur if you run the test script at the link below, please?

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python#streaming-using-rspipeline

CBTezcan, 2024-03-30:
[image]
When I run that example code or update the firmware version, I still get the same error . Do you think the error, could be hardware or connectivity related, or related to software? 
@MartyG-RealSense

MartyG-RealSense, 2024-03-30:
The **Frame didn't arrive within 5000** indicates that new frames didn't arrive from the camera within a 5 second period.  This could suggest a problem related to USB communications.  

Are you using the official 1 meter USB cable supplied with the camera or another cable of your own choice, please?

CBTezcan, 2024-04-02:
I am using non official cable but does the length of the cable matter?

MartyG-RealSense, 2024-04-02:
Yes, the length of the USB cable is important.  RealSense cameras can transmit a high volume of data, so a high quality cable should be used that is designed for data transfer and is not an inexpensive cable intended only for device charging.  Using a lower quality cable can cause performance reduction and camera disconnections.

If the cable is longer than 5 meters then it is also recommendable to use one containing **active repeater** components that boost the signal over distance.

A company called **Newnex** supply RealSense-validated high quality USB cables, including ones with active repeaters.

https://newnex.com/realsense-3d-camera-connectivity.php

https://www.youtube.com/watch?v=GLQgR1jT04M

closed 2024-04-28 by MartyG-RealSense