realsenseai/librealsense #12806: Where is the camera origin of the D405 and its coordinate system, preferably with a picture

ybefe, 2024-03-28:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 
| Firmware Version                   | | 
| Operating System & Version |   Win (10)   | 
| Kernel Version (Linux Only)    | | 
| Platform                                 |  |
| SDK Version                                    | 
| Language                            | python                         | 
| Segment			

### Issue Description
<Describe your issue / question / feature request / etc..>

MartyG-RealSense, 2024-03-28:
Hi @ybefe  The origin of depth is the center-line of the left infrared sensor.  When viewing the camera from the front, this is the sensor on the right side of the camera.

MartyG-RealSense, 2024-04-15:
Case closed due to no further comments received.

closed 2024-04-15 by MartyG-RealSense

dut-zjh, 2025-03-11:
@MartyG-RealSense The camera's coordinates are from my red circle, right?

dut-zjh, 2025-03-11:
[image]

dut-zjh, 2025-03-11:
self-recording

[image]

MartyG-RealSense, 2025-03-11:
Hi @dut-zjh  The origin of depth is the left-side infrared sensor.  When the tripod screw-hole on the camera is facing downwards and you are looking at the camera from the front (like in your circled image), the left-side sensor is the one on the **right** side of the camera

The placement of the USB connector on your image tells me that the right sensor is the one that you have circled in red.