realsenseai/librealsense #12807: Jetson 6.0DP and "No Device Detected" in Python only

lesreaper, 2024-03-29:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { d455 } | 
| Firmware Version                   | (2.54.2) | 
| Operating System & Version |  Linux  | 
| Kernel Version (Linux Only)    |  (5.15.122-tegra)                                        | 
| Platform                                 | NVIDIA Jetson  |
| SDK Version                            |  {2.54.2 }                          | 
| Language                            |  {python }                          | 
| Segment			|  {n/A}                          | 

### Issue Description
I'm trying to run the through Python the D455 camera.

Runs fine in Realsense Viewer. I check this [related issue](https://github.com/IntelRealSense/librealsense/issues/10610), and already rebuilt from scratch using this script:
```
echo "[INFO] pyrealsense2 bindings build has been started!"
sleep 2
# Give the CUDA path to CMake
sed -i '3iset(CMAKE_CUDA_COMPILER /usr/local/cuda/bin/nvcc)\' ../librealsense/CMakeLists.txt
# Set the flags so that librealsense is compatible with python
cmake ../librealsense -DBUILD_PYTHON_BINDINGS:bool=true -DPYTHON_EXECUTABLE=/usr/bin/python3 -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true -DBUILD_WITH_CUDA:bool=true -DFORCE_LIBUVC=true

sudo make uninstall && sudo make clean
echo "[INFO] Building is starting, it will take a long time like half an hour or more!"
sleep 2
sudo make -j$(($(nproc)-1)) && sudo make install

echo 'export PYTHONPATH="$PYTHONPATH:/home/nvidia/Repos/librealsense/build/release"
' >> ~/.bashrc

sudo cp ~/Repos/librealsense/config/99-realsense-libusb.rules /etc/udev/rules.d/ 
sudo cp ~/Repos/librealsense/config/99-realsense-d4xx-mipi-dfu.rules /etc/udev/rules.d/
sudo udevadm control --reload-rules && sudo udevadm trigger
```
I can load `import pyrealsense2 as rs` no problem. When I run this basic script:
```
import pyrealsense2 as rs

pipeline = rs.pipeline()
config = rs.config()

config.enable_stream(rs.stream.depth, 1280, 720, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 1280, 720, rs.format.bgr8, 30)
config.enable_stream(rs.stream.infrared, 1280, 720, rs.format.y8, 30)
# depth_sensor = profile.get_device().first_depth_sensor()

# Start streaming
# profile = config.resolve(pipeline)
profile = pipeline.start(config)
```
I get this error:
```
 -- profile = pipeline.start(config)
RuntimeError: No device connected
```

MartyG-RealSense, 2024-03-29:
Hi @lesreaper  Is the camera detected if you alter your code to use the method shown below, please?

```
import pyrealsense2 as rs

pipeline = rs.pipeline()
config = rs.config()

# Get device product line for setting a supporting resolution
pipeline_wrapper = rs.pipeline_wrapper(pipeline)
pipeline_profile = config.resolve(pipeline_wrapper)
device = pipeline_profile.get_device()

config.enable_stream(rs.stream.depth, 1280, 720, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 1280, 720, rs.format.bgr8, 30)
config.enable_stream(rs.stream.infrared, 1280, 720, rs.format.y8, 30)

# Start streaming
profile = pipeline.start(config)
```

lesreaper, 2024-03-29:
No, it does not.

Which leads me to believe it's something with the Python wrapper. However, not sure how to change that config to work. 

I'd really like to avoid the `RSUSB` [install ](https://dev.intelrealsense.com/docs/nvidia-jetson-tx2-installation)since this is destined for more than a prototype, but I don't know how to get Python wrappers otherwise on a Jetson.

MartyG-RealSense, 2024-03-29:
Using the CMake flag **DFORCE_LIBUVC=true** should be essentially the same as using RSUSB, since the LIBUVC flag was deprecated in favor of using **-DFORCE_RSUSB_BACKEND=TRUE** instead, though the deprecated flag is still valid.  Both of these flags build the SDK in a mode that bypasses the Linux kernel and so avoids the need to apply a kernel patch.

The two approaches for installing successfully from source code on a Jetson are typically to either (1) use the LIBUVC / RSUSB flag, or (2) not use the RSUSB / LIBUVC flag but do **not** apply a kernel patch script.

I note in the title of this issue that you refer to **Jetson 6.0DP**, which is JetPack 6.0 Developer Preview (DP), the first release of JetPack 6. As the librealsense SDK only supports up to JetPack 5.0.2 in the **/patch-realsense-ubuntu-L4T.sh** Jetson patch script at the time of writing this, this makes it important to not apply the patch script.

What happens if you remove **config** from the brackets of the pipeline start line so that the script ignores the stream config lines and applies the camera's default stream profile instead?

`profile = pipeline.start()`

lesreaper, 2024-03-29:
If I pull the config out, it still give me the `No Device Connected` error. 

No, I didn't apply any patches.

I've tried compiling both with and without the `RSUSB / LIBUVC` flag. Same challenge. 

Ideas?

I really don't want to have to, as I get downgraded Python 3.8 and CUDA 11, but I can downgrade the system to Jetpack 5.1.3 and try again.

I need to put these into production for a residential pilot program, so I need to have stable streams coming in. What's the best way of making that happen with the various installation procedures for Jetson, and having a need for Python?

MartyG-RealSense, 2024-03-30:
Have you seen the librealsense / Python source code installation guide for Jetson at https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049

lesreaper, 2024-03-30:
Yes, I did that exact process, I was just wondering if there was a better way to do it so it was more stable. Looks like I'm going to have to downgrade for now and test.

Thank you!!

dmipx, 2024-03-31:
Hi. JP6 kernel has no hidraw support, we enable it with our mipi integration.
https://github.com/IntelRealSense/realsense_mipi_platform_driver/tree/dev

lesreaper, 2024-04-02:
Hey @MartyG-RealSense, yes, that is the exact install I used. Didn't work.

lesreaper, 2024-04-02:
Hey @dmipx , those are interesting for sure, but adding $800 to our Cost of Goods for a camera is going to price us out of the market.

I'm going to downgrade and be back to update shortly.

dmipx, 2024-04-02:
Hi. I'm not asking you to use gmsl camera, I suggested to use jp6.0 kernel we built for and use with USB camera.

lesreaper, 2024-04-02:
Ah, I see @dmipx , thank you for the clarification. I will try that again when I have some extra time hopefully.

For now, I downgraded to 5.1.3, and was able to get it running with a night's worth of work. Hopefully it will continue to operate ok.

Thank you all!!

MartyG-RealSense, 2024-04-03:
You are very welcome, @lesreaper - I'm pleased to hear that you achieved a solution with a downgrade.  Thanks very much for the update!

closed 2024-04-03 by lesreaper

shankar-anantak, 2024-05-17:
Hello, I am facing a similar issue, building from source completes without error, i am able to import pyrealsense2 into python scripts, however, in python, "No Device Detected" error is thrown in every script ive tried

Realsense binaries are working (rs-sensor-control, multicam, etc)

JP6, Nvidia Orin dev kit, librealsense 2.55.1

dmipx, 2024-05-19:
Hi. I suggest you run `./scripts/setup_udev_rules.sh` to properly configure rules.
Report your output of `rs-enum.sh` script and `rs-enumerate-devices`.

jeezrick, 2024-07-03:
> Hi. I'm not asking you to use gmsl camera, I suggested to use jp6.0 kernel we built for and use with USB camera.

Hi, I am not quiet sure what I should to do to 'use jp6.0 kernel' as you said according to the github repo https://github.com/IntelRealSense/realsense_mipi_platform_driver/tree/dev, can you shed some light?

MartyG-RealSense, 2024-07-03:
@jeezrick  Using the MIPI driver on a USB connection with JetPack 6 provides support for the IMU.