realsenseai/librealsense #12809: d415 the relationship between the exposure time on GUI and the rolling shutter speed.

jinshangcheng, 2024-03-29:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (10)  | 
| Kernel Version (Linux Only)    |                                      | 
| Platform                                 | PC  |
| SDK Version                            |                       | 
| Language                            |                          | 
| Segment			|                          | 

### Issue Description
Hello,

I want to know the relationship between exposure time and rolling shutter speed on the GUI. From previous answers, I know t_line=15us. In the GUI, the exposure time can be set to 1us-165000us.

Thanks.

jinshangcheng, 2024-03-29:
D415

MartyG-RealSense, 2024-03-29:
Hi @jinshangcheng  The shutter speed on RealSense cameras is fixed.  The t_line value is calculated using the **Horizontal Total Size (HTS)** for a particular resolution, as described at https://github.com/IntelRealSense/librealsense/issues/2724#issuecomment-442546730

MartyG-RealSense, 2024-03-29:
The maximum depth exposure time was previously 165000 but it can now be set in the RealSense Viewer tool to a range between 1 and 200000 us. So you can set any manual exposure value within this range.  The default value is 33000.

jinshangcheng, 2024-03-29:
thanks martyg

closed 2024-03-29 by jinshangcheng

jinshangcheng, 2024-04-23:
Hi martyg,

Are T_line and T_row the same thing? T_row means the temporal offset between i-th row and i+1-th row. Thanks.
From my experiment, T_row is bigger than 15us under 1280*720 30fps. Can you tell me the exact parameter?

MartyG-RealSense, 2024-04-23:
The librealsense SDK does not have a parameter called **T_row**.  

 T-Line relates to a RealSense sensor's shutter speed.

jinshangcheng, 2024-04-23:
Does T-line mean the temporal offset between i-th row and i+1-th row? Thanks.

MartyG-RealSense, 2024-04-23:
T_line is determined by the Horizontal Total Size (HTS), as described at https://github.com/IntelRealSense/librealsense/issues/2724#issuecomment-442546730