realsenseai/librealsense #12812: Unsupported kernel version 5.10.160 . The Focal patches are maintained for Ubuntu LTS with kernel 5.4, 5.8, 5.11 only

lxd551326, 2024-04-01:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Operating System & Version |   Linux (Ubuntu 20.04  | 
| Kernel Version (Linux Only)    |  5.10.160                               | 
| Platform                                 | arm64  rk3588 |
| SDK Version                            |  2.54.2              |  
| Segment			|  {Robot }                          | 

### Issue Description
i use this script
**Ubuntu 14/16/18/20 with LTS kernel (< 5.13) **
./scripts/patch-realsense-ubuntu-lts.sh

it run has a error that:
Unsupported kernel version 5.10.160 . The Focal patches are maintained for Ubuntu LTS with kernel 5.4, 5.8, 5.11 only

MartyG-RealSense, 2024-04-01:
Hi @lxd551326  For computing devices based on RK3588 it is recommended to build the librealsense SDK from source code with the libuvc or RSUSB installation methods.  These methods bypass the Linux kernel, so that it is not necessary to apply a kernel patch script.

A quick and easy way to install is to use the libuvc build script at the link below.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

MartyG-RealSense, 2024-04-19:
Case closed due to no further comments received.

closed 2024-04-19 by MartyG-RealSense

Justinchan99, 2025-05-27:
Hi！@MartyG-RealSense It works! Thank you very much!

MartyG-RealSense, 2025-05-27:
You are very welcome, @Justinchan99 - I'm pleased to hear that it worked for you!