realsenseai/librealsense #12815: Cannot set auto exposure to True on D405 in Python after FW update to 5.15.1

elishafer, 2024-04-02:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D405 | 
| Firmware Version                   | 5.15.1 | 
| Operating System & Version |   Ubuntu 22.04  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.54.2.5684                        | 
| Language                            |  Python                      | 
| Segment			|  Robot               | 

### Issue Description
When I start the realsense using [this code](https://github.com/rail-berkeley/serl/blob/main/serl_robot_infra/franka_env/camera/rs_capture.py) on a D405 with 5.15.1 I get over-exposed images. When using a D405 that had FW 5.12.14.100, the images were correctly exposed.
When I tried adding
```
depth_sensor = self.profile.get_device().query_sensors()[0]
depth_sensor.set_option(rs.option.enable_auto_exposure, True)
```
The camera froze. After closing the program, I entered realsense viewer and couldn't receive a video stream for the colour image. Only after playing around with the controls/turning the camera off and on again did I get a stream for the colour image.

MartyG-RealSense, 2024-04-02:
Hi @elishafer  Are you using your D405 on a USB 2.1 or USB 3.2 connection, please?  There was a past case at the link below where a D405 experienced over-exposure on USB 2 only.  You can tell which type of connection the camera is being detected as being on by looking at the number beside the camera name at the top of the RealSense Viewer's options side-panel (2.1 or 3.2).

https://support.intelrealsense.com/hc/en-us/community/posts/12707550490259-D405-Color-Instability-on-USB-2-0-and-Outdoor-lighting

I note though that exposure is correct for you on firmware 5.12.14.100.

We have not had a similar case reported where exposure works better on an older firmware on D405.  So if you are confirmed to be on a USB 3.2 connection then I would recommend continuing with the older firmware 5.12.14.100 until the next release of the RealSense SDK and an updated firmware is released to see whether the next firmware resolves your issue.

elishafer, 2024-04-02:
@MartyG-RealSense It's USB3.2, I bought a PCIe card with 5Gb/s per USB port and Delock cables to make sure of it and checked it on the `realsense-viewer`. btw, when two cameras with diffferent FW versions are working at the same time the old FW has good exposure while the old is overexposed.
I didn't know you can downgrade the firmware. I'll check it out, thanks!

elishafer, 2024-04-02:
@MartyG-RealSense Quick question: Do I understand correctly that FW ver 15.12.14.100 the same as 15.13.0.50? [ref](https://dev.intelrealsense.com/docs/firmware-releases#d400-series-firmware-downloads)
Also that there is a limited amount of FW downgrades that I can do?

MartyG-RealSense, 2024-04-02:
5.13.0.50 is 5 months newer than 5.12.14.  Sometimes the changes in firmware versions are very minor.  You should not downgrade your D405 to a version older than the 5.12.14 one that is currently installed.  The ideal situation is to use the firmware recommended for the particular SDK version that you are using, though that is causing you over-exposure on your D405.

There is a limit of 20 downgrades before downgrading becomes locked and you can only upgrade to a newer version than the currently installed one.

elishafer, 2024-04-07:
Just updating that downgrading the FW to 5.13.0.50 worked

MartyG-RealSense, 2024-04-07:
I'm pleased to hear that you achieved a solution, @elishafer - thanks very much for the update!

closed 2024-04-15 by MartyG-RealSense