realsenseai/librealsense #12816: D455 imu frequency

chen579, 2024-04-02:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D455 } | 
| Firmware Version                   | 5.15.0.2 | 
| Operating System & Version |   { Linux (Ubuntu 20.04)   | 
| Kernel Version (Linux Only)    |  (Linux 5.15.0-97-generic x86_64)                                         | 
| Platform                                 | laptop  |
| SDK Version                            |  { legacy / 2.<54>.<2> }                          | 
| Language                            |  {python}                          | 
| Segment			|  {others }                          | 

### Issue Description
<import rospy
from sensor_msgs.msg import Image, Imu
from cv_bridge import CvBridge
import rosbag
import pyrealsense2 as rs
import numpy as np
import time

# Initialize ROS node
rospy.init_node('realsense_bag_recorder')

# Initialize ROS bag
bag = rosbag.Bag('realsense_imu.bag', 'w')

# Initialize RealSense pipeline
pipeline = rs.pipeline()
config = rs.config()
# Configure streams
config.enable_stream(rs.stream.infrared, 1, 640, 480, rs.format.y8, 30)  # Enable left infrared stream
config.enable_stream(rs.stream.infrared, 2, 640, 480, rs.format.y8, 30)  # Enable right infrared stream
config.enable_stream(rs.stream.accel,rs.format.motion_xyz32f, 200)
config.enable_stream(rs.stream.gyro,rs.format.motion_xyz32f, 400)
# Start streaming
pipeline.start(config)

# Create IMU messages buffer
imu_buffer = []
try:
    while not rospy.is_shutdown():
    
    # Wait for a coherent set of frames: depth, color, and infrared
        frames = pipeline.wait_for_frames()
        
        # Get IMU data
        accel_frame = frames.first_or_default(rs.stream.accel)
        gyro_frame = frames.first_or_default(rs.stream.gyro)
        
        # Convert IMU data to ROS message
        imu_msg = Imu()
        imu_msg.header.stamp = rospy.Time.now()
        imu_msg.linear_acceleration.x = accel_frame.as_motion_frame().get_motion_data().x
        imu_msg.linear_acceleration.y = accel_frame.as_motion_frame().get_motion_data().y
        imu_msg.linear_acceleration.z = accel_frame.as_motion_frame().get_motion_data().z
        imu_msg.angular_velocity.x = gyro_frame.as_motion_frame().get_motion_data().x
        imu_msg.angular_velocity.y = gyro_frame.as_motion_frame().get_motion_data().y
        imu_msg.angular_velocity.z = gyro_frame.as_motion_frame().get_motion_data().z
        imu_buffer.append(imu_msg)
        # Process buffered IMU messages
        while imu_buffer:
            imu_msg = imu_buffer.pop(0)  # Get the first message
            # Write IMU messages to bag file
            bag.write('/imu/data', imu_msg)
        #get image
        left_infrared_frame = frames.get_infrared_frame(1)  # Get left infrared frame
        right_infrared_frame = frames.get_infrared_frame(2)  # Get right infrared frame
        # Convert depth and infrared frames to ROS image messages
        bridge = CvBridge()
        left_infrared_image_msg = bridge.cv2_to_imgmsg(np.array(left_infrared_frame.get_data()), encoding="mono8")
        right_infrared_image_msg = bridge.cv2_to_imgmsg(np.array(right_infrared_frame.get_data()), encoding="mono8")
        
        
         # Set timestamps for image messages
        current_time = rospy.Time.now()
        left_infrared_image_msg.header.stamp = current_time
        right_infrared_image_msg.header.stamp = current_time
        # Write image messages to bag file
        bag.write('/camera/left_infrared/image_raw', left_infrared_image_msg)
        bag.write('/camera/right_infrared/image_raw', right_infrared_image_msg)
        

        
except Exception as e:
    print("error:", e)
I try to use the above code to save the left and right infrared images and IMU information of d455 as a bag file, when saving the IMU data separately, the recorded bag file can have about 300hz data, but when the left and right infrared images are introduced and the update frequency of the image is changed to 30hz (the update frequency of the IMU remains unchanged), the update frequency of all the data in the final recorded bag file is 30hz, how to make the IMU update frequency greater than 100hz when ensuring that the update frequency of the image is 30hz>

MartyG-RealSense, 2024-04-02:
Hi @chen579  D455 cameras manufactured after mid 2022 have IMU default values of 100 Accel and 200 Gyro.  It also supports 200 Accel and 400 Gyro.  I see that you have already configured the 200 / 400 pairing in your script, so you cannot make the IMU operate any faster.

Typically, an IMU script will work well when it is using the gyro and accel streams on their own but performance reduces when adding other types of stream such as depth and color.  A solution for this in Python is to configure two separate pipelines, one for the IMU streams and the other for the non-IMU streams.  The best available example script for doing so is at https://github.com/IntelRealSense/librealsense/issues/5628#issuecomment-575943238 - you will need to modify it for use with a bag file instead of a live camera.

chen579, 2024-04-02:
Thank you very much for your prompt reply. However, according to the code you provided for the connection, I conducted tests and the IMU data update rate obtained is still consistent with the image update rate of 30Hz. Afterwards, I analyzed and found that writing both the image and IMU information to the bag file in a loop would cause a decrease in speed. After debugging the code accordingly, it can run normally.Below is my modified code
`import rospy
from sensor_msgs.msg import Image, Imu
from cv_bridge import CvBridge
import rosbag
import pyrealsense2 as rs
import numpy as np
import time
# set
enable_imu = True
enable_image = True
# Initialize ROS node
rospy.init_node('realsense_bag_recorder')
# Initialize ROS bag
bag = rosbag.Bag('realsense_imu.bag', 'w')
if enable_imu:
    imu_pipeline = rs.pipeline()
    imu_config = rs.config()
    imu_config.enable_stream(rs.stream.accel, rs.format.motion_xyz32f, 100) # acceleration
    imu_config.enable_stream(rs.stream.gyro, rs.format.motion_xyz32f, 200)  # gyroscope
    imu_profile = imu_pipeline.start(imu_config)
if enable_image:
    pipeline = rs.pipeline()
    config = rs.config()
    config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
    config.enable_stream(rs.stream.infrared, 1, 640, 480, rs.format.y8, 30)  # Enable left infrared stream
    config.enable_stream(rs.stream.infrared, 2, 640, 480, rs.format.y8, 30)  # Enable right infrared stream
    # Start streaming
    profile = pipeline.start(config)
    # Getting the depth sensor's depth scale
    depth_sensor = profile.get_device().first_depth_sensor()
    depth_scale = depth_sensor.get_depth_scale()
    print("Depth Scale is: ", depth_scale)

# Create IMU messages buffer
imu_buffer = []
try:
    imu_count = 0
    while not rospy.is_shutdown():
        
        imu_frames = imu_pipeline.wait_for_frames()
        # Get IMU data
        accel_frame = imu_frames.first_or_default(rs.stream.accel,rs.format.motion_xyz32f)
        gyro_frame = imu_frames.first_or_default(rs.stream.gyro,rs.format.motion_xyz32f)
        
        # Convert IMU data to ROS message
        imu_msg = Imu()
        imu_msg.header.stamp = rospy.Time.now()
        imu_msg.linear_acceleration.x = accel_frame.as_motion_frame().get_motion_data().x
        imu_msg.linear_acceleration.y = accel_frame.as_motion_frame().get_motion_data().y
        imu_msg.linear_acceleration.z = accel_frame.as_motion_frame().get_motion_data().z
        imu_msg.angular_velocity.x = gyro_frame.as_motion_frame().get_motion_data().x
        imu_msg.angular_velocity.y = gyro_frame.as_motion_frame().get_motion_data().y
        imu_msg.angular_velocity.z = gyro_frame.as_motion_frame().get_motion_data().z
        imu_buffer.append(imu_msg)
        # Process buffered IMU messages
        while imu_buffer:
            imu_msg = imu_buffer.pop(0)  # Get the first message
            # Write IMU messages to bag file
            bag.write('/camera/imu', imu_msg)
            imu_count = imu_count+1
        #
        if imu_count%3==0:
            frames = pipeline.wait_for_frames()
            #get image
            left_infrared_frame = frames.get_infrared_frame(1)  # Get left infrared frame
            right_infrared_frame = frames.get_infrared_frame(2)  # Get right infrared frame
            # Convert depth and infrared frames to ROS image messages
            bridge = CvBridge()
            left_infrared_image_msg = bridge.cv2_to_imgmsg(np.array(left_infrared_frame.get_data()), encoding="mono8")
            right_infrared_image_msg = bridge.cv2_to_imgmsg(np.array(right_infrared_frame.get_data()), encoding="mono8")
            # Set timestamps for image messages
            current_time = rospy.Time.now()
            left_infrared_image_msg.header.stamp = current_time
            right_infrared_image_msg.header.stamp = current_time
            # Write image messages to bag file
            bag.write('/camera/infra1/image_rect_raw', left_infrared_image_msg)
            bag.write('/camera/infra2/image_rect_raw', right_infrared_image_msg)

        
finally:
    # Stop streaming
    pipeline.stop()
    imu_pipeline.stop()
`

MartyG-RealSense, 2024-04-02:
I'm pleased to hear that you were able to develop a script that runs normally after you debugged it.  Thanks very much for sharing your code!

closed 2024-04-17 by MartyG-RealSense