realsenseai/librealsense #12824: Frame didn't arrive within 15000 ROS2

Sol3N1n, 2024-04-03:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400| 
| Firmware Version                   |5.15.1 | 
| Operating System & Version |  Linux Ubunu 22.04 LTS | 
| Kernel Version (Linux Only)    | 5.15.0-1049-raspi  | 
| Platform                                 | Raspberry Pi4 4GB |
| SDK Version                            | 2.54.2| 
| Language                            |  C++ | 
| Segment			|  Robot                     | 

### Issue Description
I am currently working on a robot running a raspberry Pi4 with Ubuntu 22.04. 
I am trying to set up a camera stream from the robot to a server using ROS2 Humble. I am not yet using the depth camera module

I had everything working but at some point the camera started disconnecting once the program was stopped and restarted, saying:
```
Segmentation Fault
```
This to me indicates that there is something wrong with the driver installation, but for the life of me I could not figure it out. I was experiencing some other issues aswell which led me to the decision to reinstall everything from scratch. I had reinstalled everything to get a fresh start to to speak. When I tried running the code, everything seemed to work fine for about 30 seconds to a minute, with the following text appearing on the screen:
```
[INFO] [1712155774.684152821] [FFMPEGEncoder]: enc: libx264 prof:  preset:
[INFO] [1712155774.684362816] [FFMPEGEncoder]: qmax: 10 bitrate: 8242880 gop: 15
```
Then the stream froze and I got an error message on the raspberry pi saying:
```
terminate called after throwing an instance of 'rs2::error
what():  Frame didn't arrive within 15000 
```
Now I have looked through many issues here but I cant seem to find any solution that works for me, any help with this would be appreciated

This is the code i have been running the entire time. A friend suggested that a the framebuffers from the depth module might be overflowing and stopping all other modules, but I doubt that this might be the case
```
#include "cv_bridge/cv_bridge.h"
#include "image_transport/image_transport.hpp"
#include "opencv2/core/mat.hpp"
#include "rclcpp/rclcpp.hpp"
#include <librealsense2/rs.hpp>

int main(int argc, char ** argv)
{
    rclcpp::init(argc, argv);
    rclcpp::NodeOptions options;
    auto node = rclcpp::Node::make_shared("image_publisher", options);
    image_transport::ImageTransport it(node);
    auto pub = it.advertise("camera/image", 1);

    // Open color camera
    rs2::pipeline pipe;
    rs2::config cfg;
    cfg.enable_stream(RS2_STREAM_COLOR, 1280, 720, RS2_FORMAT_BGR8, 30);
    pipe.start(cfg);

    cv::Mat frame;

    rclcpp::WallRate loop_rate(30);
    while (rclcpp::ok()) {
        // Wait for the next set of frames
        auto frames = pipe.wait_for_frames();

        // Get RGB frame
        rs2::video_frame rgb_frame = frames.get_color_frame();

        // Convert RealSense frame to OpenCV Mat
        frame = cv::Mat(cv::Size(1280, 720), CV_8UC3, (void*)rgb_frame.get_data(), cv::Mat::AUTO_STEP);

        auto msg = cv_bridge::CvImage(std_msgs::msg::Header(), "bgr8", frame).toImageMsg();
        pub.publish(msg);

        rclcpp::spin_some(node);
        loop_rate.sleep();
    }

    rclcpp::shutdown();
    return 0;
}
```

MartyG-RealSense, 2024-04-03:
Hi @Sol3N1n  The error **Frame didn't arrive within 15000** indicates that new frames stopped arriving for a period of 15 seconds before the error was triggered.

How does it perform if you reduce the color FPS in your cfg instruction from 30 to 6 (or 5 if you have the D455 camera model).

Sol3N1n, 2024-04-03:
This does fix the issue, but the project requires a framerate of above 20, and most likely at minimum 720p.

Would the issue be caused by the fact that USB3.2 from the camera and the USB3.0 from the raspberry are at such significantly different speeds that it just causes it to freeze?

MartyG-RealSense, 2024-04-04:
In addition to the reduced transmission speed on USB2, the range of supported resolution / FPS combinations on RealSense cameras is also more limited on USB2.   For example, to achieve 30 FPS the resolution would have to be dropped from 1280x720 to 640x480.

On USB 3.2, 30 FPS at 1280x720 is supported.  Pi 4 has a USB 3 port.  So it sounds as though your camera is likely on a USB 3 connection but frame delivery is stopping when you use 30 FPS.  Is your program able to work if you use 15 FPS?

Sol3N1n, 2024-04-04:
15 FPS seems to work okay, I cant seem to get up to the full 15 fps on the receiving side, but that might be unrelated due to network or ROS2 specific issues

closed 2024-04-17 by MartyG-RealSense