realsenseai/librealsense #12826: No Device Detected using Pyrealsense2 but detected on librealsense commands

SereneFeng, 2024-04-03:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | SDK 2.0 | 
| Operating System & Version |  WSL2 - Ubuntu 22.04LTS  |
| Kernel Version (Linux Only)    |    5.15.133.1                                     | 
| Platform                                 | PC  |
| SDK Version                            |   2.0            | 
| Language                            |     NA             | 
| Segment			|  NA                | 

### Issue Description
I have librealsense installed following these instructions (https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md). 
I am trying to be able to access the camera using [example python scripts](https://dev.intelrealsense.com/docs/python2) using pyrealsense2. I'm try to run [python-tutorial-1-depth.py](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/python-tutorial-1-depth.py) but it returns with "No device connected":
[image]
I'm wondering if there is more to installing the library than just `pip install pyrealsense2`. 

Commands like `rs-depth`, `rs-enumerate-devices`, ect. work, so the device is detected and accessible, which makes me think that librealsense is working but not connected to pyrealsense2 correctly / I did not install pyrealsense2 properly.
[image]
[image]

Any suggested would be greatly appreciated!
Also, probably irrelevant but I'm not able to run realsense-viewer as it says "command not found."

MartyG-RealSense, 2024-04-04:
Hi @SereneFeng  Which Python 3 version are you using, please?  At the time of writing this, **pip install pyrealsense2** supports Python 3.7 to 3.11 on a PC.  If you are using Python 3.6 or a version newer than 3.11 then the pyreasense2 wrapper will have to be built from source code instead.

To install examples and tools that use graphics, including realsense-viewer, you should use the CMake flag **-DBUILD_GRAPHICAL_EXAMPLES=TRUE** instead of setting it to false like the instruction page suggests.

MartyG-RealSense, 2024-04-05:
Hi @nutsintheshell  The RealSense ZR300 model is very old and is not compatible with the RealSense SDK 2.0 and so would not be detectable by it.  The pyrealsense2 wrapper would therefore also not work with it.  

The original 1.0 SDK, now known as Legacy Librealsense, is compatible with it.

https://github.com/IntelRealSense/librealsense/tree/v1.12.1

The 1.0 SDK is compatible with the original **pyrealsense** instead of pyrealsense2.

https://github.com/toinsson/pyrealsense

nutsintheshell, 2024-04-08:
> Hi @nutsintheshell The RealSense ZR300 model is very old and is not compatible with the RealSense SDK 2.0 and so would not be detectable by it. The pyrealsense2 wrapper would therefore also not work [...]

thanks for your reply. But another error occurred.
Building wheels for collected packages: pyrealsense
  Building wheel for pyrealsense (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [13 lines of output]
      cc1plus: warning: command-line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++
      In file included from /home/jishengyin/anaconda3/envs/dexteleop/lib/python3.8/site-packages/numpy/core/include/numpy/ndarraytypes.h:1948,
                       from /home/jishengyin/anaconda3/envs/dexteleop/lib/python3.8/site-packages/numpy/core/include/numpy/ndarrayobject.h:12,
                       from /home/jishengyin/anaconda3/envs/dexteleop/lib/python3.8/site-packages/numpy/core/include/numpy/arrayobject.h:5,
                       from pyrealsense/rsutilwrapper.cpp:1272:
      /home/jishengyin/anaconda3/envs/dexteleop/lib/python3.8/site-packages/numpy/core/include/numpy/npy_1_7_deprecated_api.h:17:2: warning: #warning "Using deprecated NumPy API, disable it with " "#define NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION" [-Wcpp]
         17 | #warning "Using deprecated NumPy API, disable it with " \
            |  ^~~~~~~
      pyrealsense/rsutilwrapper.cpp:1278:10: fatal error: rs.h: No such file or directory
       1278 | #include "rs.h"
            |          ^~~~~~
      compilation terminated.
      error: command '/usr/bin/gcc' failed with exit code 1
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for pyrealsense
  Running setup.py clean for pyrealsense
Failed to build pyrealsense
ERROR: Could not build wheels for pyrealsense, which is required to install pyproject.toml-based projects

could you please tell me how to deal with it

MartyG-RealSense, 2024-04-08:
@nutsintheshell  It is difficult to say what the problem may be, as the original pyrealsense wrapper's PyPi repository has not been updated since September 2017.  At that time, the latest supported Python 3 version was 3.6.1 rather than the 3.8 that you are using.

https://pypi.org/project/pyrealsense/

With the current pyrealsense2 wrapper, if a particular Python 3 version is not supported by pip install then the wrapper can be built from source code instead and the wrapper will work with the newer Python version.  This may also be the case with the original pyrealsense wrapper.

KaushalkumarPatel, 2024-04-09:
> Hi @SereneFeng Which Python 3 version are you using, please? At the time of writing this, **pip install pyrealsense2** supports Python 3.7 to 3.11 on a PC. If you are using Python 3.6 or a version [...]

Hi, even in my case where python version is 3.8.10, gives me the same error. When I run the python-tutorial-1-depth.py, get the following error. 

realsense2.py:30: SyntaxWarning: "is" with a literal. Did you mean "=="?
  if y%20 is 19:
No device connected

MartyG-RealSense, 2024-04-09:
Hi @KaushalkumarPatel  Do you also experience problems if you use the pyrealsense2 wrapper's test script at the link below, please?

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python#streaming-using-rspipeline

KaushalkumarPatel, 2024-04-10:
> Hi @KaushalkumarPatel Do you also experience problems if you use the pyrealsense2 wrapper's test script at the link below, please?
> 
> https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python#streaming-using-rspipeline

yes, exact same error which I mentioned in my previous comment.

MartyG-RealSense, 2024-04-10:
@KaushalkumarPatel  Does the problem continue if you unplug the camera, plug it back in and run your script again?

KaushalkumarPatel, 2024-04-11:
> @KaushalkumarPatel Does the problem continue if you unplug the camera, plug it back in and run your script again?

I am using D435 to detect the 3D object using yolov8 and ROS noetic. But the problem is when I launch the rs_camera.launch and my py script which is yolo node to detect the object in ros msg format in which I imported pyrealsense2 as a module, it gives me error from my py script from line `pipeline.start(config)` says that `RuntimeError: No device connected` 

But when I launch only the py script without the pyrealsense2 module than everything works fine. I need this pyrealsense2 to detect the distance. and also need to launch rs_camera.launch because to listen the camera and depth topics.

MartyG-RealSense, 2024-04-11:
The librealsense SDK operates under rules called the Multi-Streaming Model that dictates that only one application at a time can access the same camera sensor.  So if a ROS node was launched first then another process that is launched secondly afterwards may be unable to access the sensor because the first process already has a claim on it.  Until the claim is released, the other process cannot access the sensor.

https://github.com/IntelRealSense/librealsense/blob/master/doc/rs400_support.md#multi-streaming-model

You can detect distance from within the RealSense ROS wrapper instead of using the pyrealsense2 wrapper by using the SDK's provided Python node script **show_center_depth.py**

https://github.com/IntelRealSense/realsense-ros/blob/ros1-legacy/realsense2_camera/scripts/show_center_depth.py

KaushalkumarPatel, 2024-04-12:
OK. Thanks for the link. I understood.

MartyG-RealSense, 2024-04-19:
Hi @KaushalkumarPatel  Do you require further assistance with this case, please?  Thanks!

KaushalkumarPatel, 2024-04-20:
Hi,

Thanks for the msg. Everything is working now. Thanks for the help.

MartyG-RealSense, 2024-04-20:
That's great to hear, @KaushalkumarPatel - thanks very much for the update!

KaushalkumarPatel, 2024-04-22:
Hi @MartyG-RealSense , 

I have one more question. I detected the object and also calculated the distance from camera. Now I am into calculating the roll, pitch angles of the Bounding Box wrt Camera. Is there any resources for D345 ?

MartyG-RealSense, 2024-04-23:
@KaushalkumarPatel  The RealSense project in Python language at the link below sounds like what you are seeking

https://github.com/monkeyrom/pointcloud_by_open3D

closed 2024-05-17 by MartyG-RealSense