realsenseai/librealsense #12827: Non-uniformity in the depth data

A-Robotics-mama, 2024-04-04:
| | |
|---|---|
|**librealsense**|2.54.2 RELEASE|
|**OS**|Linux|
|**Name**|Intel RealSense D435I|
|**Serial Number**|241122073106|
|**Firmware Version**|5.15.1|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|D400|
|**Asic Serial Number**|244523062794|
|**Firmware Update Id**|244523062794|

Please provide a description of the problem
I try to calibrate camera, place it 120 cm from white wall but have not uniform depth data and after on-chip calibration it is the same
[image]
[image]

Please advise

MartyG-RealSense, 2024-04-04:
Hi @A-Robotics-mama  It looks as though the camera is having difficulty accurately analyzing the wall for depth, as you say, since a flat wall surface that is relatively the same distance from the camera would be expected to be mostly the same color (light and dark blue at around 1 meter from the camera).

Instead of using on-chip calibration, please try instead resetting the camera to its factory-new default calibration values using the instructions at https://github.com/IntelRealSense/librealsense/issues/10182#issuecomment-1019854487

MartyG-RealSense, 2024-04-04:
The camera is operating at a high temperature above 40 degrees C, so this could be negatively affecting the depth image.  Is the camera in a hot location, please? 

[image]

If the location is cool then a high temperature could be an indication of a fault in the USB system (either with the USB port or USB cable) that is causing overheating.

A-Robotics-mama, 2024-04-05:
Thank you @MartyG-RealSense for assistance, I will monitor this parameter

closed 2024-04-24 by MartyG-RealSense