realsenseai/librealsense #12828: Use d435 on linux，The usb is not recognized

EthanUNC, 2024-04-04:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435i | 
| Firmware Version                   || 
| Operating System & Version | Ubuntu18.04 | 
| Kernel Version (Linux Only)    |                                        | 
| Platform                                 | PC |
| SDK Version                            |                 | 
| Language                            | python }                          | 
| Segment			|                        | 

### Issue Description
I know when i want to use d435i on linux ,I need get the access,but Linux doesn't recognize my d435..
[image]
[image]

MartyG-RealSense, 2024-04-04:
Hi @EthanUNC  Are you using the camera with the official 1 meter USB cable supplied with it or a cable of your own choice, please?

Has the camera worked previously or has it not been able to be detected since your purchased it?

If it has worked previously, do you know which firmware driver version is currently installed in the camera?

EthanUNC, 2024-04-05:
@MartyG-RealSense  Using the official 1 meter usb cable.and i have run the 3d point cloud example on pycharm successfully.I download the sdk2.0.But i have not downlaod any other driver. I using the official 1 meter usb cable.and i have run the 3d point cloud example on pycharm successfully.I download the sdk2.0.But i have not downlaod any other driver.

MartyG-RealSense, 2024-04-05:
Is the camera detected if you install the SDK using the **libuvc backend** build script at the link below, please?

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

EthanUNC, 2024-04-07:
@MartyG-RealSense thanks for your reply!it is solved now.

closed 2024-04-07 by EthanUNC

MartyG-RealSense, 2024-04-07:
You are very welcome, @EthanUNC - I'm pleased to hear that it worked for you.  Thanks very much for the update!