realsenseai/librealsense #12829: d435i IMU calibration: fixing accelerometer inaccuracies and calibrating twice.

TakShimoda, 2024-04-04:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i  | 
| Firmware Version                   | (5.13.0.50) | 
| Operating System & Version |   Ubuntu 20.04 | 
| Kernel Version (Linux Only)    |  (5.15.0-94-generic)                                         | 
| Platform                                 | PC/NVIDIA Jetson Nano 4GB  |
| SDK Version                            |  2.54.2                          | 
| Language                            |  python                          | 
| Segment			|  Robot                          | 

### Issue Description
Hello, I have numerous d435i cameras mounted on turtlebot3 robots to run VIO algorithms, such as ORB-SLAM3, VINS-Fusion, OKVIS, Kimera-VIO, etc..

The issue is that the the IMU data may be causing issues with the algorithms. For example, ORB-SLAM3 works well with just VO, but crashes with VIO. OKVIS and VINS-fusion sees significant drift (e.g. a circular motion looks like an arc with a very large radius). This is even after doing the camera-IMU calibration with kalibr (including the calculations of allan variance deviations) for each of the robots (I've tested 2 so far with similar results). 

I decided just to check the raw IMU data from one of the camera while it's stationary as follows:

`ros2 launch realsense2_camera rs_launch.py enable_gyro:=true enable_accel:=true enable_infra1:=true enable_infra2:=true gyro_fps:=400.0 accel_fps:=250.0 unite_imu_method:='linear_interpolation'`
`ros2 topic echo /camera/imu sensor_msgs/msg/Imu --csv --qos-history keep_all --qos-reliability reliable > imu.csv`

- When I check the csv file, I noticed the accelerometer data had inaccuracies: x was reading about -0.13, and z about  -0.45, while y was about -9.9. These values should be [x, y, z] = [0, -9.80665, 0]. 
- From here, I ran the [IMU calibration script](https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/RealSense_Depth_D435i_IMU_Calib.pdf?language=en_US) and wrote the results to the camera's EEPROM.
- When I tried again, there were still poor results, where it typically showed x in the range of -0.15 to-0.19, and z from -0.11 to -0.15. y is improved but still around -9.77. I checked the gyroscope's angular velocity values and they were all in the order of magnitude of 10^-6, which I think is fair. I also checked the motion on `rs-motion`, and it seemed reasonable. 
   - I also checked the motion correction module in the realsense-viewer and noticed that although it does a fairly good job of correcting the y-axis with correction (about 9.40 without it vs 9.8 with it), the x and z values are still inaccurate, if not worse: x is about -0.18 with correction and -0.19 without, while z is about -0.14 with correction and about 0.05 without it. 
- I rotated the camera around and gravity did stay consistently around 9.80 (unlike for example [this user](https://github.com/IntelRealSense/librealsense/issues/5177#issuecomment-549790528), and combined with what I see with rs-motion, it seems orientation is fairly accurate, but the camera is reading noticeable acceleration in the x-z directions when it shouldn't, and I think this is causing poor performance for VIO, especially for a ground robot that only moves in the ground plane.

From here, I noticed that my IMU calibration could be improved, because I simply used my hands to try to keep the camera still during calibration at the 6 positions, but saw that [the box](https://github.com/IntelRealSense/librealsense/tree/development/tools/rs-imu-calibration#rs-imu-calibration-tool) should be used to keep all positions perfectly perpendicular. 
- I just had a question before proceeding with the calibration again. When checking the ROS2 imu topic, I noticed it already gives the "corrected" data after accounting for the first calibration I did. From the IMU calibration white paper, the intrinsics are used to correct the raw data with:

[image]
- Would running the already calibrated data as raw measurements into a new calibration test for a new set of intrinsics be the proper way to go, or is there a way to clear the EEPROM to the factory values, and should I calibrate from that? I saw issues such as #3239 and was wondering if this could be an issue.
- Also, what is an acceptable level for the x, y, z values at rest? Should the deviation from the ideal values be in the order of 10^-6 like the gyroscope values?

Thanks.

MartyG-RealSense, 2024-04-05:
Hi @TakShimoda  I recommend repeating the IMU calibration script with the camera placed on the box in order to attempt to obtain a better calibration.  If the results are not an improvement on the calibration that you already have then you have the option to reject the new calibration and try again.

The PDF guide for the IMU calibration tool provides an example output to compare your own xyz values to.

https://www.intelrealsense.com/wp-content/uploads/2020/07/IMU_Calibration_Tool_for_Intel_RealSense-Depth_Cameras_Whitepaper.pdf

[image]

MartyG-RealSense, 2024-04-06:
I looked carefully though the source code of librealsense and the ROS wrapper but did not find code that would clearly achieve the bias result that you want if it was changed, unfortunately.

SLAM is typically more difficult to implement on ROS2.  There is a recent discussion of various approaches to ROS2 SLAM at https://github.com/IntelRealSense/realsense-ros/issues/3046

In regard to IMU data visualization, **imu_tools** for ROS might suit your requirements.  You can find versions of it for ROS1 and ROS2 by selecting the ROS version that you want from the branch drop-down.

https://github.com/CCNYRoboticsLab/imu_tools

The workings of unite_imu_method's linear_interpolation and copy settings are discussed in detail at https://github.com/IntelRealSense/realsense-ros/issues/898

Regarding implementing a separate integration script, I do not have information about that. I do apologize.

TakShimoda, 2024-04-08:
Hello Marty, 

Thanks, I will try out **imu_tools**. It seems the magwick filter implemented there is what I was looking for.  For the discussion on the unite_imu_method, doronhi explains  [here](https://github.com/IntelRealSense/realsense-ros/issues/729),  which is 5 years old so I don't know if it's the same, since in  the issue you mentioned, they seem to have updated the linear interpolation scheme to come in bursts so it's periodic. I also noticed the frequency for the copy method doesn't match the explanation in issue 729 from realsense-ros, since he says it's added, but if that was the case then it would be 400(from gyro) + 250(from accel) = 650Hz, but it remains 400 (same as linear interpolation).

Regards,

MartyG-RealSense, 2024-04-09:
There is not an available explanation for how unite_imu_method functions after the update of its implementation, unfortunately.

In mid 2022 the IMU component in IMU-equipped RealSense cameras was changed.  So cameras manufactured before mid 2022 support 63 or 250 Accel and 200 or 400 Gyro.  Cameras with the different IMU component manufactured after mid 2022 support 100 or 200 Accel and 200 or 400 Gyro.

TakShimoda, 2024-04-09:
Thanks Marty,

I did realize all of our cameras were purchased sometime in 2021 if I recall, so I think we should be using the old IMU. I confirmed with `rs-enumerate-devices` and it supported 63/250Hz and 200/400Hz for accel/gyro respectively. 

Your comment did alert me to the fact that I was probably using the wrong IMU to check parameters when doing the camera-imu calibration with kalibr. I was using the datasheet from the [Bosch BMI055](https://www.mouser.ca/ProductDetail/Bosch-Sensortec/BMI055?qs=d72FGnIDsgRto2b9Ru3roQ%3D%3D) to check against random walk and noise density parameters when performing allan variance analysis. I believe the old IMU as shown from the 2019 white imu calibration white paper is [this one](https://www.st.com/content/ccc/resource/technical/document/application_note/a0/f0/a0/62/3b/69/47/66/DM00119044.pdf/files/DM00119044.pdf/jcr:content/translations/en.DM00119044.pdf). I can't really find the noise parameters I'm looking for on their data sheet which could help me see if the allan variance analysis is accurate before doing the camera-imu calibration.

MartyG-RealSense, 2024-04-10:
Bosch BMI055 is the IMU used by the old cameras.  The post-2022 IMU is Bosch BMI085.  So the datasheet that you have been using should be okay for your cameras.

TakShimoda, 2024-04-10:
Thank you for your help Marty.

closed 2024-04-10 by TakShimoda