realsenseai/librealsense #12830: Record .bag file of RealSense camera with Python

AnoukDeBock, 2024-04-05:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description

Hello, 

My RealSense camera is connected to an Intel NUC mini PC and I want to record .bag files. I don't have direct access to my NUC so I am not able to manually record the .bag file from my RealSense viewer. Therefore I want to write a python script that is able to record and save .bag files at a certain time each day. 

Is it possible to obtain the same information as manually pressing the Realsense Viewer record button compared to control this via a python script? 

Thank you for the response.

MartyG-RealSense, 2024-04-05:
Hi @AnoukDeBock  An approach that you could consider would be to continuously monitor the computer's current time with the **datetime** Python instruction and when the time reaches a certain hour and minute, activate the bag recording code.

The link below has advice about checking hour, minute and second with datetime.

https://phoenixnap.com/kb/get-current-date-time-python

This link provides an example of retrieving hour, minute and second.

```
import datetime
e = datetime.datetime.now()
print ("Current date and time = %s" % e)
print ("Today's date:  = %s/%s/%s" % (e.day, e.month, e.year))
print ("The current time is: = %s:%s:%s" % (e.hour, e.minute, e.second))
```

You can then add code to your script to take an action based on the retrieved hour and minute values when they are in a certain range.

MartyG-RealSense, 2024-04-15:
Hi @AnoukDeBock  Do you require further assistance with this case, please?  Thanks!

AnoukDeBock, 2024-04-16:
Hi @MartyG-RealSense, I don't need further assistance. Thank you for your help!