realsenseai/librealsense #12831: RealSense D456 not working on Jetson Orin Nano with ROS2 packages

Krzo99, 2024-04-05:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D456| 
| Firmware Version                   | 5.15.0.2 | 
| Operating System & Version |  Ubuntu 22.04  | 
| Kernel Version (Linux Only)    | 5.15.122-tegra                                         | 
| Platform                                 | Nvidia Jetson  |
| SDK Version                            |  2.54.1                          | 
| Language                            |  Python launch file         | 
| Segment			|  ROS2 humble, Jetpack 6.0                         | 

### Issue Description
I have an Intel RealSense D456 that I'm trying to setup on Jetson Orin Nano 8GB, using Jetpack 6.0 and ROS2 Humble.

The camera works well on another amd64 PC, running the same operating system, same ROS2 packages and everything.
It can be launched with: `ros2 launch realsense2_camera rs_launch.py` normally. The output is as follows:

```
[realsense2_camera_node-1] [INFO]  [camera.camera]: RealSense ROS v4.54.1
[realsense2_camera_node-1] [INFO]  [camera.camera]: Built with LibRealSense v2.54.1
[realsense2_camera_node-1] [INFO]  [camera.camera]: Running with LibRealSense v2.54.1
[realsense2_camera_node-1] [INFO]  [camera.camera]: Device with serial number xxxxxxxxxxxx was found.

```

When the same launch is made on the Jetson, the camera could not be found:

```
[realsense2_camera_node-1] [INFO]  [camera.camera]: RealSense ROS v4.54.1
[realsense2_camera_node-1] [INFO]  [camera.camera]: Built with LibRealSense v2.54.1
[realsense2_camera_node-1] [INFO]  [camera.camera]: Running with LibRealSense v2.54.1
[realsense2_camera_node-1]  05/04 16:32:48,265 WARNING [281473022093536] (d400-factory.cpp:1195) DS5 group_devices is empty.
[realsense2_camera_node-1] [WARN] [1712327568.265842988] [camera.camera]: No RealSense devices were found!

```

Same behavior is seen when running rs-enumerate-devices, either as sudo or user.

The 2 udev permissions files are in /etc/udev/rules.d/ 

I can't seem to figure out why the same camera works only on the PC.

Thanks for your help!

MartyG-RealSense, 2024-04-05:
Hi @Krzo99  I note that you are using JetPack 6.  The RealSense SDK does not have JetPack 6 support at the time of writing this, as 2.54.1 / 2.54.2 support Jetson 5.0.2 as the latest JetPack.  https://github.com/IntelRealSense/realsense-ros/issues/3015 is an example of camera non-detection on a JetPack 6 Orin Nano.

If you build the SDK from source code with the **RSUSB backend** installation method though then it should be able to work on a JetPack 6 Orin Nano because an RSUSB build of the SDK bypasses the kernel.

Krzo99, 2024-04-06:
Thanks for the reply. If anybody else is looking for the solution (similar to solution in above commend, but with ROS2) . Here it is:

1. Make sure camera is disconnected. Additionaly, just to check installation, you can uninstall all ros realsense wrappers, (Check installed ones with: `apt list --installed | grep realsense`) but make sure to reinstall them at the end, to enable ROS2 publishing.
2. `git clone https://github.com/IntelRealSense/librealsense.git`
3. `cd librealsense`
4. `mkdir build && cd build`
5. `cmake ../ -DFORCE_RSUSB_BACKEND=ON -DBUILD_PYTHON_BINDINGS:bool=true - DPYTHON_EXECUTABLE=/usr/bin/python -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true -DBUILD_WITH_CUDA:bool=true`
6.  `make -j6`
7. `sudo make install` (This will install libs, bins & includes in /usr/local/... If you want to uninstall, just do `sudo make uninstall` in the build folder)
8. `sudo cp ../config/99-realsense-libusb.rules /etc/udev/rules.d/`
9. `sudo udevadm control --reload-rules && sudo udevadm trigger`
10. Add /usr/local/lib to begining(!) of $LD_LIBRARY_PATH. `export LD_LIBRARY_PATH=/usr/local/lib:$LD_LIBRARY_PATH`. This makes sure the new libs are found before broken ROS2 ones.
11. If you uninstalled ROS2 packages at the begining, you can reinstall them and test with: `ros2 launch realsense2_camera rs_launch.py`

closed 2024-04-06 by Krzo99

MartyG-RealSense, 2024-04-07:
Hi @Krzo99  I'm pleased to hear that you achieved a solution.  Thanks so much for sharing the details of it!

Nico-Sander, 2024-04-10:
> Thanks for the reply. If anybody else is looking for the solution (similar to solution in above commend, but with ROS2) . Here it is:
>
>     1. Make sure camera is disconnected. Additionaly, just to [...]

Thanks so much for this solution. I didn't really find any detailed instructions on how to install librealsense using the RSUSB backend in the official librealsense documentation. Espeacially steps 8-10 that you provided really helped. Got everything working now on Jetson AGX Orin running Jetpack 6.0.

Zeki411, 2024-07-14:
Hi, i have a problem while trying this on jetson orin nano.

> Thanks for the reply. If anybody else is looking for the solution (similar to solution in above commend, but with ROS2) . Here it is:
>
>     1. Make sure camera is disconnected. Additionaly, just to [...]

At the step `cmake ../ -DFORCE_RSUSB_BACKEND=ON -DBUILD_PYTHON_BINDINGS:bool=true -DPYTHON_EXECUTABLE=/usr/bin/python -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true -DBUILD_WITH_CUDA:bool=true`, it raises errors as belows:

> -- The CXX compiler identification is GNU 11.4.0
-- The C compiler identification is GNU 11.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Checking internet connection...
-- Internet connection identified
-- Info: REALSENSE_VERSION_STRING=2.55.1
-- Setting Unix configurations
-- Info: Building with CUDA requires CMake v3.8+
-- The CUDA compiler identification is unknown
CMake Error at CMake/cuda_config.cmake:3 (enable_language):
  No CMAKE_CUDA_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "CUDACXX" or the CMake cache entry CMAKE_CUDA_COMPILER to the full
  path to the compiler, or to the compiler name if it is in the PATH.
Call Stack (most recent call first):
  CMake/global_config.cmake:63 (include)
  CMakeLists.txt:46 (global_set_flags)

-- Configuring incomplete, errors occurred!
See also "/home/serene/librealsense/build/CMakeFiles/CMakeOutput.log".
See also "/home/serene/librealsense/build/CMakeFiles/CMakeError.log".

does anyone know to handle the problem ?

MartyG-RealSense, 2024-07-14:
Hi @Zeki411  https://github.com/jetsonhacks/buildLibrealsense2TX/issues/13 has some suggestions for dealing with the **No CMAKE_CUDA_COMPILER could be found** error.

You could also try using the JetsonHacks website's **buildLibrealsense.sh** build script at the link below, which should do the same procedure as the list of commands above.

https://github.com/JetsonHacksNano/installLibrealsense

Jiajian-Chang, 2024-08-29:
> Hi, i have a problem while trying this on jetson orin nano.
>
>
> At the step `cmake ../ -DFORCE_RSUSB_BACKEND=ON -DBUILD_PYTHON_BINDINGS:bool=true -DPYTHON_EXECUTABLE=/usr/bin/python [...]

I had the same issue. My solution is to execute the script without sudo and export CUDACXX=/usr/local/cuda-$version$/bin/nvcc

circuitseeker, 2025-04-15:
The solution is working and camera is now able to detect but,  now the new issue is their is no camera feed in 2D RGB its just shows up the black screen , I have also changed the resolutions and tried every thing possible.

MartyG-RealSense, 2025-04-15:
Hi @circuitseeker  Are you using the RealSense Viewer tool?  If you are, can you confirm please that you have clicked on the '2D' option in the Viewer's top corner to ensure that you are in 2D display mode?  If you were in 3D mode then the display would be black because 3D mode only displays data if depth is also enabled - RGB on its own does not appear in 3D mode.

[image]