realsenseai/librealsense #12833: 软件打不开

biaofeng111000, 2024-04-06:
| | |
|---|---|
|**librealsense**|2.53.1 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
GL_INVALID_VALUE - C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\common\opengl3.cpp:572

```

MartyG-RealSense, 2024-04-06:
Hi @biaofeng111000  Please try using the instructions at https://github.com/IntelRealSense/librealsense/issues/10544#issuecomment-1138569631 to download and run the old RealSense Viewer version 2.21.0 and then closing it and launching the 2.53.1 viewer again to see whether the issue is corrected.

MartyG-RealSense, 2024-04-19:
Hi @biaofeng111000  Do you require further assistance with this case, please?  Thanks!

您好@biaofeng111000 您是否需要对此案的进一步帮助？ 谢谢！

MartyG-RealSense, 2024-04-24:
Case closed due to no further comments received.

由于没有收到进一步的评论，案件已结。

closed 2024-04-24 by MartyG-RealSense