realsenseai/librealsense #12838: Launching multiple L515 cameras at once take time to start the camera

KIL299, 2024-04-08:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  L515 |
| Firmware Version                   | 1.5.8.1 |
| Operating System & Version |   Windows 10 |
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC Intel NUC i5 - 10210 CPU |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  UNITY |
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
I have 8 L515 sensors which are connected to a powered USB hub.
I want to start the sensor at once and capture frames. However, the time required to start all sensors is 25 to 30 secs. Now this affects my user flow. To avoid this i want to start the sensors in minimum time.
I tried starting the devices on thread as well as unity thread but i am not seeing any difference in timings.
**Is there any way to reduce this time?**

When i start the sensor one by on in my application, each sensor is taking different time and it varies from 3 sec to 8 sec.
If i keep connceted only one sensor the time to start that one sensor is 0.5 to 0.7 sec
When i try to start those sensors in intel realsense viewer one by one they seem to take same time to start around 1.5 to 2 sec.

MartyG-RealSense, 2024-04-08:
Hi @KIL299  This is a difficult question as the RealSense Unity wrapper is primarily intended for single-camera use.  It has been possible in the past to use multiple cameras with the wrapper though by defining a separate RsDevice for each camera, as described at the link below.

https://support.intelrealsense.com/hc/en-us/community/posts/360047049173-Community-contribution-for-using-multiple-cameras-in-RealSense-Unity-wrapper

KIL299, 2024-04-08:
Hi @MartyG-RealSense 
I am able to start multiple sensors and have no issues in doing so.
Only that the time required to start the sensors is more than expected.
As described in issue if only one sensor is connected the time to start the sensor is less, but as i go on increasing the sensors the time also increases.
But same behaviour is not being observed in realsense viewer

MartyG-RealSense, 2024-04-08:
I believe that the Viewer (which is based on C++ code rather than Unity's C#) is creating a separate pipeline for each camera.  https://github.com/IntelRealSense/librealsense/issues/3432#issuecomment-472570380 suggests equivalent code for creating one pipeline per camera in C#

KIL299, 2024-04-09:
Hi @MartyG-RealSense , apologies for delayed response.
we tested the CPU and memory usage
Below are the screenshots for the same.
First is the usage while starting the sensor and not the streaming.
[image]

Second is starting the sensor and the streaming
[image]

MartyG-RealSense, 2024-04-09:
It does seem as though most of the available memory on the computer is being consumed.  From the values displayed in the memory column, I would estimate that the particular computer that you are using is equipped with a total of 8 gigabyes memory capacity, which is not a large amount (it sounds like a laptop PC).  

The high memory consumption may not be completely due to the eight cameras as the Unity Editor can be memory-hungry.  You might get better performance if you are able to **Build and Run** the project instead of running it in editor mode with the play triangle at the top of the Unity window (if you are not using Build and Run mode already).

KIL299, 2024-04-09:
Hi @MartyG-RealSense 
I will test it with the build and also test it on another PC with good configuration.
And let you know by tomorrow if any issues identified.

MartyG-RealSense, 2024-04-09:
Thanks very much, @KIL299 - good luck!

MartyG-RealSense, 2024-04-10:
Another way of comparing multiple camera performance outside of Unity to its performance within Unity would be to launch the example program 'rs-multicam' at `C: > Program Files (x86) > Intel RealSense SDK 2.0 > tools`

If the eight L515 display almost immediately in rs-multicam then this would be further evidence that the multicam slowdown is a problem that is specific to Unity.  

A RealSense user at https://github.com/IntelRealSense/librealsense/issues/11910 who was also using eight L515 in the Unity wrapper found that they would stop receiving new frames after some time had passed.

KIL299, 2024-04-10:
Hi @MartyG-RealSense 
We tried it and the time required to start is almost same that is required with in unity
Below we have attached a reference

https://github.com/IntelRealSense/librealsense/assets/129760179/494788ff-96b0-44ea-a6cc-d93b0834e62b

MartyG-RealSense, 2024-04-11:
If this is an L515-specific issue then there is likely to be nothing that can be done, unfortunately.  This is because support for the L515 camera model has ceased and so a fix would not be created for it if there was a bug that only affected L515 cameras.

KIL299, 2024-04-12:
@MartyG-RealSense 
Isn't there anything that we can do to reduce that timing?

MartyG-RealSense, 2024-04-12:
The RealSense Viewer and rs-multicam both create separate pipelines for each attached camera.  But the cameras work normally with the Viewer whilst the timing problem occurs with rs-multicam.  So without knowing what else could be accounting for the difference in performance between the two tools, it would be difficult to know where to start in developing a solution for correcting the timing. There is also the added difficulty that these tools are written in C++ whilst Unity and its programs are C#, so a solution in C++ might not be applicable to Unity.

KIL299, 2024-04-16:
Hi @MartyG-RealSense 
I have one more issue that is not addressed yet.
Let go of the issue where it takes time to start those sensors when trying to start them parallelly.
As i said earlier even if i start them one after the other the time is different.
Is there any possible explanation for this?

MartyG-RealSense, 2024-04-16:
A characteristic of the L515 camera model is that it performs better when the micro-sized USB cable in the base of the camera is inserted in one particular orientation of the two available (USB-C cables are two-way insertion at the micro-sized end).  Does reversing the connector make a difference to start time?

KIL299, 2024-04-16:
Yes i tested it but no any difference in time.

MartyG-RealSense, 2024-04-17:
Is the start-up time normal if only one L515 at a time is tested and the others are unplugged?  If each individual camera tested on its own has a normal start-up time and the time only starts increasing as more cameras are attached to the computer then that would be evidence that adding multiple cameras may be putting a processing strain on your particular model of computer.

KIL299, 2024-04-17:
is the start-up time normal if only one L515 at a time is tested and the others are unplugged?
-> Yes if i connect only one sensor connected, the time required to initilise is 0.6 to 1 sec. Let's say this sensor as 'xyz'

Regarding the second point, if i keep connected all sensor and now when i try to turn on the same above 'xyz' sensor the time is more.
And this happens even if high configuration PC is used.

MartyG-RealSense, 2024-04-18:
Intel's L515 multiple camera white paper explains how available USB data bandwidth reduces and power demands increase as the number of simultaneously active cameras increases above 3 or 4 cameras.  This is why the ability to turn camera capture on and off in sequence is offered instead of having all cameras enabled all the time in order to minimize these demands.

https://dev.intelrealsense.com/docs/lidar-camera-l515-multi-camera-setup#2-multi-camera-considerations

KIL299, 2024-04-18:
Yes we have looked into the documentation for the power supply and usb bandwidth part. 
but here i am trying to start one sensor at a time
And each individual sensor is taking different time to initialise.

MartyG-RealSense, 2024-04-18:
Thanks very much for the clarification.  So you are only using one L515 at a time and not starting 8 cameras one by one from 1 to 8 until all 8 cameras are simultaneously enabled.

What method do you use to stop a camera and start another - for example, using start() and stop() instructions.

How are you accessing specific cameras - by their unique serial number?

KIL299, 2024-04-19:
Yes 
For starting the sensor at first we create a pipepline and then using start method we start that particular sensor.
For stopping a sensor, we use pipeline.stop() method.

And for accessing a particular sensor we are setting device serial number in device configuration of RsDevice.

below are the code snippets:
1. creating a device configuration variable, we are setting the "RequestedSerialNumber" externally to the sensor serial number that we want to start.

public RsConfiguration DeviceConfiguration = new RsConfiguration
    {
        mode = RsConfiguration.Mode.Live,
        RequestedSerialNumber = string.Empty,
        Profiles = new RsVideoStreamRequest[] {
            new RsVideoStreamRequest {Stream = Stream.Depth, StreamIndex = -1, Width = 640, Height = 480, Format = Format.Z16 , Framerate = 30 },
            new RsVideoStreamRequest {Stream = Stream.Infrared, StreamIndex = -1, Width = 640, Height = 480, Format = Format.Y8 , Framerate = 30 },
            new RsVideoStreamRequest {Stream = Stream.Color, StreamIndex = -1, Width = 640, Height = 480, Format = Format.Rgb8 , Framerate = 30 }
        }
    };

2. To start a sensor
void OnEnableThread()
    {
        
        m_pipeline = new Pipeline();
       
        try
        {
            startTime = System.DateTime.Now;
            using (var cfg = DeviceConfiguration.ToPipelineConfig())
            {

                ActiveProfile = m_pipeline.Start(cfg);

            }
            DeviceConfiguration.Profiles = ActiveProfile.Streams.Select(RsVideoStreamRequest.FromProfile).ToArray();

            if (processMode == ProcessMode.Multithread)
            {
                stopEvent.Reset();
                worker = new Thread(WaitForFrames);
                worker.IsBackground = true;
                worker.Start();
            }
            deviceFoundStatus = true;

        }
        catch (Exception)
        {
            deviceFoundStatus = false;
            Debug.Log("device  " + DeviceConfiguration.RequestedSerialNumber + "   not found");
        }

        hasInitilizationDone = true;
    }

3. To stop a sensor
void DeInitializeDevice()
    {
        if(deviceInitializationThread != null && deviceInitializationThread.IsAlive)
        {
            deviceInitializationThread.Abort();
        }
        
        OnNewSample = null;
        // OnNewSampleSet = null;

        if (worker != null)
        {
            stopEvent.Set();
            worker.Join();
        }

        if (Streaming && OnStop != null)
            OnStop();

        if (ActiveProfile != null)
        {
            ActiveProfile.Dispose();
            ActiveProfile = null;
        }

        if (m_pipeline != null)
        {
            // if (Streaming)
            m_pipeline.Stop();
            m_pipeline.Dispose();
            m_pipeline = null;
        }

        Streaming = false;
        OnDeviceDeInitializationDone?.Invoke(DeviceConfiguration.RequestedSerialNumber);
        Debug.Log("device deinitialized");
    }

MartyG-RealSense, 2024-04-19:
If the camera is unplug from the USB port then that should also act as a pipeline close mechanism.  So an alternative way to test whether a pipeline close method would make a difference to timing might be to have all cameras unplugged except one, and when one camera is finished with, unplug it from the USB port and insert the next camera.

KIL299, 2024-04-20:
Hi Marty
This alternative is not the best choice for us.
Because it will hinder the performance of the software.
Plus we need to insert those sensors sequentially within usb hub so as to know that port number where a particular sensor is  connected.

MartyG-RealSense, 2024-04-20:
I did not mean the suggestion as a permanent solution, but just as a one-off to test whether ensuring the pipeline is closed causes an improvement in start-up timing.

KIL299, 2024-04-20:
Yeah that was tested earlier
When we unplug all sensor except one the iniitialisation time significantly reduces, that means when i keep only one sensor connected the initialisation time is less.

MartyG-RealSense, 2024-04-20:
Another way to reset the pipeline in C# without unplugging cameras may be to use **HardwareReset()** to perform a reset of the camera, as shown in a script at https://github.com/IntelRealSense/librealsense/issues/11787#issuecomment-1542096597 - when a camera is reset, it is disconnected and then reconnected.

```
public void HardwareReset()
{            
NativeMethods.rs2_hardware_reset(base.Handle, out object _);
}
```

KIL299, 2024-04-23:
NativeMethods.rs2_hardware_reset(base.Handle, out object _);
 is not present in the current sdk we are using

MartyG-RealSense, 2024-04-23:
This hardware reset mechanism for C# was added to the SDK in February 2019, as described at https://github.com/IntelRealSense/librealsense/issues/2878 and is still in the latest 2.55.1 SDK.

https://github.com/IntelRealSense/librealsense/blob/ff8a9fb213ec1227394de4060743b0ed61171985/wrappers/csharp/Intel.RealSense/Devices/Device.cs#L117

KIL299, 2024-04-24:
We are using UNITY Realsense SDK as our application is developed using UNITY.

KIL299, 2024-04-24:
And for a work around we tried hardware reset from realsense viewer and then start that sensor in our application.
Still no improvement in time.

MartyG-RealSense, 2024-04-24:
I understand that you are using the Unity wrapper.  My reason for mentioning the hardware reset machanism for C# is that if something is accessible in C# scripting in the RealSense SDK then that C# instruction should be accessible from the Unity wrapper too, as advised at https://github.com/IntelRealSense/librealsense/issues/1570 - I note though that performing a reset first in the Viewer did not resolve your issue.

I do not have further suggestions for possible solutions, unfortunately.

KIL299, 2024-04-24:
Thanks for #1570 .

MartyG-RealSense, 2024-05-18:
Hi @KIL299  Were you able to achieve a solution in this case, please?  Thanks!

closed 2024-06-03 by MartyG-RealSense